ptrkrysik/gr-gsm #453: is there anyone have cfile  with contains of A5/3 with kc code already?

arifkyi, 2019-03-15:
i want to test Decode with cfile A5/3 and known KC code, there are  some operators 2G that already use the A5/3 for their 2G decription 
the purpose: i want to test the grgsm_decode with -a 3 (A5/3)  

if someone already have cfile with known Kc code, kindly share with me and i will test it .

thank you very much.

arifkyi, 2019-03-21:
@velichkov maybe you can help?  if you have 'cfile' with A5/3 encryption and kc code?  i just need it to explore the grgsm_decode -e 3  , so i can get experience the decoding of this A5/3 encryption?

velichkov, 2019-03-21:
Hi @arifkyi,

>  if you have 'cfile' with A5/3 encryption and kc code?

Sorry I don't have such capture.

arifkyi, 2019-03-22:
thank you so much Mr. @velichkov   
hello Mr @ptrkrysik , i need you kindly help if you have the 'Cfile'  with A5/3 encryption and with known kc code? i just need it to explore the grgsm_decode -e 3 , so i can get experience and feel the decoding of this A5/3 encryption producing either voice and sms ?

closed 2019-03-29 by arifkyi