ptrkrysik/gr-gsm #454: how do we know if the particular ARFCN downlink is using the Frequency Hoping?

arifkyi, 2019-03-25:
Hello 
how do we know if the particular ARFCN downlink is using the Frequency Hoping?

is it from System Information 1,2,3 or 5/Immediate Assignment  ?
and which particular info  Messages in Wireshark that show that this one is Hoping frequency?

velichkov, 2019-03-25:
Hi @arifkyi 

It's better to ask such questions in our [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

> how do we know if the particular ARFCN downlink is using the Frequency Hoping?
is it from System Information 1,2,3 or 5/Immediate Assignment ?

Yes its from from System Information Type 1, Immediate Assignment and Assignment Command.

> and which particular info Messages in Wireshark that show that this one is Hoping frequency?

List of ARFCNs,  Hopping Channel: Yes, MAIO and HSN.
```
GSM CCCH - System Information Type 1
    L2 Pseudo Length
    .... 0110 = Protocol discriminator: Radio Resources Management messages (0x6)
    Message Type: System Information Type 1
    Cell Channel Description
        00.. 000. = Format Identifier: bit map 0 (0x00)
        List of ARFCNs = 49 48 47 25 24 23 22 21 20 19 18 17
    RACH Control Parameters
    SI 1 Rest Octets
```
```
GSM CCCH - Immediate Assignment
    L2 Pseudo Length
    .... 0110 = Protocol discriminator: Radio Resources Management messages (0x6)
    Message Type: Immediate Assignment
    Page Mode
    Dedicated mode or TBF
    Channel Description
        0100 1... = SDCCH/8 + SACCH/C8 or CBCH (SDCCH/8): 9
        Subchannel: 1
        .... .010 = Timeslot: 2
        100. .... = Training Sequence: 4
        ...1 .... = Hopping Channel: Yes
        Hopping channel MAIO: 4
        HSN: 22
    Request Reference
    Timing Advance
```
```GSM A-I/F DTAP - Assignment Command
    Protocol Discriminator: Radio Resources Management messages (6)
    DTAP Radio Resources Management Message Type: Assignment Command (0x2e)
    Channel Description 2 - Description of the First Channel, after time
        0001 1... = TCH/H + ACCHs: 3
        Subchannel: 1
        .... .001 = Timeslot: 1
        101. .... = Training Sequence: 5
        ...1 .... = Hopping Channel: Yes
        Hopping channel MAIO: 0
        HSN: 2
    Power Command
    Cell Channel Description
        Element ID: 0x62
        00.. 000. = Format Identifier: bit map 0 (0x00)
        List of ARFCNs = 13 3 1
    Channel Mode - Mode of the First Channel(Channel Set 1)
    Mobile Allocation - Mobile Allocation, after time
        Element ID: 0x72
        Length: 1
        Bitmap of increasing ARFCNs included in the Mobile Allocation: 11100000
    MultiRate configuration
````

arifkyi, 2019-03-26:
Hello @velichkov  noted and thanks.

I will try to consult to mailing list as well for such similar question.

Thanks again.

closed 2019-03-26 by arifkyi