ptrkrysik/gr-gsm #455: airprobe compilation issue

nbalkanas, 2019-03-25:
Hi,

My apologies for posting here. Unfortunately airprobe is gone:( Let me know if I should stop using this group.
I tried various gnuradio versions to get them to export gsm group for gsm_receiver.py. Finally tried a bootstrap in airprobe's gsm-receiver. Configure looks better than ever, almost everything checks out:)That's when all hell broke loose, and I got filled with swing errors from src/lib/gsm.cc:

gsm.cc: In function 'void* _p_gr_headTo_p_gr_block(void*, int*)':
gsm.cc:5486:37: error: 'gr_head' was not declared in this scope
     return (void *)((gr_block *)  ((gr_head *) x));
                                     ^
Relevant piece of code in gsm.cc:

/* -------- TYPE CONVERSION AND EQUIVALENCE RULES (BEGIN) -------- */
static void *_p_gr_headTo_p_gr_block(void *x, int *SWIGUNUSEDPARM(newmemory)) {
    return (void *)((gr_block *)  ((gr_head *) x));
}

I started including in all gnuradio header files to satisfy declarations, but I'm not sure this is the right approach. There are too many of them and besides some of the declarations are not in the header files:(

Any help will be appreciated:)
Nikos

velichkov, 2019-03-25:
Hi @nbalkanas,

I just posted a [response](https://groups.google.com/d/msg/gr-gsm/9L3iT2yt_dg/HI8ns-ZABAAJ) in the mailing list

nbalkanas, 2019-03-26:
Thx Vasili,

Issue was caused by what it seems to be a corrupted airprobe build tree. Was working OK, until I ran bootstrap on it and finished it off:(
Downloading the new patched iamckn sources fixed all that:)

nbalkanas, 2019-03-26:
Issue fixed

closed 2019-03-26 by nbalkanas