ptrkrysik/gr-gsm #456: Issue: Decode voice with multirate 20151737d0 is remain hang

arifkyi, 2019-03-29:
Hello,

Need your kindly expertise.
This might similar with #450 , but i already use this argument **--voice-boundary**

**short description of issue**: Decode voice is hang
**Long Description:** Decode voice with this command is running but never finished even i wait until more than 2 hours, no error populated but it's never End it's just remain hamppered like below more than 2 hours.

**Command Line:** grgsm_decode -c voicecall.cfile  -s 2e5 -a 85 -m TCHH -t 4 -e 1 -k '8F71A98356B954D8' --multi-rate=20151737d0 -o rifkyVoice.au.gsm --voice-boundary
Using Volk machine: avx_64_mmx_orc
multi_rate configuration: 20151737d0
Enabled AMR Codecs:
**4,75 kbit/s codec rate: is part of the subset**
**6,90 kbit/s codec rate: is part of the subset**
**7,40 kbit/s codec rate: is part of the subset**

Known:  KC code (seen in the command up) we able to get Assignment Command (of course)
CFile:    can be download in this [link](https://drive.google.com/open?id=1I1KBndeDo81t1WhsTm3XXU_gP6JAmE7I) 

Assignment Command: below also if you want to [downlod](https://drive.google.com/open?id=1M4wGDyYa8pcOwu-MNDSudgeVsmWIUBNb) here in pcap format  in this topic 
(to validate the multi-rate value is correct)

**My question:** 
1. Do i need to update again?  since i saw in Master branch that around 10 days ago there is update with regards to TCHH 

2.The Multirate by wireshark is 20151737d0 for this bit presents are: "7.4, 5.9  and 4.75"
but the one that populated by grgsm_Decode is 6.9 instead of 5.9? 

3. if the answer for question one is YES, let me know the procedure

GSM A-I/F DTAP - Assignment Command
    Protocol Discriminator: Radio Resources Management messages (6)
    DTAP Radio Resources Management Message Type: Assignment Command (0x2e)
    Channel Description 2 - Description of the First Channel, after time
        0001 0... = TCH/H + ACCHs: 2
        Subchannel: 0
        .... .100 = Timeslot: 4
        101. .... = Training Sequence: 5
        ...0 .... = Hopping Channel: No
        ..10 .... = Spare: 0x02
        Single channel ARFCN: 85
    Power Command
    Channel Mode - Mode of the First Channel(Channel Set 1)
        Element ID: 0x63
        Channel Mode: speech full rate or half rate version 3(FR AMR or HR AMR) (65)
    MultiRate configuration
        Element ID: 0x03
        Length: 5
        001. .... = Multirate speech version: Adaptive Multirate speech version 1 (1)
        ...0 .... = NSCB: Noise Suppression Control Bit: Noise Suppression can be used (default) (0)
        .... 0... = ICMI: Initial Codec Mode Indicator: The initial codec mode is defined by the implicit rule provided in 3GPP TS 05.09 (0)
        .... ..00 = Start Mode: 0
        0... .... = 12,2 kbit/s codec rate: is not part of the subset
        .0.. .... = 10,2 kbit/s codec rate: is not part of the subset
        ..0. .... = 7,95 kbit/s codec rate: is not part of the subset
        ...1 .... = 7,40 kbit/s codec rate: is part of the subset
        .... 0... = 6,70 kbit/s codec rate: is not part of the subset
        .... .1.. = 5,90 kbit/s codec rate: is part of the subset
        .... ..0. = 5,15 kbit/s codec rate: is not part of the subset
        .... ...1 = 4,75 kbit/s codec rate: is part of the subset
        ..01 0111 = AMR Threshold: 11.5 dB (23)
        0011 .... = AMR Hysteresis: 1.5 dB (3)
        .... 0111  11.. .... = AMR Threshold: 15.5 dB (31)
        ..01 00.. = AMR Hysteresis: 2.0 dB (4)

velichkov, 2019-03-29:
Hi @arifkyi,

> This might similar with #450 , 

Yes it is.

> but i already use this argument **--voice-boundary**

The **--voice-boundary** is not suppose to fix this.

> **short description of issue**: Decode voice is hang
> **Long Description:** Decode voice with this command is running but never finished even i wait until more than 2 hours, no error populated but it's never End it's just remain hamppered like below more than 2 hours.

Stop it with `Ctrl + C`

> **Command Line:** grgsm_decode -c voicecall.cfile -s 2e5 -a 85 -m TCHH -t 4 -e 1 -k '8F71A98356B954D8' --multi-rate=20151737d0 -o rifkyVoice.au.gsm --voice-boundary

In the above command the sample is rate wrong - it should be `2e6` and not `2e5`

>     1. Do i need to update again?  since i saw in Master branch that around 10 days ago there is update with regards to TCHH

Update regularly and often. The TCHH related change does not solve the hang issue.

> 2.The Multirate by wireshark is 20151737d0 for this bit presents are: "7.4, 5.9 and 4.75"
> but the one that populated by grgsm_Decode is 6.9 instead of 5.9?

Thanks for reporting this, I'm going to fix it. It's just a typo in the log message that does not affect decoding so you could ignore it for now.
https://github.com/ptrkrysik/gr-gsm/blob/57431ed50cb3e2530e6f5337fea9a98c4e6b4b2d/lib/decoding/tch_h_decoder_impl.cc#L180-L182

arifkyi, 2019-03-29:
Thank you very much @velichkov, 

woooww how come i didn;t notice if i put that typo :(  ( i can't forgive my self).

The last but not list.  if i want to update regularly, i just need to rerun these commands below right? or is there any shorter way? : 

**git clone https://git.osmocom.org/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
make
sudo make install
sudo ldconfig**

velichkov, 2019-03-30:
Hi @arifkyi,

> **2.** But the command remain hampered and never close/stop, the last lines like this below (nearly similar with #450)
> Error! frame_nr:752355 mod26:19 fn_is_odd:1 length:-1
> Error! frame_nr:752359 mod26:23 fn_is_odd:0 length:-1
> Error! frame_nr:752364 mod26:2 fn_is_odd:1 length:-1
> Error! frame_nr:752398 mod26:10 fn_is_odd:1 length:-1
> Error! frame_nr:752403 mod26:15 fn_is_odd:0 length:-1
> Error! frame_nr:752407 mod26:19 fn_is_odd:1 length:-1
> 
> is there a way to fix this without Control+C

You could fix the hanging by upgrading gnuradio, it is described in #450. You could ignore the "Error! frame_nr...." messages.
 
> **3.** The last but not list. if i want to update regularly, i just need to rerun these commands below right? or is there any shorter way? :
> 
> **git clone https://git.osmocom.org/gr-gsm cd gr-gsm mkdir build cd build cmake .. mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ make sudo make install sudo ldconfig**

Most of the above commands are only needed the first time. Here is how to do an update.
```
cd gr-gsm/build
git pull
make
sudo make install
```

closed 2023-09-24 by arifkyi