ptrkrysik/gr-gsm #458: Lime SoapySSDPEndpoint failed to join group udp://239.255.255.190:1900

mozoho77, 2019-04-15:
Is Lime mini compatible with gr-gsm?

uild# grgsm_livemon
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

[WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
  setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]

velichkov, 2019-04-15:
Hi @mozoho77,

> Is Lime mini compatible with gr-gsm?

`gr-gsm` uses [gr-osmosdr](https://osmocom.org/projects/gr-osmosdr/wiki) project that have not direct support for Lime devices but `gr-osmosdr` could be compiled with [SoapySDR](https://github.com/pothosware/SoapySDR/wiki) support so in theory there should be support for Lime mini 

I saw you have already opened two other issues regarding this
* https://github.com/pothosware/SoapyUHD/issues/20
* https://github.com/jocover/OpenUSRP/issues/12

> [WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]

See https://stackoverflow.com/a/3188060/2315085

Which OS do you use and which version?
How did you install `gnuradio`, `gr-omsosdr` and `gr-gsm` - from the package repository, manually or using `pybombs`?

Run the following commands and provide the full output
```
SoapySDRUtil --info
SoapySDRUtil --find
SoapySDRUtil --probe
```

guruofquality, 2019-04-15:
> [WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]

FYI, this is probably a network configuration issue, and its a print from SoapyRemote while its trying to discover devices. @mozoho77 isnt using SoapyRemote, so he can ignore this issue or even uninstall SoapyRemote. It should be harmless.

Since gr-gsm using gr-osmosdr, you have two options as far as I can see:

* Build SoapySDR, LimeSuite with SoapySDR support, and gr-osmosdr with SoapySDR support. Make sure these options are compiled in
* Or the UHD path: Build LimeSuite, then build OpenUSRP, and then build gr-osmosdr with UHD support

mozoho77, 2019-04-15:
Gnuradio - apt 
gr-osmosdr - https://github.com/Nuand/gr-osmosdr
gr-gsm https://github.com/ptrkrysik/gr-gsm

cSDR/cmake# SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g9a942698
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libLMS7Support.so (19.01.0-357ad5d)
Module found: /usr/local/lib/SoapySDR/modules0.8/libuhdSupport.solinux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

  (0.3.6-8777242)
Available factories... lime, uhd
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

make# SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

Found device 0
  addr = 24607:1027
  driver = lime
  label = LimeSDR Mini [USB 2.0] x
  media = USB 2.0
  module = FT601
  name = LimeSDR Mini
  serial = x

root@debian:~/CubicSDR/cmake# SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

[INFO] Make connection: 'LimeSDR Mini [USB 2.0] x'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeSDR-Mini
  boardSerialNumber=x
  firmwareVersion=6
  gatewareVersion=1.29
  hardwareVersion=2
  protocolVersion=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
  Registers: BBIC
  GPIOs: MAIN

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=2048]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
  Antennas: NONE, LNAH, LNAL_NC, LNAW, Auto
  Corrections: DC removal, DC offset, IQ balance
  Full gain range: [-12, 61] dB
    TIA gain range: [0, 12] dB
    LNA gain range: [0, 30] dB
    PGA gain range: [-12, 19] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-10, 10] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-1e+07, 1e+07]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-1e+07, 1e+07], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 65] MSps
  Filter bandwidths: [1.4, 130] MHz
  Sensors: lo_locked
  Other Settings:
     * TSP DC Level - Digital DC level in LMS7002M TSP chain.
       [key=TSP_CONST, type=int, range=[0, 32767]]

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CF32, CS12, CS16
  Native format: CS16 [full-scale=2048]
  Stream args:
     * Buffer Length - The buffer transfer size over the link.
       [key=bufferLength, units=samples, default=0, type=int]
     * Link Format - The format of the samples over the link.
       [key=linkFormat, default=CS16, type=string, options=(CS16, CS12)]
     * Skip Calibration - Skip automatic activation calibration.
       [key=skipCal, default=false, type=bool]
  Antennas: NONE, BAND1, BAND2, Auto
  Corrections: DC offset, IQ balance
  Full gain range: [-12, 64] dB
    PAD gain range: [0, 52] dB
    IAMP gain range: [-12, 12] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-10, 10] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-1e+07, 1e+07]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-1e+07, 1e+07], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 65] MSps
  Filter bandwidths: [5, 40], [50, 130] MHz
  Sensors: lo_locked
  Other Settings:
     * TSP DC Level - Digital DC level in LMS7002M TSP chain.
       [key=TSP_CONST, type=int, range=[0, 32767]]

velichkov, 2019-04-15:
Hi @mozoho77,

> gr-osmosdr - https://github.com/Nuand/gr-osmosdr

Why Nuand's version and not Osmocom - git://git.osmocom.org/gr-osmosdr?

When running `grgsm_livemon` (or any of the other gr-gsm tool) it's good to specify the device in the `--args=` option otherwise when there are multiple devices (especially when SoapyAudio is installed) it's going to pick the first device which could be the audio device and it's not going to work.
 
>     * Build SoapySDR, LimeSuite with SoapySDR support, and gr-osmosdr with SoapySDR support. Make sure these options are compiled in

In this case I guess it should be `grgsm_livemon --args=soapy,driver=lime`

>     * Or the UHD path: Build LimeSuite, then build OpenUSRP, and then build gr-osmosdr with UHD support

And in this case it should be 
* `grgsm_livemon --args=uhd,type=lime`
* `grgsm_livemon --args="uhd,name=USB 2.0 (LimeSDR Mini)"`. 

See also https://files.ettus.com/manual/page_identification.html#id_identifying_common

Run the above commands and provide the full output enclosed in 3 back-ticks (`), see https://help.github.com/en/articles/basic-writing-and-formatting-syntax#quoting-code

mozoho77, 2019-04-15:
` grgsm_livemon --args="uhd,name=USB 2.0 (LimeSDR Mini)"
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 357, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 345, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 98, in __init__
    self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + args )
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1173, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: Wrong device arguments specified. Missing nchan?
root@debian:~/gr-osmosdr/build#
`
`
root@debian:~/gr-osmosdr/build# grgsm_livemon --args=uhd,type=lime
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

gr-osmosdr d8014f4d (0.1.5git) gnuradio 3.7.10
built-in source types: file fcd uhd rfspace soapy redpitaya 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 357, in <module>
    main()
  File "/usr/local/bin/grgsm_livemon", line 345, in main
    tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
  File "/usr/local/bin/grgsm_livemon", line 98, in __init__
    self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + args )
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1173, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: LookupError: KeyError: No devices found for ----->
Device Address:
    type: lime`
`root@debian:~/gr-osmosdr/build# grgsm_livemon --args=uhd,24607:1027
linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown

gr-osmosdr d8014f4d (0.1.5git) gnuradio 3.7.10
built-in source types: file fcd uhd rfspace soapy redpitaya 

UHD Error:
    Device discovery error: unknown key format: 24607:1027

UHD Error:
    Device discovery error: unknown key format: 24607:1027

UHD Error:
    Device discovery error: unknown key format: 24607:1027
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

[WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
  setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

-- Using subdev spec '0:0'.

UHD Warning:
    Setting DC offset compensation is not possible on this device.
-- Using format CF32.

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

RtApiAlsa: dump hardware params just after device open:

ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [44100 192000]
PERIOD_TIME: (83 5944309)
PERIOD_SIZE: [16 262144]
PERIOD_BYTES: [128 2097152]
PERIODS: [2 32]
BUFFER_TIME: (166 11888617)
BUFFER_SIZE: [32 524288]
BUFFER_BYTES: [128 2097152]
TICK_TIME: ALL

RtApiAlsa: dump hardware params after installation:

ACCESS:  RW_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 192000
PERIOD_TIME: (10666 10667)
PERIOD_SIZE: 2048
PERIOD_BYTES: 16384
PERIODS: 4
BUFFER_TIME: (42666 42667)
BUFFER_SIZE: 8192
BUFFER_BYTES: 65536
TICK_TIME: 0

RtApiAlsa: dump software params after installation:

tstamp_mode: NONE
tstamp_type: MONOTONIC
period_step: 1
avail_min: 2048
start_threshold: 2048
stop_threshold: -1
silence_threshold: 0
silence_size: 1073741824
boundary: 1073741824`

velichkov, 2019-04-15:
> I removed nuand and installed Osmocom version.
> 
> ```
> linux; GNU C++ version 6.3.0 20170221; Boost_106200; UHD_003.009.005-0-unknown
> 
> Traceback (most recent call last):
>   File "/usr/local/bin/grgsm_livemon", line 357, in <module>
>     main()
>   File "/usr/local/bin/grgsm_livemon", line 345, in main
>     tb = top_block_cls(args=options.args, collector=options.collector, collectorport=options.collectorport, fc=options.fc, gain=options.gain, osr=options.osr, ppm=options.ppm, samp_rate=options.samp_rate, serverport=options.serverport, shiftoff=options.shiftoff)
>   File "/usr/local/bin/grgsm_livemon", line 98, in __init__
>     self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + args )
>   File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1173, in make
>     return _osmosdr_swig.source_make(*args, **kwargs)
> RuntimeError: Wrong device arguments specified. Missing nchan?
> root@debian:~/gr-osmosdr/build#
> ```

Which command is this output from? Please always copy the exact command including all it's arguments.

> ```
> root@debian:~/gr-osmosdr/build# grgsm_livemon --args=uhd,type=lime
> .....................................
>   File "/usr/local/bin/grgsm_livemon", line 98, in __init__
>     self.rtlsdr_source_0 = osmosdr.source( args="numchan=" + str(1) + " " + args )
>   File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 1173, in make
>     return _osmosdr_swig.source_make(*args, **kwargs)
> RuntimeError: LookupError: KeyError: No devices found for ----->
> Device Address:
>     type: lime 
> ```

Make sure that your device is connected and accessible? Also check if `uhd_find_devices` and `uhd_find_devices --args=type=soapy` find the device.  Try with `grgsm_livemon --args=uhd,type=soapy` as well.

> ```
>  root@debian:~/gr-osmosdr/build# grgsm_livemon --args=uhd,24607:1027
> 
> UHD Error:
>     Device discovery error: unknown key format: 24607:1027
> ```

You are not specifying the address correctly, you need `addr=...` in the args, it should be `grgsm_livemon --args=uhd,addr=24607:1027`

velichkov, 2019-04-15:
Have you installed the `SoapyAudio` module or have you removed it at some point?

Some of the logs you have provided indicate that you have `SoapyAudio` and some indicate that you don't. If you don't need `SoapyAudio` for something else then try removing it, check that `SoapySDRUtil --find` does not find audio devices and finds __only__ your Lime mini device and run `grgsm_livemon` without `--args`.

ptrkrysik, 2026-11-09:
This is a gr-osmosdr/SoapySDR configuration question rather than a gr-gsm defect; use a gr-osmosdr build with SoapySDR support and select the device via --args (e.g. soapy,driver=lime). Please reopen with fresh details if you still hit a problem with the current release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik