ptrkrysik/gr-gsm #459: No install method:autotools-bootstrap while installing gr-gsm on Kali

AndrewSto, 2019-04-17:
How to install autotools-bootstrap?

velichkov, 2019-04-17:
Hi @AndrewSto,

> How to install autotools-bootstrap?

It seems you are using the `pybombs` installation method which is the most difficult and error prone. My advice is to try [this](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition) first.

See also https://github.com/ptrkrysik/gr-gsm/issues/400 and https://github.com/ptrkrysik/gr-gsm/issues/459

AndrewSto, 2019-04-17:
When installing, there are problems:
Package 'cmake' has no installation candidate
Package 'autoconf' has no installation candidate
Package 'libtool' has no installation candidate
Package 'pkg-config' has no installation candidate
Unable to locate package libcppunit-dev
Unable to locate package swig
Unable to locate package doxygen
Unable to locate package liblog4cpp5-dev
Package 'python-scipy' has no installation candidate
Unable to locate package gnuradio-dev
Unable to locate package gr-osmosdr
Unable to locate libosmocore-dev

AndrewSto, 2019-04-17:
> Hi @AndrewSto,
>
>
> It seems you are using the `pybombs` installation method which is the most difficult and error prone. My advice is to try [...]

When installing, there are problems:
Package 'cmake' has no installation candidate
Package 'autoconf' has no installation candidate
Package 'libtool' has no installation candidate
Package 'pkg-config' has no installation candidate
Unable to locate package libcppunit-dev
Unable to locate package swig
Unable to locate package doxygen
Unable to locate package liblog4cpp5-dev
Package 'python-scipy' has no installation candidate
Unable to locate package gnuradio-dev
Unable to locate package gr-osmosdr
Unable to locate libosmocore-dev

velichkov, 2019-04-17:
Have you tried the other installation method - https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition?

Could you be more specific and provides all commands that you have executed including all arguments?

If you are following a tutorial or blog post post a link to it.

velichkov, 2019-04-17:
Provide the full output of `apt-cache policy`. I just tested installation in a clean Kali container and it works as expected

```
docker run -it --rm kalilinux/kali-linux-docker /bin/bash
apt-get update
apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev
```

Which Kali version are you using?

AndrewSto, 2019-04-17:
Linux version: 4.19.0-kali3-amd64

For commands to work, do you need to do something beforehand?

velichkov, 2019-04-17:
>  For commands to work, do you need to do something beforehand?

You need the `apt` repositories to be configured correctly and that's why I've asked for the full output of `apt-cache policy`, you could provide all the full output of 
```
cat /etc/apt/sources.list
cat /etc/apt/sources.list.d/*
```

velichkov, 2019-04-17:
> Linux version: 4.19.0-kali3-amd64

This is the kernel version and not the Kali verison, see https://www.kali.org/kali-linux-releases/. It seems you are somewhere between `Kali 2018.4` and `Kali 2019.1`. How did you install it?

AndrewSto, 2019-04-17:
Kali 2019.1

AndrewSto, 2019-04-17:
> > Linux version: 4.19.0-kali3-amd64
> 
> This is the kernel version and not the Kali verison, see https://www.kali.org/kali-linux-releases/. It seems you are somewhere between `Kali 2018.4` and `Kali 2019.1`. How did you install it?

Kali 2019.1

AndrewSto, 2019-04-17:
> Provide the full output of `apt-cache policy`. I just tested installation in a clean Kali container and it works as expected
>
> ```
> docker run -it --rm kalilinux/kali-linux-docker /bin/bash
> apt-get [...]

For teams to work, do you need to do something beforehand?

velichkov, 2019-04-17:
Could you provide the full output of the commands I've asked in the previous posts?
Try running `apt-get upgrade`

AndrewSto, 2019-04-17:
> Provide the full output of `apt-cache policy`. I just tested installation in a clean Kali container and it works as expected
>
> ```
> docker run -it --rm kalilinux/kali-linux-docker /bin/bash
> apt-get [...]

Reading package lists... Done
Building dependency tree 
Reading state information... Done
Package autoconf is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libtool is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package pkg-config is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package cmake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-scipy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'cmake' has no installation candidate
E: Package 'autoconf' has no installation candidate
E: Package 'libtool' has no installation candidate
E: Package 'pkg-config' has no installation candidate
E: Unable to locate package libcppunit-dev
E: Unable to locate package swig
E: Unable to locate package doxygen
E: Unable to locate package liblog4cpp5-dev
E: Package 'python-scipy' has no installation candidate
E: Unable to locate package gnuradio-dev
E: Unable to locate package gr-osmosdr
E: Unable to locate package libosmocore-dev

AndrewSto, 2019-04-17:
Package files:
 100 /var/lib/dpkg/status
        release a=now
Pinned packages:

AndrewSto, 2019-04-17:
> Could you provide the full output of the commands I've asked in the previous posts?
> Try running `apt-get upgrade`

Reading package lists... Done
Building dependency tree 
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

velichkov, 2019-04-17:
> Package files:
> 100 /var/lib/dpkg/status
> release a=now
> Pinned packages:

It seems you don't have any repository in `/etc/apt/sources.list`. How did you install Kali? Are you using a live ISO image or something? Execute the following
```
echo "deb http://http.kali.org/kali kali-rolling main contrib non-free" >> /etc/apt/sources.list
apt-get update
apt-get upgrade
```
and provide the full output.

AndrewSto, 2019-04-17:
root@kali:~# echo "deb http://http.kali.org/kali kali-rolling maincontrib non-free" » /etc/apt/sources.list
root@kali:~# apt-get update
Ign:1 https://download.docker.com/linux/debian kali-rolling InRelease
Err:2 https://download.docker.com/linux/debian kali-rolling Release
404 Not Found [IP: 52.85.239.19 443]
Get:3 http://mirror-1.truenetwork.ru/kali kali-rolling InRelease [30.5 kB]
Get:4 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 Packages [187 kB]
Reading package lists... Done 
E: The repository 'https://download.docker.com/linux/debian kali-rolling Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Skipping acquire of configured file 'maincontrib/binary-amd64/Packages' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/i18n/Translation-en_US' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/i18n/Translation-en' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/Components-amd64.yml' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/icons-48x48.tar' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/icons-64x64.tar' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
libwhisker2-perl
Use 'apt autoremove' to remove it.
The following packages have been kept back:
wpscan
The following packages will be upgraded:
firmware-zd1211 intel-microcode nikto
3 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Need to get 2,193 kB of archives.
After this operation, 563 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 firmware-zd1211 all 1:1.5-6 [14.7 kB]
Get:2 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 intel-microcode amd64 3.20190312.1 [1,765 kB]
Get:3 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 nikto all 1:2.1.6+git20190310-0kali2 [413 kB]
Fetched 2,193 kB in 14s (162 kB/s) 
Reading changelogs... Done
(Reading database ... 374598 files and directories currently installed.)
Preparing to unpack .../firmware-zd1211_1%3a1.5-6_all.deb ...
Unpacking firmware-zd1211 (1:1.5-6) over (1:1.5-4) ...
Preparing to unpack .../intel-microcode_3.20190312.1_amd64.deb ...
Unpacking intel-microcode (3.20190312.1) over (3.20180807a.2) ...
Preparing to unpack .../nikto_1%3a2.1.6+git20190310-0kali2_all.deb ...
Unpacking nikto (1:2.1.6+git20190310-0kali2) over (1:2.1.6+git20150709-0kali1) ...
Setting up intel-microcode (3.20190312.1) ...
update-initramfs: deferring update (trigger activated)
intel-microcode: microcode will be updated at next boot
Setting up firmware-zd1211 (1:1.5-6) ...
Setting up nikto (1:2.1.6+git20190310-0kali2) ...
Installing new version of config file /etc/nikto.conf ...
Processing triggers for man-db (2.8.5-1) ...
Processing triggers for initramfs-tools (0.132) ...
update-initramfs: Generating /boot/initrd.img-4.19.0-kali3-amd64

AndrewSto, 2019-04-17:
I use linux installed on virtual box.

velichkov, 2019-04-17:
> I use linux installed on virtual box.

Could please be more specific and provide the link to the image you have downloaded and all the commands that you have executed? If you have followed a tutorial or blog post or something provide a link to it.

After `apt-get upgrade` finishes you could try again installing `gr-gsm` by following [this](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)

AndrewSto, 2019-04-18:
https://cdimage.kali.org/kali-2019.1a/kali-linux-2019.1a-amd64.iso - Linux
https://www.rtl-sdr.com/rtl-sdr-tutorial-analyzing-gsm-with-airprobe-and-wireshark/ - blog

AndrewSto, 2019-04-18:
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
libwhisker2-perl
Use 'apt autoremove' to remove it.
The following packages have been kept back:
wpscan
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python-gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
Hit:1 http://mirror-1.truenetwork.ru/kali kali-rolling InRelease
Ign:2 https://download.docker.com/linux/debian kali-rolling InRelease
Err:3 https://download.docker.com/linux/debian kali-rolling Release
404 Not Found [IP: 13.32.56.60 443]
Reading package lists... Done
W: Skipping acquire of configured file 'maincontrib/binary-amd64/Packages' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/i18n/Translation-en_US' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/i18n/Translation-en' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/Components-amd64.yml' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/icons-48x48.tar' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/icons-64x64.tar' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
E: The repository 'https://download.docker.com/linux/debian kali-rolling Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

AndrewSto, 2019-04-18:
> > I use linux installed on virtual box.
> 
> Could you please be more specific and provide a link to the image you have downloaded and all the commands that you have executed? If you have followed a tutorial or blog post or something provide a link to it.
> 
> After `apt-get upgrade` finishes you could try again installing `gr-gsm` by following [this](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition)

root@kali:# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
libwhisker2-perl
Use 'apt autoremove' to remove it.
The following packages have been kept back:
wpscan
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@kali:# sudo apt-get update && \

sudo apt-get install -y 
cmake 
autoconf 
libtool 
pkg-config 
build-essential 
python-docutils 
libcppunit-dev 
swig 
doxygen 
liblog4cpp5-dev 
python-scipy 
python-gtk2 
gnuradio-dev 
gr-osmosdr 
libosmocore-dev
Hit:1 http://mirror-1.truenetwork.ru/kali kali-rolling InRelease
Ign:2 https://download.docker.com/linux/debian kali-rolling InRelease
Err:3 https://download.docker.com/linux/debian kali-rolling Release
404 Not Found [IP: 13.32.56.60 443]
Reading package lists... Done
W: Skipping acquire of configured file 'maincontrib/binary-amd64/Packages' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/i18n/Translation-en_US' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/i18n/Translation-en' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/Components-amd64.yml' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/icons-48x48.tar' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
W: Skipping acquire of configured file 'maincontrib/dep11/icons-64x64.tar' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)
E: The repository 'https://download.docker.com/linux/debian kali-rolling Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

velichkov, 2019-04-19:
> Ign:2 https://download.docker.com/linux/debian kali-rolling InRelease
Err:3 https://download.docker.com/linux/debian kali-rolling Release

Have you tried to install docker and if yes how? Try to disable the docker apt repository as they don't provide support for Kali. Search for `download.docker.com` in `/etc/apt/sources.list` and in all files in the `/etc/apt/sources.list.d/` directory and comment them out.

> W: Skipping acquire of configured file 'maincontrib/binary-amd64/Packages' as repository 'http://http.kali.org/kali kali-rolling InRelease' doesn't have the component 'maincontrib' (component misspelt in sources.list?)

Open `/etc/apt/sources.list` and make sure that there is a space between `main` and `contrib` as stated in the above warning message.

AndrewSto, 2019-04-19:
> Have you tried to install docker and if yes how? Try to disable the docker apt repository as they don't provide support for Kali. Search for `download.docker.com` in `/etc/apt/sources.list` and in [...]

How to install docker?

velichkov, 2019-04-19:
> How to install docker?

You don't have to. Somehow you have added docker repositories so I thought you have tried to install it. Remove all docker repositories from the sources.list and sources.list.d/* files and try again.

AndrewSto, 2019-04-19:
> > How to install docker?
> 
> You don't have to. Somehow you have added docker repositories so I thought you have tried to install it. Remove all docker repositories from the sources.list and sources.list.d/* files and try again.

Nothing happens. Is there any complete installation instruction? I want to do it all over again.

velichkov, 2019-04-19:
> > > How to install docker?
> > 
> > 
> > You don't have to. Somehow you have added docker repositories so I thought you have tried to install it. Remove all docker repositories from the sources.list and sources.list.d/* files and try again.
> 
> Nothing happens. 

What do you mean by "Nothing happens"? I've already asked you to be more specific and provide all commands with all arguments and their full output. Don't make me ask again.

> Is there any complete installation instruction? I want to do it all over again.

https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

velichkov, 2019-04-19:
Here are some video tutorials:

https://www.youtube.com/watch?v=tl3fc6rzMQo&list=PLRovDyowOn5F_TFotx0n8A79ToZYD2lOv&index=3
https://www.youtube.com/watch?v=LHxFL2eqBko&list=PLRovDyowOn5F_TFotx0n8A79ToZYD2lOv&index=4
https://www.youtube.com/watch?v=ZGVwRxwhv-c&list=PLRovDyowOn5F_TFotx0n8A79ToZYD2lOv&index=5

AndrewSto, 2019-04-20:
> What do you mean by "Nothing happens"? I've already asked you to be more specific and provide all commands with all arguments and their full output. Don't make me ask again.

root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python-gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
Reading package lists... Done
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Package autoconf is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libtool is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package pkg-config is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package cmake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-scipy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'cmake' has no installation candidate
E: Package 'autoconf' has no installation candidate
E: Package 'libtool' has no installation candidate
E: Package 'pkg-config' has no installation candidate
E: Unable to locate package libcppunit-dev
E: Unable to locate package swig
E: Unable to locate package doxygen
E: Unable to locate package liblog4cpp5-dev
E: Package 'python-scipy' has no installation candidate
E: Unable to locate package gnuradio-dev
E: Unable to locate package gr-osmosdr
E: Unable to locate package libosmocore-dev
root@kali:~# apt-cache policy
Package files:
100 /var/lib/dpkg/status
release a=now
Pinned packages:
root@kali:~# cat /etc/apt/sources.list
# 

# deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

#deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

# This system was installed using small removable media
# (e.g. netinst, live or single CD). The matching "deb cdrom"
# entries were disabled at the end of the installation process.
# For information about how to configure apt package sources,
# see the sources.list(5) manual.
root@kali:~# cat /etc/apt/sources.list.d/*
cat: '/etc/apt/sources.list.d/*': No such file or directory
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@kali:~# echo "deb http.kali.org/kali kali-rolling main contrib non-free" » /etc/apt/sources.list
root@kali:~# apt-get update
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# apt-get upgrade
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~#

AndrewSto, 2019-04-20:
> What do you mean by "Nothing happens"? I've already asked you to be more specific and provide all commands with all arguments and their full output. Don't make me ask again. [...]

root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python-gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
Reading package lists... Done
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Package autoconf is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libtool is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package pkg-config is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package cmake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-scipy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'cmake' has no installation candidate
E: Package 'autoconf' has no installation candidate
E: Package 'libtool' has no installation candidate
E: Package 'pkg-config' has no installation candidate
E: Unable to locate package libcppunit-dev
E: Unable to locate package swig
E: Unable to locate package doxygen
E: Unable to locate package liblog4cpp5-dev
E: Package 'python-scipy' has no installation candidate
E: Unable to locate package gnuradio-dev
E: Unable to locate package gr-osmosdr
E: Unable to locate package libosmocore-dev
root@kali:~# apt-cache policy
Package files:
100 /var/lib/dpkg/status
release a=now
Pinned packages:
root@kali:~# cat /etc/apt/sources.list
# 

# deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

#deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

# This system was installed using small removable media
# (e.g. netinst, live or single CD). The matching "deb cdrom"
# entries were disabled at the end of the installation process.
# For information about how to configure apt package sources,
# see the sources.list(5) manual.
root@kali:~# cat /etc/apt/sources.list.d/*
cat: '/etc/apt/sources.list.d/*': No such file or directory
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@kali:~# echo "deb http.kali.org/kali kali-rolling main contrib non-free" » /etc/apt/sources.list
root@kali:~# apt-get update
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# apt-get upgrade
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~#

AndrewSto, 2019-04-20:
root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python-gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
Reading package lists... Done
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Package autoconf is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libtool is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package pkg-config is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package cmake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package python-scipy is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'cmake' has no installation candidate
E: Package 'autoconf' has no installation candidate
E: Package 'libtool' has no installation candidate
E: Package 'pkg-config' has no installation candidate
E: Unable to locate package libcppunit-dev
E: Unable to locate package swig
E: Unable to locate package doxygen
E: Unable to locate package liblog4cpp5-dev
E: Package 'python-scipy' has no installation candidate
E: Unable to locate package gnuradio-dev
E: Unable to locate package gr-osmosdr
E: Unable to locate package libosmocore-dev
root@kali:~# apt-cache policy
Package files:
100 /var/lib/dpkg/status
release a=now
Pinned packages:
root@kali:~# cat /etc/apt/sources.list
# 

# deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

#deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

# This system was installed using small removable media
# (e.g. netinst, live or single CD). The matching "deb cdrom"
# entries were disabled at the end of the installation process.
# For information about how to configure apt package sources,
# see the sources.list(5) manual.
root@kali:~# cat /etc/apt/sources.list.d/*
cat: '/etc/apt/sources.list.d/*': No such file or directory
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree 
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@kali:~# echo "deb http.kali.org/kali kali-rolling main contrib non-free" » /etc/apt/sources.list
root@kali:~# apt-get update
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# apt-get upgrade
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# sudo apt-get update && \
> sudo apt-get install -y \
> cmake \
> autoconf \
> libtool \
> pkg-config \
> build-essential \
> python-docutils \
> libcppunit-dev \
> swig \
> doxygen \
> liblog4cpp5-dev \
> python-scipy \
> python gtk2 \
> gnuradio-dev \
> gr-osmosdr \
> libosmocore-dev
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~#

velichkov, 2019-04-20:
> root@kali:# echo "deb http.kali.org/kali kali-rolling main contrib non-free" » /etc/apt/sources.list

This line is incorrect, it seems you didn't copy-pasted the command but retyped it put an incorrect symbol `»` , this should be two greater than signs `>` put together one after another.

> root@kali:# apt-get upgrade
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.

Open `/etc/apt/sources.list` with your favorite text editor,  **delete** everything and insert only the following line exactly as you see it, the white spaces are important. The whole content should be just this one line, nothing else.
```
deb http://http.kali.org/kali kali-rolling main contrib non-free
```
and then `apt-get update` and `apt-get upgrade` should succeed. 

Also when posting outputs please enclose them with triple back-ticks symbols ` ``` ` to make the output more readable. See https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code-and-syntax-highlighting

AndrewSto, 2019-04-20:
```
root@kali:~# echo "deb http://http.kali.org/kali kali-rolling main coroot@kali:~# echo "deb http://http.kali.org/kali kali-rolling main contrib non-free" >> /etc/apt/sources.list
root@kali:~# apt-get update
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# apt-get upgrade
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
E: Malformed entry 12 in list file /etc/apt/sources.list (URI parse)
E: The list of sources could not be read.
root@kali:~# cat /etc/apt/sources.list
# 

# deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

#deb cdrom:[Debian GNU/Linux 2019.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20190228-17:12]/ kali-last-snapshot contrib main non-free

# This system was installed using small removable media
# (e.g. netinst, live or single CD). The matching "deb cdrom"
# entries were disabled at the end of the installation process.
# For information about how to configure apt package sources,
# see the sources.list(5) manual.
deb http.kali.org/kali kali-rolling main contrib non-free
deb http://http.kali.org/kali kali-rolling main contrib non-free
root@kali:~# gedit /etc/apt/sources.list
root@kali:~# apt-get update
Get:1 http://mirror-1.truenetwork.ru/kali kali-rolling InRelease [30.5 kB]
Get:2 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 Packages [17.1 MB]
Get:3 http://mirror-1.truenetwork.ru/kali kali-rolling/contrib amd64 Packages [105 kB]
Get:4 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 Packages [187 kB]
Fetched 17.4 MB in 3min 8s (92.5 kB/s)                              
Reading package lists... Done
```

AndrewSto, 2019-04-20:
```root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gdal-bin gdal-data libaec0 libarmadillo9 libarpack2
  libboost-program-options1.67.0 libboost-serialization1.67.0
  libboost-test1.67.0 libboost-timer1.67.0 libcgal13 libcharls1
  libcrypt2 libdap25 libdapclient6v5 libdapserver7v5 libepsilon1
  libfcgi-bin libfcgi0ldbl libfreexl1 libfyba0 libgdal20
  libgeotiff2 libhdf4-0-alt libhdf5-100 libkmlbase1
  libkmlconvenience1 libkmldom1 libkmlengine1 libkmlregionator1
  libkmlxsd1 liblwgeom-2.5-0 liblwgeom-dev libnetcdf13 libogdi3.2
  libpoppler80 libpyside1.2 libqca2 libqca2-plugins
  libqgis-analysis2.18.25 libqgis-core2.18.25 libqgis-customwidgets
  libqgis-gui2.18.25 libqgis-networkanalysis2.18.25
  libqgis-server2.18.25 libqgispython2.18.25 libqhull7 libqtwebkit4
  libqwt6abi1 libsfcgal1 libshiboken1.2v5 libspatialindex4v5
  libspatialite7 libsuperlu5 libsz2 liburiparser1 libwhisker2-perl
  libxerces-c3.2 odbcinst odbcinst1debian2
  python-backports.functools-lru-cache python-cycler python-gdal
  python-kiwisolver python-matplotlib python-owslib python-pyproj
  python-pyside.qtcore python-pyside.qtgui python-pyside.qtnetwork
  python-pyside.qtwebkit python-pyspatialite python-qgis
  python-qgis-common python-qt4-sql python-shapely
  python-subprocess32 qt4-designer ruby-faraday
Use 'apt autoremove' to remove them.
The following packages have been kept back:
  beef-xss fwupd fwupd-amd64-signed gdal-bin gjs gnome-shell
  gnome-shell-common gnome-shell-extension-desktop-icons
  gnome-sushi initramfs-tools initramfs-tools-core irqbalance
  kali-linux-full king-phisher libgdal20 libgjs0g libldb1
  libmariadb3 libnetcdf13 libpython3-dev libpython3-stdlib
  libqgis-customwidgets libradare2-dev libsmbclient libwbclient0
  linux-headers-amd64 linux-image-amd64 nodejs python-bs4
  python-faraday python-gdal python-impacket python-ldb
  python-matplotlib python-pil python-pil.imagetk python-pyv8
  python-qgis python-qgis-common python-samba python-webargs
  python3 python3-apt python3-brlapi python3-brotli python3-bs4
  python3-cffi-backend python3-dbus python3-dev python3-distutils
  python3-gst-1.0 python3-lib2to3 python3-lxml python3-markupsafe
  python3-matplotlib python3-minimal python3-mpltoolkits.basemap
  python3-mysqldb python3-numpy python3-pil python3-psycopg2
  python3-pycurl python3-pyproj python3-scipy python3-sip
  python3-sqlalchemy python3-sqlalchemy-ext python3-tk
  python3-tornado python3-yaml radare2 ruby-activesupport ruby-i18n
  ruby-rack ruby-sinatra samba samba-common samba-common-bin
  samba-dsdb-modules samba-libs samba-vfs-modules smbclient sslyze
  usbutils va-driver-all winexe
The following packages will be upgraded:
  accountsservice acl afflib-tools alsa-tools android-libbacktrace
  android-libbase android-libcutils android-liblog android-libutils
  android-libziparchive apache2 apache2-bin apache2-data
  apache2-utils apparmor apt apt-listchanges apt-utils arj aspell
  at-spi2-core attr b43-fwcutter base-passwd bash bash-completion
  bind9-host binfmt-support binutils binutils-common
  binutils-x86-64-linux-gnu braa bubblewrap build-essential bundler
  busybox ca-certificates ca-certificates-java cewl cheese-common
  cherrytree chkrootkit chrome-gnome-shell chromium chromium-common
  chromium-driver chromium-sandbox clang-7 commix console-setup
  console-setup-linux coreutils cpp cpp-8 cracklib-runtime cron
  cups-bsd cups-client cups-common curl cython debconf debconf-i18n
  debian-faq desktop-base dh-python diffutils dmeventd dmsetup
  dns-root-data dnsrecon dnsutils docutils-common docutils-doc dpkg
  dpkg-dev dradis easy-rsa evince evince-common
  evolution-data-server evolution-data-server-common ewf-tools
  exe2hexbat exim4-base exim4-config exim4-daemon-light exploitdb
  fern-wifi-cracker file findutils firebird3.0-common
  firebird3.0-common-doc firefox-esr firmware-b43-installer
[... 2009 lines cut ...]
Err:948 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 wireshark amd64 2.6.8-1
  404  Not Found [IP: 94.247.111.11 80]
Err:949 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 wpscan all 3.5.1-0kali1
  404  Not Found [IP: 94.247.111.11 80]
Err:950 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xbrlapi amd64 5.6-10
  404  Not Found [IP: 94.247.111.11 80]
Err:951 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xfonts-base all 1:1.0.5
  404  Not Found [IP: 94.247.111.11 80]
Err:952 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-input-libinput amd64 0.28.2-2
  404  Not Found [IP: 94.247.111.11 80]
Err:953 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-video-amdgpu amd64 18.1.99+git20190207-1
  404  Not Found [IP: 94.247.111.11 80]
Err:954 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-video-radeon amd64 1:18.1.99+git20190207-1
  404  Not Found [IP: 94.247.111.11 80]
Err:955 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-video-ati amd64 1:18.1.99+git20190207-1
  404  Not Found [IP: 94.247.111.11 80]
Err:956 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-video-nouveau amd64 1:1.0.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:957 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 zsh amd64 5.7.1-1
  404  Not Found [IP: 94.247.111.11 80]
Err:958 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 zsh-common all 5.7.1-1
  404  Not Found [IP: 94.247.111.11 80]
Err:959 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 liburcu6 amd64 0.10.2-1
  404  Not Found [IP: 94.247.111.11 80]
Fetched 96.8 MB in 21min 8s (76.3 kB/s) ```

AndrewSto, 2019-04-20:
```
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gdal-bin gdal-data libaec0 libarmadillo9 libarpack2
  libboost-program-options1.67.0 libboost-serialization1.67.0
  libboost-test1.67.0 libboost-timer1.67.0 libcgal13 libcharls1
  libcrypt2 libdap25 libdapclient6v5 libdapserver7v5 libepsilon1
  libfcgi-bin libfcgi0ldbl libfreexl1 libfyba0 libgdal20
  libgeotiff2 libhdf4-0-alt libhdf5-100 libkmlbase1
  libkmlconvenience1 libkmldom1 libkmlengine1 libkmlregionator1
  libkmlxsd1 liblwgeom-2.5-0 liblwgeom-dev libnetcdf13 libogdi3.2
  libpoppler80 libpyside1.2 libqca2 libqca2-plugins
  libqgis-analysis2.18.25 libqgis-core2.18.25 libqgis-customwidgets
  libqgis-gui2.18.25 libqgis-networkanalysis2.18.25
  libqgis-server2.18.25 libqgispython2.18.25 libqhull7 libqtwebkit4
  libqwt6abi1 libsfcgal1 libshiboken1.2v5 libspatialindex4v5
  libspatialite7 libsuperlu5 libsz2 liburiparser1 libwhisker2-perl
  libxerces-c3.2 odbcinst odbcinst1debian2
  python-backports.functools-lru-cache python-cycler python-gdal
  python-kiwisolver python-matplotlib python-owslib python-pyproj
  python-pyside.qtcore python-pyside.qtgui python-pyside.qtnetwork
  python-pyside.qtwebkit python-pyspatialite python-qgis
  python-qgis-common python-qt4-sql python-shapely
  python-subprocess32 qt4-designer ruby-faraday
Use 'apt autoremove' to remove them.
The following packages have been kept back:
  beef-xss fwupd fwupd-amd64-signed gdal-bin gjs gnome-shell
  gnome-shell-common gnome-shell-extension-desktop-icons
  gnome-sushi initramfs-tools initramfs-tools-core irqbalance
  kali-linux-full king-phisher libgdal20 libgjs0g libldb1
  libmariadb3 libnetcdf13 libpython3-dev libpython3-stdlib
  libqgis-customwidgets libradare2-dev libsmbclient libwbclient0
  linux-headers-amd64 linux-image-amd64 nodejs python-bs4
  python-faraday python-gdal python-impacket python-ldb
  python-matplotlib python-pil python-pil.imagetk python-pyv8
  python-qgis python-qgis-common python-samba python-webargs
  python3 python3-apt python3-brlapi python3-brotli python3-bs4
  python3-cffi-backend python3-dbus python3-dev python3-distutils
  python3-gst-1.0 python3-lib2to3 python3-lxml python3-markupsafe
  python3-matplotlib python3-minimal python3-mpltoolkits.basemap
  python3-mysqldb python3-numpy python3-pil python3-psycopg2
  python3-pycurl python3-pyproj python3-scipy python3-sip
  python3-sqlalchemy python3-sqlalchemy-ext python3-tk
  python3-tornado python3-yaml radare2 ruby-activesupport ruby-i18n
  ruby-rack ruby-sinatra samba samba-common samba-common-bin
  samba-dsdb-modules samba-libs samba-vfs-modules smbclient sslyze
  usbutils va-driver-all winexe
The following packages will be upgraded:
  accountsservice acl afflib-tools alsa-tools android-libbacktrace
  android-libbase android-libcutils android-liblog android-libutils
  android-libziparchive apache2 apache2-bin apache2-data
  apache2-utils apparmor apt apt-listchanges apt-utils arj aspell
  at-spi2-core attr b43-fwcutter base-passwd bash bash-completion
  bind9-host binfmt-support binutils binutils-common
  binutils-x86-64-linux-gnu braa bubblewrap build-essential bundler
  busybox ca-certificates ca-certificates-java cewl cheese-common
  cherrytree chkrootkit chrome-gnome-shell chromium chromium-common
  chromium-driver chromium-sandbox clang-7 commix console-setup
  console-setup-linux coreutils cpp cpp-8 cracklib-runtime cron
  cups-bsd cups-client cups-common curl cython debconf debconf-i18n
  debian-faq desktop-base dh-python diffutils dmeventd dmsetup
  dns-root-data dnsrecon dnsutils docutils-common docutils-doc dpkg
  dpkg-dev dradis easy-rsa evince evince-common
  evolution-data-server evolution-data-server-common ewf-tools
  exe2hexbat exim4-base exim4-config exim4-daemon-light exploitdb
  fern-wifi-cracker file findutils firebird3.0-common
[... 190 lines cut ...]
  ruby-http ruby-http-form-data ruby-httpclient ruby-mime
  ruby-mini-exiftool ruby-net-http-persistent ruby-nokogiri
  ruby-opt-parse-validator ruby-pkg-config ruby-rack-protection
  ruby-snmp ruby-spider ruby-twitter ruby2.5 ruby2.5-dev
  ruby2.5-doc rygel sane-utils smali sntp
  software-properties-common software-properties-gtk sparta
  speech-dispatcher speech-dispatcher-audio-plugins
  speech-dispatcher-espeak-ng sqlite3 sqlmap squashfs-tools sslscan
  stunnel4 swaks sysstat systemd systemd-sysv sysvinit-utils tar
  tasksel tasksel-data tcl tcl8.6 tcpreplay tdb-tools telnet
  tex-common texlive-base texlive-latex-base
  thin-provisioning-tools tk tk8.6 tmux tracker tracker-extract
  tracker-miner-fs tshark tzdata udev udisks2 unattended-upgrades
  unzip upower vim vim-common vim-gtk vim-gui-common vim-runtime
  vim-tiny vlan vpnc-scripts wfuzz whois wine wine64 wireshark
  wireshark-common wireshark-qt wpasupplicant wpscan xbrlapi
  xfonts-base xkb-data xml-core xserver-xorg-input-libinput
  xserver-xorg-video-amdgpu xserver-xorg-video-ati
  xserver-xorg-video-nouveau xserver-xorg-video-radeon xterm xxd
  xz-utils yelp yelp-xsl zsh zsh-common
959 upgraded, 0 newly installed, 0 to remove and 86 not upgraded.
Need to get 1,405 MB of archives.
After this operation, 162 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
```

AndrewSto, 2019-04-20:
```
root@kali:~# apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  gdal-bin gdal-data libaec0 libarmadillo9 libarpack2
  libboost-program-options1.67.0 libboost-serialization1.67.0
  libboost-test1.67.0 libboost-timer1.67.0 libcgal13 libcharls1
  libcrypt2 libdap25 libdapclient6v5 libdapserver7v5 libepsilon1
  libfcgi-bin libfcgi0ldbl libfreexl1 libfyba0 libgdal20
  libgeotiff2 libhdf4-0-alt libhdf5-100 libkmlbase1
  libkmlconvenience1 libkmldom1 libkmlengine1 libkmlregionator1
  libkmlxsd1 liblwgeom-2.5-0 liblwgeom-dev libnetcdf13 libogdi3.2
  libpoppler80 libpyside1.2 libqca2 libqca2-plugins
  libqgis-analysis2.18.25 libqgis-core2.18.25 libqgis-customwidgets
  libqgis-gui2.18.25 libqgis-networkanalysis2.18.25
  libqgis-server2.18.25 libqgispython2.18.25 libqhull7 libqtwebkit4
  libqwt6abi1 libsfcgal1 libshiboken1.2v5 libspatialindex4v5
  libspatialite7 libsuperlu5 libsz2 liburiparser1 libwhisker2-perl
  libxerces-c3.2 odbcinst odbcinst1debian2
  python-backports.functools-lru-cache python-cycler python-gdal
  python-kiwisolver python-matplotlib python-owslib python-pyproj
  python-pyside.qtcore python-pyside.qtgui python-pyside.qtnetwork
  python-pyside.qtwebkit python-pyspatialite python-qgis
  python-qgis-common python-qt4-sql python-shapely
  python-subprocess32 qt4-designer ruby-faraday
Use 'apt autoremove' to remove them.
The following packages have been kept back:
  beef-xss fwupd fwupd-amd64-signed gdal-bin gjs gnome-shell
  gnome-shell-common gnome-shell-extension-desktop-icons
  gnome-sushi initramfs-tools initramfs-tools-core irqbalance
  kali-linux-full king-phisher libgdal20 libgjs0g libldb1
  libmariadb3 libnetcdf13 libpython3-dev libpython3-stdlib
  libqgis-customwidgets libradare2-dev libsmbclient libwbclient0
  linux-headers-amd64 linux-image-amd64 nodejs python-bs4
  python-faraday python-gdal python-impacket python-ldb
  python-matplotlib python-pil python-pil.imagetk python-pyv8
  python-qgis python-qgis-common python-samba python-webargs
  python3 python3-apt python3-brlapi python3-brotli python3-bs4
  python3-cffi-backend python3-dbus python3-dev python3-distutils
  python3-gst-1.0 python3-lib2to3 python3-lxml python3-markupsafe
  python3-matplotlib python3-minimal python3-mpltoolkits.basemap
  python3-mysqldb python3-numpy python3-pil python3-psycopg2
  python3-pycurl python3-pyproj python3-scipy python3-sip
  python3-sqlalchemy python3-sqlalchemy-ext python3-tk
  python3-tornado python3-yaml radare2 ruby-activesupport ruby-i18n
  ruby-rack ruby-sinatra samba samba-common samba-common-bin
  samba-dsdb-modules samba-libs samba-vfs-modules smbclient sslyze
  usbutils va-driver-all winexe
The following packages will be upgraded:
  accountsservice acl afflib-tools alsa-tools android-libbacktrace
  android-libbase android-libcutils android-liblog android-libutils
  android-libziparchive apache2 apache2-bin apache2-data
  apache2-utils apparmor apt apt-listchanges apt-utils arj aspell
  at-spi2-core attr b43-fwcutter base-passwd bash bash-completion
  bind9-host binfmt-support binutils binutils-common
  binutils-x86-64-linux-gnu braa bubblewrap build-essential bundler
  busybox ca-certificates ca-certificates-java cewl cheese-common
  cherrytree chkrootkit chrome-gnome-shell chromium chromium-common
  chromium-driver chromium-sandbox clang-7 commix console-setup
  console-setup-linux coreutils cpp cpp-8 cracklib-runtime cron
  cups-bsd cups-client cups-common curl cython debconf debconf-i18n
  debian-faq desktop-base dh-python diffutils dmeventd dmsetup
  dns-root-data dnsrecon dnsutils docutils-common docutils-doc dpkg
  dpkg-dev dradis easy-rsa evince evince-common
  evolution-data-server evolution-data-server-common ewf-tools
  exe2hexbat exim4-base exim4-config exim4-daemon-light exploitdb
  fern-wifi-cracker file findutils firebird3.0-common
[... 290 lines cut ...]
Get:77 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 ifupdown amd64 0.8.35 [81.5 kB]
Get:78 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 systemd amd64 241-3 [3,494 kB]
Get:79 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libsystemd0 amd64 241-3 [329 kB]
Get:80 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 systemd-sysv amd64 241-3 [97.8 kB]
Get:81 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 netbase all 5.6 [19.4 kB]
Get:82 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 attr amd64 1:2.4.48-4 [41.4 kB]
Get:83 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libattr1 amd64 1:2.4.48-4 [21.3 kB]
Get:84 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 acl amd64 2.2.53-4 [60.3 kB]
Get:85 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libacl1 amd64 2.2.53-4 [30.4 kB]
Get:86 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libapparmor1 amd64 2.13.2-10 [94.7 kB]
Get:87 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libcap2 amd64 1:2.25-2 [17.6 kB]
Get:88 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libgpg-error0 amd64 1.35-1 [71.1 kB]
Get:89 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libxtables12 amd64 1.8.2-4 [80.0 kB]
Get:90 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 iptables amd64 1.8.2-4 [399 kB]
[... cut ...]
Get:96 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libkmod2 amd64 26-1 [52.7 kB]
Get:97 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libseccomp2 amd64 2.3.3-4 [41.4 kB]
Get:98 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libapt-pkg5.0 amd64 1.8.0 [966 kB]
Get:99 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libapt-inst2.0 amd64 1.8.0 [204 kB]
Get:100 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 apt amd64 1.8.0 [1,354 kB]
```

AndrewSto, 2019-04-20:
```
Err:100 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 apt amd64 1.8.0
  Connection timed out [IP: 94.247.111.11 80]
Err:101 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 apt-utils amd64 1.8.0
  Connection failed [IP: 94.247.111.11 80]
Err:102 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libdebconfclient0 amd64 0.249
  Connection failed [IP: 94.247.111.11 80]
Err:103 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 base-passwd amd64 3.5.46
  Connection failed [IP: 94.247.111.11 80]
Err:104 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libc-bin amd64 2.28-8
  Connection failed [IP: 94.247.111.11 80]
Err:105 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 ncurses-base all 6.1+20181013-2
  404  Not Found [IP: 94.247.111.11 80]
Err:106 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 rsyslog amd64 8.1901.0-1
  404  Not Found [IP: 94.247.111.11 80]
Err:107 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 tasksel-data all 3.50
  404  Not Found [IP: 94.247.111.11 80]
Err:108 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 tasksel all 3.50
  404  Not Found [IP: 94.247.111.11 80]
Err:109 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 groff-base amd64 1.22.4-3
  404  Not Found [IP: 94.247.111.11 80]
Err:110 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpipeline1 amd64 1.5.1-2
  404  Not Found [IP: 94.247.111.11 80]
Err:111 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 man-db amd64 2.8.5-2
  404  Not Found [IP: 94.247.111.11 80]
Err:112 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 mime-support all 3.62
  404  Not Found [IP: 94.247.111.11 80]
Err:113 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python2.7-dev amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:114 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python2.7 amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:115 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython2.7-dev amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:116 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython2.7 amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:117 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libtinfo-dev amd64 6.1+20181013-2
  404  Not Found [IP: 94.247.111.11 80]
Err:118 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libncurses-dev amd64 6.1+20181013-2
  404  Not Found [IP: 94.247.111.11 80]
Err:119 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libncurses6 amd64 6.1+20181013-2
  404  Not Found [IP: 94.247.111.11 80]
Err:120 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libtinfo6 amd64 6.1+20181013-2
  404  Not Found [IP: 94.247.111.11 80]
Err:121 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libncursesw6 amd64 6.1+20181013-2
  404  Not Found [IP: 94.247.111.11 80]
Err:122 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 sqlite3 amd64 3.27.2-2
  404  Not Found [IP: 94.247.111.11 80]
Err:123 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libsqlite3-0 amd64 3.27.2-2
  404  Not Found [IP: 94.247.111.11 80]
Err:124 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython2.7-stdlib amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:125 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python2.7-minimal amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:126 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython2.7-minimal amd64 2.7.16-2
  404  Not Found [IP: 94.247.111.11 80]
Err:127 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython2-stdlib amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:128 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python-all amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:129 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python-dev amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:130 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python-all-dev amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:131 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython-all-dev amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:132 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython-dev amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:133 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libpython2-dev amd64 2.7.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:134 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 python2-dev amd64 2.7.16-1
[... 1627 lines cut ...]
Err:948 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 wireshark amd64 2.6.8-1
  404  Not Found [IP: 94.247.111.11 80]
Err:949 http://mirror-1.truenetwork.ru/kali kali-rolling/non-free amd64 wpscan all 3.5.1-0kali1
  404  Not Found [IP: 94.247.111.11 80]
Err:950 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xbrlapi amd64 5.6-10
  404  Not Found [IP: 94.247.111.11 80]
Err:951 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xfonts-base all 1:1.0.5
  404  Not Found [IP: 94.247.111.11 80]
Err:952 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-input-libinput amd64 0.28.2-2
  404  Not Found [IP: 94.247.111.11 80]
[... cut ...]
Err:956 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 xserver-xorg-video-nouveau amd64 1:1.0.16-1
  404  Not Found [IP: 94.247.111.11 80]
Err:957 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 zsh amd64 5.7.1-1
  404  Not Found [IP: 94.247.111.11 80]
Err:958 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 zsh-common all 5.7.1-1
  404  Not Found [IP: 94.247.111.11 80]
Err:959 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 liburcu6 amd64 0.10.2-1
  404  Not Found [IP: 94.247.111.11 80]
```

AndrewSto, 2019-04-20:
```

Fetched 96.8 MB in 21min 8s (76.3 kB/s)                   
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/a/apt/apt_1.8.0_amd64.deb  Connection timed out [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/a/apt/apt-utils_1.8.0_amd64.deb  Connection failed [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/c/cdebconf/libdebconfclient0_0.249_amd64.deb  Connection failed [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/b/base-passwd/base-passwd_3.5.46_amd64.deb  Connection failed [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/g/glibc/libc-bin_2.28-8_amd64.deb  Connection failed [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/n/ncurses/ncurses-base_6.1+20181013-2_all.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/r/rsyslog/rsyslog_8.1901.0-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/t/tasksel/tasksel-data_3.50_all.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/t/tasksel/tasksel_3.50_all.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/g/groff/groff-base_1.22.4-3_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/libp/libpipeline/libpipeline1_1.5.1-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/m/man-db/man-db_2.8.5-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/m/mime-support/mime-support_3.62_all.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/python2.7-dev_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/python2.7_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/libpython2.7-dev_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/libpython2.7_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/n/ncurses/libtinfo-dev_6.1+20181013-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/n/ncurses/libncurses-dev_6.1+20181013-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/n/ncurses/libncurses6_6.1+20181013-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/n/ncurses/libtinfo6_6.1+20181013-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/n/ncurses/libncursesw6_6.1+20181013-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/s/sqlite3/sqlite3_3.27.2-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/s/sqlite3/libsqlite3-0_3.27.2-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/libpython2.7-stdlib_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/python2.7-minimal_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python2.7/libpython2.7-minimal_2.7.16-2_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/libpython2-stdlib_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/python-all_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/python-dev_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/python-all-dev_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/libpython-all-dev_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/libpython-dev_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/libpython2-dev_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/python2-dev_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/p/python-defaults/python2-minimal_2.7.16-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
[... cut ...]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/z/zsh/zsh-common_5.7.1-1_all.deb  404  Not Found [IP: 94.247.111.11 80]
E: Failed to fetch http://mirror-1.truenetwork.ru/kali/pool/main/libu/liburcu/liburcu6_0.10.2-1_amd64.deb  404  Not Found [IP: 94.247.111.11 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@kali:~# apt-get upgrade
E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
```

AndrewSto, 2019-04-20:
> This line is incorrect, it seems you didn't copy-pasted the command but retyped it put an incorrect symbol `»` , this should be two greater than signs `>` put together one after another.
>
>
> Open [...]

???

velichkov, 2019-04-21:
> Get:98 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libapt-pkg5.0 amd64 1.8.0 [966 kB]
> Get:99 http://mirror-1.truenetwork.ru/kali kali-rolling/main amd64 libapt-inst2.0 amd64 1.8.0 [...]

After the 99 packet you either start having problems with the internet connection or this mirror has failed. Retry after some time or change the mirror.

> root@kali:~# apt-get upgrade
> E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
> E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?

Most probably the packagekit is running, try stopping it or wait and retry.

AndrewSto, 2019-04-21:
> When installing, there are problems:
> Package 'cmake' has no installation candidate
> Package 'autoconf' has no installation candidate
> Package 'libtool' has no installation candidate
> Package [...]

I have everything set. Gnu Radio additionally need to install?

velichkov, 2019-04-21:
> Gnu Radio additionally need to install?

No.

sahilc556, 2020-04-26:
> root@kali:~# echo "deb http://http.kali.org/kali kali-rolling maincontrib non-free" » /etc/apt/sources.list root@kali:~# apt-get update
> Ign:1 https://download.docker.com/linux/debian kali-rolling [...]

nano /etc/apt/sources.list

past sources.list:
**deb http://kali.cs.nctu.edu.tw/kali kali-rolling main contrib non-free
# For source package access, uncomment the following line
# deb-src http://kali.cs.nctu.edu.tw/kali kali-rolling main contrib non-free**
 
Ctrl: o save file.
Ctrl: x exit file.

sudo apt-get update
sudo apt-get upgrade -y
sudo apt-get dist-upgrade -y

sahilc556, 2020-04-26:
> root@kali:~# echo "deb http://http.kali.org/kali kali-rolling maincontrib non-free" » /etc/apt/sources.list root@kali:~# apt-get update
> Ign:1 https://download.docker.com/linux/debian kali-rolling [...]

  GNU nano 4.9.2                                                                 New Buffer                                                                 Modified  

nano /etc/apt/sources.list

past sources.list:
deb http://kali.cs.nctu.edu.tw/kali kali-rolling main contrib non-free
# For source package access, uncomment the following line
# deb-src http://kali.cs.nctu.edu.tw/kali kali-rolling main contrib non-free
 
Ctrl: o save file.
Ctrl: x exit file.

sudo apt-get update    
sudo apt-get upgrade -y
sudo apt-get dist-upgrade -y

ptrkrysik, 2026-11-09:
This was an environment/package-manager problem with the old pybombs installation method, which is no longer used; the new release builds with plain CMake against GNU Radio 3.10. Please open a new issue if you hit a build problem with it.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik