ptrkrysik/gr-gsm #460: gsm half-rate

Ivashar, 2019-04-20:
Good day. Please tell me where to find the gsm 06.20 source code (half rate) for the format of 112 bits. If there are no sources, can anyone share the library under Win 64 ???

arifkyi, 2019-06-10:
i suggest, you raise this technical query to this miling list

since this miling pure only technical problem, details -->  gr-gsm@googlegroups.com
To join the group with any e-mail address, use this link:

https://groups.google.com/forum/#!forum/gr-gsm/join

ptrkrysik, 2026-11-09:
This is a general question about GSM half-rate codec sources rather than a gr-gsm issue; please use the mailing list for such questions.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik