ptrkrysik/gr-gsm #461: grgsm_scanner 'TCH_HS' traceback error

649, 2019-05-01:
Greetings ptrkrysik!

I'm trying to install gr-gsm on a Kali Linux build (2019.1.2) using the guide located here:
(https://harrisonsand.com/imsi-catcher/)

I was able to successfully execute every instruction until I was told to run "sudo grgsm_scanner" in which I encountered the following error:
```
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
  File "/usr/local/lib/python2.7/dist-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/local/lib/python2.7/dist-packages/grgsm/grgsm_swig.py", line 1680, in <module>
    TCH_HS = _grgsm_swig.TCH_HS
AttributeError: 'module' object has no attribute 'TCH_HS'
```

My guess is some python module is missing, not sure which module. I've looked everywhere to see if anyone had any similar issue and I couldn't find anything. Perhaps it's because Kali is defaulted to Python 2.7? I am not entirely sure.

Maybe I need to set environment paths? Or symbolic links.. Not sure. If you could tell/direct me to how I can solve this issue that would be fantastic!

velichkov, 2019-05-01:
Hi @649,

>  I'm trying to install gr-gsm on a Kali Linux build (2019.1.2) using the guide located here:
(https://harrisonsand.com/imsi-catcher/)

I just tested these instructions in a clean docker container and everything works as expected including the `grgsm_scanner`
 
> My guess is some python module is missing, not sure which module. 

I don't think it's related to python modules.

> I've looked everywhere to see if anyone had any similar issue and I couldn't find anything. 

I'm not aware of other reports.

> Perhaps it's because Kali is defaulted to Python 2.7? I am not entirely sure.

I doubt that, `gnuradio` still does not support `python 3`

> Maybe I need to set environment paths? Or symbolic links.. Not sure. 

Nope.

> If you could tell/direct me to how I can solve this issue that would be fantastic!

Most probably you ended up with a mix of new and old versions. Have you tried to install `gr-gsm` previously?  It looks like `_grgsm_swig.so` was build and installed from a previous version that does not have the `TCH_HS` attribute and for some reasons `make install` did not overwrite it with the new one. Try removing it and then reinstall.
```
sudo rm /usr/local/lib/python2.7/dist-packages/grgsm/_grgsm_swig.so
cd gr-gsm/build
sudo make install
```
or
```
cd gr-gsm/build
sudo make uninstall
sudo make install
```

649, 2019-05-01:
I ran the following as a precaution
```
sudo rm /usr/local/lib/python2.7/dist-packages/grgsm/_grgsm_swig.so
cd gr-gsm/build
sudo make uninstall
sudo make install
```
no dice :-( I'm assuming something in Kali is causing this...

velichkov, 2019-05-01:
> I'm assuming something in Kali is causing this...

I doubt that. If you have previously installed `gr-gsm` from the apt repository try removing it - `apt-get remove gr-gsm`. But if you haven't execute `find / -name _grgsm_swig.so` and provide the full output.

649, 2019-05-01:
That did the trick!
```
sudo apt-get remove gr-gsm
```
I think Kali comes with gr-gsm but an outdated version in their source repository. It conflicts with the build version and removing swig alone isn't enough to evade all the conflicts. Once I ran that command, I then ran 
```
cd gr-gsm/build
sudo make uninstall
sudo make install
```

+ Now when I run ```sudo grgsm_scanner``` I don't get any output. I get one new line and then the program just sits there with no output. I'm guessing this is hardware related? (I am using a Realtek RTL2838u dongle) max I can get it to read (based off of Gqrx) is 1,766 MHz.

I know some GSM frequencies are higher than that, would this work though?

velichkov, 2019-05-01:
> That did the trick!

Cool.
 
> I think Kali comes with gr-gsm but an outdated version in their source repository. 

I'm not a Kali user and I didn't expected that it comes preinstalled. You could use that version as well but of course some newly implemented features like TCH/H are missing in it.

>     * Now when I run `sudo grgsm_scanner` I don't get any output. I get one new line and then the program just sits there with no output.

You need to wait at least several minutes as scanning is not very fast (an not very reliable) and depending on where you live you may need to select a different band `grgsm_scanner -b GSM850`, see also `grgsm_scanner --help`

> I'm guessing this is hardware related? (I am using a Realtek RTL2838u dongle) max I can get it to read (based off of Gqrx) is 1,766 MHz.

It after waiting several minutes it still does not output anything run it with `--debug` flag and provide the full output.

> I know some GSM frequencies are higher than that, would this work though?

It won't work for `DCS1800` an `PCS1900` bands but for the other bands it should be OK.

649, 2019-05-01:
Update: Scanned a couple times with the dongle, nothing. Here's the output with the debug flag:
```
root@kali:~# grgsm_scanner --debug

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[... 115 lines cut ...]
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[... cut ...]
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOReattached kernel driver

```

Is this normal/intended behavior? If so, I can assume that there are no base stations around me that are less than ~1700 MHz. I'll then see if I can get my hands on a HackRF One, they're good investments.

velichkov, 2019-05-01:
> OOOOOOOOOOOOOOOOOOOOOOOOOOOOO

The multiple `O` in the output indicate a buffer **O**verflow. Are you running it in a VM? Try stopping all other processes that consume CPU and/or are using USB devices and try on a different USB port.

>  If so, I can assume that there are no base stations around me that are less than ~1700 MHz. 

You could find the ARFCNs of the base stations near you using an Android smartphone. See https://groups.google.com/d/msg/gr-gsm/YfcMJtPUPVU/QrUjXW38DgAJ

> I make sure the phone is connected to 2G, then I call *#4636#*#* on android to go into service mode. In service mode all neighbour cells are shown alongside the serving cell.

649, 2019-05-01:
I'm not running anything other than that (aside from background processes that Kali might do on the regular). 

Also not a VM, but this computer has 2GB physical memory with a swap disk of 10 GB.

velichkov, 2019-05-01:
> Also not a VM,

That's good

>  but this computer has 2GB physical memory with a swap disk of 10 GB.

Not sure whether this is enough or not. What CPU do you have?

You could try scanning with [kalibrate-rtl](https://github.com/steve-m/kalibrate-rtl)

Also run `rtl_test` and `cat /sys/module/usbcore/parameters/usbfs_memory_mb` and provide the full output. 

See also:
* http://lists.osmocom.org/pipermail/osmocom-sdr/2018-October/001838.html
* http://lists.osmocom.org/pipermail/osmocom-sdr/2018-October/001839.html
* http://lists.osmocom.org/pipermail/osmocom-sdr/2018-October/001841.html

ptrkrysik, 2026-11-09:
Resolved in the thread: an older distro-packaged gr-gsm conflicted with the source build (removing it fixed the TCH_HS error). The remaining no-output scan with buffer overflows is a hardware/USB/band-selection matter, not a gr-gsm defect.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik