ptrkrysik/gr-gsm #462: Error during file compilation.!!!!!

fahimWHHP, 2019-05-20:
``root@SystemAdmin:~/gr-gsm# make
[ 62%] Built target grgsm
[ 62%] Built target grgsm_swig_swig_doc
[ 64%] Built target _grgsm_swig_swig_tag
[ 68%] Built target grgsm_swig_swig_2d0df
[ 72%] Built target _grgsm_swig
[ 75%] Built target pygen_swig_826e5
[ 78%] Built target pygen_python_9fa2b
[ 80%] Built target pygen_python_misc_utils_e3f47
[ 82%] Built target pygen_python_receiver_6ad36
[ 84%] Built target pygen_python_demapping_6640b
[ 86%] Built target pygen_python_transmitter_4c5c4
[ 89%] Built target pygen_python_trx_959af
[ 90%] Generating grgsm_livemon
'options'
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1322: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
root@SystemAdmin:~/gr-gsm#

velichkov, 2019-05-20:
Hi @fahimWHHP,

> [ 90%] Generating grgsm_livemon
'options'
Error during file compilation.

Which OS are you using and which version?
How did you install gnuradio and gr-osmosdr - from the package repository, using pybombs or manually?
Have you tried following our [installation instructions](https://osmocom.org/projects/gr-gsm/wiki/Installation)?

In case you are using Kali I recommend using [this](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition) instructions.

fahimWHHP, 2019-05-22:
I am Using Kali Linux.(Linux SystemAdmin 4.15.0-kali3-amd64 #1 SMP Debian 4.15.17-1kali1 (2018-04-25) x86_64 GNU/Linux).
Plesase help...with the issue below..
root@SystemAdmin:~/gr-gsm# make
[ 62%] Built target grgsm
[ 62%] Built target grgsm_swig_swig_doc
[ 64%] Built target _grgsm_swig_swig_tag
[ 68%] Built target grgsm_swig_swig_2d0df
[ 72%] Built target _grgsm_swig
[ 75%] Built target pygen_swig_826e5
[ 78%] Built target pygen_python_9fa2b
[ 80%] Built target pygen_python_misc_utils_e3f47
[ 82%] Built target pygen_python_receiver_6ad36
[ 84%] Built target pygen_python_demapping_6640b
[ 86%] Built target pygen_python_transmitter_4c5c4
[ 89%] Built target pygen_python_trx_959af
[ 90%] Generating grgsm_livemon
'options'
Error during file compilation.
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:1323: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
root@SystemAdmin:~/gr-gsm# 
this happends when I try to "make".from "gr-gsm" folder.But when I installed it with "apt-get install gr-gsm" .Then I ran this command .." grgsm_livemon_headless " I get the error below:
linux; GNU C++ version 7.3.0; Boost_106200; UHD_3.13.1.0-3

Exception in static block reg_basic_and_lf_dboards
  LookupError: KeyError: The dboard id 0x0000 is already registered to Basic TX.
Exception in static block reg_rfx_dboards
  LookupError: KeyError: The dboard id pair [0x0024, 0x0028] is already registered to RFX400.
Exception in static block reg_xcvr2450_dboard
  LookupError: KeyError: The dboard id pair [0x0061, 0x0060] is already registered to XCVR2450.
Exception in static block reg_sbx_dboards
  LookupError: KeyError: The dboard id pair [0x0054, 0x0055] is already registered to SBX.
Exception in static block reg_ubx_dboards
  LookupError: KeyError: The dboard id pair [0x0074, 0x0073] is already registered to UBX v0.3.
Exception in static block reg_wbx_simple_dboards
  LookupError: KeyError: The dboard id pair [0x0053, 0x0052] is already registered to WBX.
Exception in static block reg_dbsrx_dboard
  LookupError: KeyError: The dboard id 0x000d is already registered to DBSRX.
Exception in static block reg_unknown_dboards
  LookupError: KeyError: The dboard id 0xfff0 is already registered to Unknown TX.
Exception in static block reg_tvrx_dboard
  LookupError: KeyError: The dboard id 0x0040 is already registered to TVRX.
Exception in static block reg_dbsrx2_dboard
  LookupError: KeyError: The dboard id 0x0012 is already registered to DBSRX2.
Exception in static block reg_tvrx2_dboard
  LookupError: KeyError: The dboard id 0x0046 is already registered to TVRX2.
Exception in static block reg_twinrx_dboards
  LookupError: KeyError: The dboard id 0x0091 is already registered to TwinRX Rev A.
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Found Rafael Micro R820T tuner
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
corrupted size vs. prev_size
Aborted
And when I run "grgsm_livemon" I get:
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 28, in <module>
    from gnuradio import qtgui
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtGui.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'
And When I Run "gnuradio-companion" I get:
Traceback (most recent call last):
  File "/usr/local/bin/gnuradio-companion", line 99, in <module>
    run_main()
  File "/usr/local/bin/gnuradio-companion", line 92, in run_main
    exit(main())
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/main.py", line 51, in main
    install_prefix=gr.prefix()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/gui/Platform.py", line 38, in __init__
    _Platform.__init__(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/core/Platform.py", line 75, in __init__
    self.build_block_library()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/core/Platform.py", line 175, in build_block_library
    hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/core/utils/odict.py", line 35, in __getitem__
    return self._data[key]
KeyError: 'options'
I installed all packages using apt-get install {package}

velichkov, 2019-05-22:
Hi @fahimWHHP, 

Please use [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages.

> I am Using Kali Linux.(Linux SystemAdmin 4.15.0-kali3-amd64 #1 SMP Debian 4.15.17-1kali1 (2018-04-25) x86_64 GNU/Linux).

You have provided the Linux Kernel version but what I was asking for is the version of the distribution you are using, e.g. Kali version. See https://www.kali.org/kali-linux-releases/. Would it be possible for you to update to the latest Kali version, e.g. Kali 2019.2?

> Plesase help...with the issue below..
>```
> root@SystemAdmin:~/gr-gsm# make
> [ 90%] Generating grgsm_livemon
> 'options'
> Error during file compilation.
> make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
> make[1]: *** [CMakeFiles/Makefile2:1323: apps/CMakeFiles/pygen_apps.dir/all] Error 2
> make: *** [Makefile:141: all] Error 2
> root@SystemAdmin:~/gr-gsm#
>```

The update of Kali most probably will solve this problem.

> But when I installed it with "apt-get install gr-gsm" .Then I ran this command .." grgsm_livemon_headless " I get the error below:
> ```
> linux; GNU C++ version 7.3.0; Boost_106200; UHD_3.13.1.0-3 [...]
Most probably a version mismatch between the multiple installed versions.

> And when I run "grgsm_livemon" I get:
>```
> Traceback (most recent call last):
> File "/usr/bin/grgsm_livemon", line 28, in
> from gnuradio import qtgui
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/qtgui/**init**.py", line 37, in
> from range import Range, RangeWidget
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in
> class RangeWidget(QtGui.QWidget):
> AttributeError: 'module' object has no attribute 'QWidget'
>```

This is a well known problem, see https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-356401976, https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-366277650 and https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-368032541

> And When I Run "gnuradio-companion" I get:
>```
> Traceback (most recent call last):
> File "/usr/local/bin/gnuradio-companion", line 99, in
> run_main()
> File "/usr/local/bin/gnuradio-companion", line 92, in run_main
> exit(main())
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/main.py", line 51, in main
> install_prefix=gr.prefix()
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/gui/Platform.py", line 38, in **init**
> _Platform.**init**(self, *args, **kwargs)
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/core/Platform.py", line 75, in **init**
> self.build_block_library()
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/core/Platform.py", line 175, in build_block_library
> hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
> File "/usr/local/lib/python2.7/dist-packages/gnuradio/grc/core/utils/odict.py", line 35, in **getitem**
> return self._data[key]
> KeyError: 'options'
> ```

This `KeyError: 'options'` looks like the first problem.

> I installed all packages using apt-get install {package}

Most probably you have used `pybombs` as well or installed `gnuradio` manually as you have one version in `/usr/local/bin/gnuradio-companion` (see above logs) and apt packages never install files in `/usr/local/`. Please remove everything you have installed with `pybombs` or manually as having multiple versions installed at the same time could cause a lot of hard to debug problems.

I highly recommend you to upgrade your system to the latest Kali version.

fahimWHHP, 2019-05-23:
How to remove everything I have installed with pybombs?

velichkov, 2019-05-23:
> How to remove everything I have installed with pybombs?
```
pybombs inv
pybombs remove uhd gnuradio # add the names of other packages listed by `pybombs inv`
pip uninstall pybombs
```

ptrkrysik, 2026-11-09:
This was a build failure caused by an outdated Kali/pybombs GNU Radio setup and the old build system. Please try the new GNU Radio 3.10 release and open a new issue with full logs if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik