ptrkrysik/gr-gsm #463: Not able to do a livemon

jorgemqr, 2019-05-21:
hello newbie here: 

while trying to execute command:

root@kali:~/Descargas/gr-gsm-0.41.2# grgsm_livemon
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'

is there any fix ? thank you in advance.

jorgemqr, 2019-05-21:
i was able to fix it by doing the installation process from https://osmocom.org/projects/gr-gsm/wiki/Installation just followed the instrucctions of section Manual compilation and installation, i first installed it with PyBombs (not neccesary to delete the PyBombs work cause it is running), also just for your information i am using a Kali GNU/Linux Rolling \n \l

ptrkrysik, 2026-11-09:
This was a broken GNU Radio/Qt installation on an old Python 2.7 Kali system and the reporter solved it by reinstalling; not a gr-gsm defect.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik