ptrkrysik/gr-gsm #464: No module named _grgsm_swig

btlvr, 2019-05-31:
I have grgsm installed on Arch Linux. 

When I try to start any of the `grgsm_*` programs, I get the following error:

```
Traceback (most recent call last):
  File "/usr/bin/grgsm_scanner", line 33, in <module>
    import grgsm
  File "/usr/lib/python2.7/site-packages/grgsm/__init__.py", line 45, in <module>
    from grgsm_swig import *
  File "/usr/lib/python2.7/site-packages/grgsm/grgsm_swig.py", line 17, in <module>
    _grgsm_swig = swig_import_helper()
  File "/usr/lib/python2.7/site-packages/grgsm/grgsm_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_grgsm_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _grgsm_swig
```

I have executed `sudo ldconfig` and 
```
export PYTHONPATH=/usr/local/lib/python2.7/dist-packages/grgsm:$PYTHONPATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
```

Neither of these solved the problem.

velichkov, 2019-05-31:
Hi @btlvr,

> I have grgsm installed on Arch Linux.

How did you install it - manually, using pacman or using pybombs? 

> When I try to start any of the `grgsm_*` programs, I get the following error:
>
> ```
> Traceback (most recent call last):
>   File "/usr/bin/grgsm_scanner", line 33, in <module>
>     import grgsm
>   File [...]

You have installed in in `/usr/lib/` and not in `/usr/local/lib` so try changing the path in the above two variables accordingly.

Also provide the full output of the following commands.

```
sudo find / -name _grgsm_swig.so
ldd /usr/lib/python2.7/site-packages/grgsm/_grgsm_swig.so
echo PYTHONPATH $PYTHONPATH
echo LD_LIBRARY_PATH $LD_LIBRARY_PATH
```

btlvr, 2019-06-01:
@velichkov it was installed with pacman. 

I have tried changing the `/usr/local/lib/` to `/usr/lib` in the two `export` statements, but that did not fix the issue. 

```
# sudo find / -name _grgsm_swig.so 2>/dev/null
/usr/lib/python2.7/site-packages/grgsm/_grgsm_swig.so
```
```
ldd /usr/lib/python2.7/site-packages/grgsm/_grgsm_swig.so
	linux-vdso.so.1 (0x00007ffe7abf9000)
	libpython2.7.so.1.0 => /usr/lib/libpython2.7.so.1.0 (0x00007febee0c6000)
	libgrgsm-0.42.2..so.0.0.0 => /usr/lib/libgrgsm-0.42.2..so.0.0.0 (0x00007febedfbf000)
	libboost_filesystem.so.1.68.0 => not found
	libboost_system.so.1.68.0 => not found
	libgnuradio-runtime-3.7.13.4.so.0.0.0 => not found
	libgnuradio-pmt-3.7.13.4.so.0.0.0 => not found
	libgnuradio-filter-3.7.13.4.so.0.0.0 => not found
	libgnuradio-fft-3.7.13.4.so.0.0.0 => not found
	libvolk.so.1.4 => /usr/lib/libvolk.so.1.4 (0x00007febedc94000)
	libosmocore.so.11 => /usr/lib/libosmocore.so.11 (0x00007febedc6b000)
	libosmocodec.so.0 => /usr/lib/libosmocodec.so.0 (0x00007febedc65000)
	libosmogsm.so.10 => /usr/lib/libosmogsm.so.10 (0x00007febedc0c000)
	libosmocoding.so.0 => /usr/lib/libosmocoding.so.0 (0x00007febedbf1000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007febeda61000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007febed91b000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007febed901000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007febed73c000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007febed71b000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007febed714000)
	libutil.so.1 => /usr/lib/libutil.so.1 (0x00007febed70f000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007febee681000)
	libboost_filesystem.so.1.68.0 => not found
	libboost_system.so.1.68.0 => not found
	libgnuradio-runtime-3.7.13.4.so.0.0.0 => not found
	libgnuradio-pmt-3.7.13.4.so.0.0.0 => not found
	libgnuradio-filter-3.7.13.4.so.0.0.0 => not found
	libgnuradio-fft-3.7.13.4.so.0.0.0 => not found
	liborc-0.4.so.0 => /usr/lib/liborc-0.4.so.0 (0x00007febed68e000)
	libtalloc.so.2 => /usr/lib/libtalloc.so.2 (0x00007febed678000)
	libgnutls.so.30 => /usr/lib/libgnutls.so.30 (0x00007febed4cc000)
	libcrypt.so.1 => /usr/lib/libcrypt.so.1 (0x00007febed490000)
	libp11-kit.so.0 => /usr/lib/libp11-kit.so.0 (0x00007febed360000)
	libidn2.so.0 => /usr/lib/libidn2.so.0 (0x00007febed341000)
	libunistring.so.2 => /usr/lib/libunistring.so.2 (0x00007febecfc1000)
	libtasn1.so.6 => /usr/lib/libtasn1.so.6 (0x00007febecdae000)
	libnettle.so.6 => /usr/lib/libnettle.so.6 (0x00007febecd74000)
	libhogweed.so.4 => /usr/lib/libhogweed.so.4 (0x00007febecd39000)
	libgmp.so.10 => /usr/lib/libgmp.so.10 (0x00007febecca5000)
	libffi.so.6 => /usr/lib/libffi.so.6 (0x00007febecc9a000)
```
```
# echo PYTHONPATH $PYTHONPATH          
                                       
PYTHONPATH /usr/lib/python2.7/dist-packages/grgsm:
```

```
# echo LD_LIBRARY_PATH $LD_LIBRARY_PATH
                                       
LD_LIBRARY_PATH /usr/lib:
```

velichkov, 2019-06-01:
Hi @btlvr,

> ldd /usr/lib/python2.7/site-packages/grgsm/_grgsm_swig.so
> 	linux-vdso.so.1 (0x00007ffe7abf9000)
> 	libpython2.7.so.1.0 => /usr/lib/libpython2.7.so.1.0 (0x00007febee0c6000)
> 	libgrgsm-0.42.2..so.0.0.0 => /usr/lib/libgrgsm-0.42.2..so.0.0.0 (0x00007febedfbf000)
> 	libboost_filesystem.so.1.68.0 => not found
> 	libboost_system.so.1.68.0 => not found
> 	libgnuradio-runtime-3.7.13.4.so.0.0.0 => not found
> 	libgnuradio-pmt-3.7.13.4.so.0.0.0 => not found
> 	libgnuradio-filter-3.7.13.4.so.0.0.0 => not found
> 	libgnuradio-fft-3.7.13.4.so.0.0.0 => not found
>.......
> 	libboost_filesystem.so.1.68.0 => not found
> 	libboost_system.so.1.68.0 => not found
> 	libgnuradio-runtime-3.7.13.4.so.0.0.0 => not found
> 	libgnuradio-pmt-3.7.13.4.so.0.0.0 => not found
> 	libgnuradio-filter-3.7.13.4.so.0.0.0 => not found
> 	libgnuradio-fft-3.7.13.4.so.0.0.0 => not found

So you are missing the `boost` and `gnuradio` dependencies. I don't have any experience with Arch Linux. I would have expected the package manager to install all dependencies but maybe that's not the case with `pacman` or there is a bug in the package (consider reporting it to the package maintainer).

Try installing the missing dependencies manually - `sudo pacman -S gnuradio boost` and also have a look at [Vadim's docker file](https://git.osmocom.org/docker-playground/tree/gr-gsm-master/Dockerfile)

aep, 2020-02-23:
> ```
> export PYTHONPATH=/usr/local/lib/python2.7/dist-packages/grgsm:$PYTHONPATH
> export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
> ```

these should probably be part of the installation manual here since they are required
https://osmocom.org/projects/gr-gsm/wiki/Installation

ptrkrysik, 2026-11-09:
This was a distro packaging/dependency problem (missing boost/GNU Radio libraries) with the old Python 2 swig build. The new release no longer uses that layout; please open a new issue if you see a similar problem with it.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik