ptrkrysik/gr-gsm #467: How to start reading the Code

moemuses, 2019-06-16:
Hi every one .

i'm have been using gr-gsm more than one year, it's amazing code and good tools

now i want to read the code to edit it , to my own purpose
my c++ is intermediate , and python is good

can someone please tell me the guide to start reading the whole code , the started point and milestone.

sorry for bad english
and thank you again for this project.

velichkov, 2019-06-17:
Hi @moemuses,

> can someone please tell me the guide to start reading the whole code , the started point and milestone.

Unfortunately there are neither starting point nor milestones. It depends a lot of what your needs are and if you describe them here maybe we will be able to give you more specific advises.

My general advice is to start small and try to understand only the parts that are directly related to the feature you want to implement or the bug you need to fix. Split the feature/bug fix on as small tasks as possibly could, fork the github repository, share your work early (don't wait until it's finished) and ask for feedback and advises in a github pull request or on the mailing list.

> sorry for bad english

Don't worry too much about your English, most of us are non-native speakers.

moemuses, 2019-06-19:
Hi @velichkov 

> Unfortunately there are neither starting point nor milestones. It depends a lot of what your needs are and if you describe them here maybe we will be able to give you more specific advises.

let's say first of all , i want to undrstand how to decode packet , i mean the first step after SDR output data to grgsm. i.e the steps grgsm do to convert packet from SDR to Readable msg that send to udp.

Thank you for your interest

velichkov, 2019-06-19:
> let's say first of all , i want to undrstand how to decode packet , i mean the first step after SDR output data to grgsm. i.e the steps grgsm do to convert packet from SDR to Readable msg that send [...]

That's like more then 50% of what gr-gsm does (recent versions do support transmit which is the reverse direction). Open `apps/grgsm_livemon.grc` in `gnuradio-companion` to get a higher level overview of the available blocks and how they are connected and try to understand each one starting from the left to right - GSM Input Adaptor, GSM Receiver, various Demappers, ...

You said that you want to "read the code to edit it , to my own purpose" and to edit it you don't necessary have to understand everything in order to implement something useful or interesting. For example I still don't understand how the GSM Input Adaptor works but this does not stop me to implement various other things. The idea is to start with something specific (and small) and work you way. 

So if you have some specific ideas in mind you could describe them here, otherwise it is really difficult to give you any advice how to understand everything (or at least half of it) all at once.

Hope this helps.

moemuses, 2019-06-19:
> That's like more then 50% of what gr-gsm does (recent versions do support transmit which is the reverse direction). Open `apps/grgsm_livemon.grc` in `gnuradio-companion` to get a higher level [...]

Sorry if my question is a general question, I'm dealing with GNU radio and all of .grc "livemon" , "decode" also i combine ( livemon with decode with multi_rtl ) grcs to decode livestream of uplink and downlink

my specific question now , if i want to know what happen inside some of this block , let's say 'GSM input adaptor' , where i can find this code
 
> Hope this helps.

Thank you.

velichkov, 2019-06-19:
> my specific question now , if i want to know what happen inside some of this block , let's say 'GSM input adaptor' , where i can find this code

All blocks are implemented in either python or c++, the python blocks are in `python` sub-directory and the c++ blocks are in `lib` sub-directory. Usually the filename contains something similar to the block name and sometimes it's easier to use `find`.

```
$ find . -name *input* | grep -v build
./python/receiver/gsm_input.py
./hier_blocks/receiver/gsm_input.grc
./grc/receiver/gsm_input.xml
```

moemuses, 2019-06-19:
> All blocks are implemented in either python or c++, the python blocks are in `python` sub-directory and the c++ blocks are in `lib` sub-directory. Usually the filename contains something similar to [...]

Ok, that's help alot , thank you for your time :)

ptrkrysik, 2026-11-09:
Answered in the thread (use apps/grgsm_livemon.grc as an overview; blocks are under lib/ and python/). Feel free to ask further questions on the mailing list.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik