ptrkrysik/gr-gsm #468: Generating gr-gsm failed

PHPoenX, 2019-06-30:
OpenSUSE Tumbleweed x86_64

```
[ 94%] Generating grgsm_livemon
cd /home/ifrit/gr-gsm/build/apps && /usr/bin/cmake -E env PYTHONPATH="/home/ifrit/gr-gsm/python:/home/ifrit/gr-gsm/python/misc_utils:/home/ifrit/gr-gsm/python/demapping:/home/ifrit/gr-gsm/python/receiver:/home/ifrit/gr-gsm/python/transmitter:/home/ifrit/gr-gsm/python/trx:/home/ifrit/gr-gsm/build/swig" GRC_BLOCKS_PATH=/home/ifrit/gr-gsm/grc /usr/bin/grcc -d /home/ifrit/gr-gsm/build/apps /home/ifrit/gr-gsm/apps/grgsm_livemon.grc
Warning: restarting the docstring loader (crashed while loading 'gsm_clock_offset_control')
Warning: restarting the docstring loader (crashed while loading 'gsm_fcch_burst_tagger')
Warning: restarting the docstring loader (crashed while loading 'gsm_input')
Warning: restarting the docstring loader (crashed while loading 'gsm_receiver_with_uplink')
CPU Features: SSE2+ SSE4.1- AVX- FMA-
Using sse2 for xtrxdsp_iq16_sc32
Using sse2 for xtrxdsp_iq8_ic16
Using sse2 for xtrxdsp_iq16_ic16i
Using sse2 for xtrxdsp_iq8_ic8i
Using sse2 for xtrxdsp_sc32i_iq16
Using sse2 for xtrxdsp_iq8_sc32
Using sse2 for xtrxdsp_iq8_sc32i
Using sse2 for xtrxdsp_iq16_sc32i
Using sse2 for xtrxdsp_sc32_iq16
Using sse2 for xtrxdsp_ic16i_iq16
Segmentation fault

velichkov, 2019-07-01:
Hi @PHPoenX,

> OpenSUSE Tumbleweed x86_64

I just successfully tested the build in OpenSUSE Tumbleweed in [a Docker container](https://github.com/velichkov/gr-gsm/blob/opensuse/tests/dockerfiles/OpenSUSE_Tumbleweed.Dockerfile) and [here are the logs](https://travis-ci.org/velichkov/gr-gsm/jobs/552557794#L460)

> ```
> [ 94%] Generating grgsm_livemon
> cd /home/ifrit/gr-gsm/build/apps && /usr/bin/cmake -E env [...]

You could try to disable building of `grgsm_livemon` and `grgsm_livemon_headless` by executing `cmake -DENABLE_GRCC=OFF ...` from the `build` directory and then if it builds successfully you could try to manually build `apps/grgsm_livemon.grc` and `apps/grgsm_livemon_headless.grc` with `gnuradio-companion`

Could you take a backtrace with gdb? Run `ulimit -c unlimited` and `make`. After it fails with `Segmentation fault` a file with name similar to `core.2322` would be generated in you working directory, you need to open it with `gdb -c core...` and then in the gdb shell execute `bt` and provide the full output.

Could you provide some more information about your system? 
Is you system fully updated?
Which gnuradio version do you use and how did you installed it -  from the distribution repository, manually or using pybombs?
What SDR device do you use and is it connected during the build? Try building with both connected and disconnected SDR. 
Do you build it in a VM and if so could you share an image of this VM?

PHPoenX, 2019-07-02:
>  Is you system fully updated?

Yes.

> What SDR device do you use and is it connected during the build?

All disconnected

PHPoenX, 2019-07-02:
>  disable building of `grgsm_livemon`

I need it

> pybombs

I tried `> sudo pybombs config default_prefix default_prx
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.7.3
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.config - INFO - Using config file: /root/.pybombs/config.yml
default_prefix: default_prx
  - Default Prefix
ifrit@ifrit-factory ~> sudo pybombs prefix init /usr/local -a default_prx -R gr-gsm
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.7.3
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.prefix - ERROR - Could not find recipe for `gr-gsm'`

velichkov, 2019-07-02:
> > Is you system fully updated?
> 
> Yes.

OK.

> > What SDR device do you use and is it connected during the build?
> 
> All disconnected

Have you tried building when the SDR device is connected?
What SDR device do you use?

> GR version is 3.7.13.5

How did you install it?

>>  disable building of `grgsm_livemon`
>
> I need it

I know, you could manually build it later using `gnuradio-companion` as described before  so I recommend trying to disable it. The more things you try the more information we get the easier will be to find and fix the problem.

> I tried
>
> ```
> PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.7.3
> PyBOMBS - INFO - PyBOMBS Version 2.3.3
> PyBOMBS.config - INFO - Using config file: /root/.pybombs/config.yml [...]

I **do not** recommend using `pybombs` but if you decide to try it first do it in a VM or prepare yourself for a long and hard recovery process. You are missing `pybombs auto-config` and `pybombs recipes add-defaults` steps. See https://osmocom.org/projects/gr-gsm/wiki/Installation#Compilation-and-installation-with-use-of-Pybombs

Have to tried to take a core file and a backtrace?

PHPoenX, 2019-07-04:
> You are missing `pybombs auto-config` and `pybombs recipes add-defaults` steps.

No

>  disable building of `grgsm_livemon` and `grgsm_livemon_headless`

It works!

PHPoenX, 2019-07-04:
> How did you install it?

I used zypper(suse pkg manager)

After updating pkgs and rebuilding with and without flag -DENABLE_GRCC=OFF all works good!

closed 2019-07-04 by PHPoenX