ptrkrysik/gr-gsm #469: XTRX Support

cyrus104, 2019-07-03:
I would like to see about getting XTRX support.

Thanks!

velichkov, 2019-07-03:
Hi @cyrus104,

`gr-gsm` uses [gr-osmosdr](https://osmocom.org/projects/gr-osmosdr/wiki) internally so all devices supported by `gr-osmosdr` are supported by `gr-gsm` as well. It seems the official `gr-osmosdr` does not have support for XTRX yet but in the [xtrx-sdr's fork](https://github.com/xtrx-sdr/gr-osmosdr) they've added support for it so probably you'll need to install this version.

cyrus104, 2019-07-04:
@velichkov thanks for the quick response. I've sent an email to the xtrx email distro asking if their fork can be merged back into gr-osmosdr. Hopefully they are able to migrate that back in and then we'll have our support.

ptrkrysik, 2019-08-21:
I'm closing this issue. If gr-osmosdr (or other gnu radio block) will support xtrx then it'll be supported. If not it won't. I don't have xtrx so I'm not able to work on it (even if I had time for it).

closed 2019-08-21 by ptrkrysik