ptrkrysik/gr-gsm #470: Sampling rate  not support  USRP X310

czl1989, 2019-07-04:
how to support USRP X310？  Thanks

velichkov, 2019-07-04:
Hi @czl1989,

> how to support USRP X310？

All USRP devices are supported. We are using `gr-osmosdr` project internally and all devices supported by `gr-osmosdr` are supported by `gr-gsm` as well.

> Sampling rate not support USRP X310

Could you describe exactly what is not supported by your device?

czl1989, 2019-07-05:
Project set rx sample rate is 270.833k*4. USRP1 clock is 52M can support 270.833k*4. USRP x310 clock is 200M can not support 270.833k*4. I change the sample rate for 390.625k*4, but the tx tag is wrong.

velichkov, 2019-07-08:
Hi @czl1989,

> I change the sample rate for 390.625k4, but the tx tag is wrong.

Is the problem in `grgsm_trx` as the other gr-gsm applications do not use tx_time tags?
Could you describe in some more details what you are trying to do?

ptrkrysik, 2019-08-21:
For grgsm_trx - devices other than USRPs B2x0 are not supported yet. I'm working on it (as part of effort to add frequency hopping to grgsm_trx) but it's not done yet.

closed 2019-08-21 by ptrkrysik