ptrkrysik/gr-gsm #471: Problem with RTL-SDR source

Krounet, 2019-07-07:
Hello,
I tried to make a FM receiver with gnuradio on a raspberry pi 3 B+ The OS is Debian Stretch. My sdr device is well recognized by the pi (an nesdr smart XTR from NOOELEC). And the RTL-SDR Source Block is recognized by gnuradio-companion.

But, during the compilation I've got this error message.

Traceback (most recent call last):
  File "/home/pi/Documents/test_gnuradio/fm_receiver/top_block.py", line 30, in <module>
    import osmosdr
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/__init__.py", line 26, in <module>
    from osmosdr_swig import *
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 21, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 20, in swig_import_helper
    return importlib.import_module('_osmosdr_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _osmosdr_swig

>>> Done

For information my swig version is 3.0.10.
I investigate a little into the repository _/usr/local/lib/python2.7/dist-packages/osmosdr/_. I find a file named _osmosdr_swig.so but I don't think it's the famous module. I must admit I don't know what is the *. so extention.

So, if somebody know how to resolve _osmosdr_swig module problem, don't hesitate, there will be a big virtual at the end :). And I am also curious to know what is a *.so extension.

Thanks

velichkov, 2019-07-08:
Hi @Krounet,

> I tried to make a FM receiver with gnuradio on a raspberry pi 3 B+ The OS is Debian Stretch.

Which gnuradio version do you use and how did you install it?

> My sdr device is well recognized by the pi (an nesdr smart XTR from NOOELEC). And the RTL-SDR Source Block is recognized by gnuradio-companion.

How did you install `gr-osmosdr`?

> But, during the compilation I've got this error message.
> 
> Traceback (most recent call last):
> File "/home/pi/Documents/test_gnuradio/fm_receiver/top_block.py", line 30, in
> import osmosdr
> File "/usr/local/lib/python2.7/dist-packages/osmosdr/**init**.py", line 26, in
> from osmosdr_swig import *
> File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 21, in
> _osmosdr_swig = swig_import_helper()
> File "/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py", line 20, in swig_import_helper
> return importlib.import_module('_osmosdr_swig')
> File "/usr/lib/python2.7/importlib/**init**.py", line 37, in import_module
> **import**(name)
> ImportError: No module named _osmosdr_swig

Provide the full output of `ldd /usr/local/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.so`

> I investigate a little into the repository _/usr/local/lib/python2.7/dist-packages/osmosdr/_. I find a file named _osmosdr_swig.so but I don't think it's the famous module.

Yes, that is the module.

> I must admit I don't know what is the *. so extention.
> And I am also curious to know what is a *.so extension.

The `.so` files are dynamic libraries similar to `DLL` in Windows, `so` stands for `shared object`. 

http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html
http://www.yolinux.com/TUTORIALS/LibraryArchives-StaticAndDynamic.html

> So, if somebody know how to resolve _osmosdr_swig module problem, don't hesitate, there will be a big virtual at the end :).

Big virtual what? :)

Krounet, 2019-07-14:
Hello velichkov, thanks for your answer.

> Which gnuradio version do you use and how did you install it?
> 

My gnuradio version is 3.7.10. I installed gnuradio via an Aptitude Repository with Debian (sudo apt-get install gnuradio). I tried first to installed gnuradio from source with cmake, But I had some problems with volk and the mako python package (my mako version was not recognized during the installation).

> How did you install gr-osmosdr?

I installed gr-osmosdr from source with cmake

```
git clone git://git.osmocom.org/gr-osmosdr
cd gr-osmosdr/
mkdir build
cd build/
cmake ../
make
sudo make install
sudo ldconfig
```
Before that i installed the rtl-sdr packages, I did a rtl_test with my sdr device and it worked well.
```
linux-vdso.so.1 (0x7e98d000)
	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76ee2000)
	libpython2.7.so.1.0 => /usr/lib/arm-linux-gnueabihf/libpython2.7.so.1.0 (0x76ba9000)
	libgnuradio-osmosdr.so.0.1.5git => /usr/local/lib/libgnuradio-osmosdr.so.0.1.5git (0x76b3c000)
	libboost_thread.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.62.0 (0x76b0a000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ae1000)
	libboost_system.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.62.0 (0x76ace000)
	libboost_chrono.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_chrono.so.1.62.0 (0x76ab8000)
	libboost_date_time.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_date_time.so.1.62.0 (0x76a9a000)
	libboost_atomic.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_atomic.so.1.62.0 (0x76a87000)
	libgnuradio-runtime.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.7.10 (0x7699e000)
	libgnuradio-pmt.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-pmt.so.3.7.10 (0x76945000)
	libgnuradio-blocks.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-blocks.so.3.7.10 (0x76718000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76701000)
	libgnuradio-fcd.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-fcd.so.3.7.10 (0x766dd000)
	librtlsdr.so.0 => /usr/lib/arm-linux-gnueabihf/librtlsdr.so.0 (0x766c0000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76578000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x764f9000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x764cc000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x7638d000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x76366000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76353000)
	libutil.so.1 => /lib/arm-linux-gnueabihf/libutil.so.1 (0x76340000)
	/lib/ld-linux-armhf.so.3 (0x76fb0000)
	libvolk.so.1.3 => /usr/lib/arm-linux-gnueabihf/libvolk.so.1.3 (0x762da000)
	libboost_program_options.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_program_options.so.1.62.0 (0x7625c000)
	libboost_filesystem.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.62.0 (0x76236000)
	libboost_regex.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.62.0 (0x7613b000)
	liblog4cpp.so.5 => /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5 (0x760fb000)
	libgnuradio-audio.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-audio.so.3.7.10 (0x760a6000)
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0x76081000)
	libasound.so.2 => /usr/lib/arm-linux-gnueabihf/libasound.so.2 (0x75f94000)
	libjack.so.0 => /usr/lib/arm-linux-gnueabihf/libjack.so.0 (0x75f4b000)
	libportaudio.so.2 => /usr/lib/arm-linux-gnueabihf/libportaudio.so.2 (0x75f12000)
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x75e9a000)
	libicudata.so.57 => /usr/lib/arm-linux-gnueabihf/libicudata.so.57 (0x7460d000)
	libicui18n.so.57 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.57 (0x743fd000)
	libicuuc.so.57 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.57 (0x74293000)
	libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0x7426f000)
	libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0x74252000)
	libopus.so.0 => /usr/lib/arm-linux-gnueabihf/libopus.so.0 (0x741ff000)
```

> Provide the full output of ldd /usr/local/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.so

```
linux-vdso.so.1 (0x7e98d000)
[... 23 lines cut ...]
	libvolk.so.1.3 => /usr/lib/arm-linux-gnueabihf/libvolk.so.1.3 (0x762da000)
	libboost_program_options.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_program_options.so.1.62.0 (0x7625c000)
	libboost_filesystem.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_filesystem.so.1.62.0 (0x76236000)
	libboost_regex.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.62.0 (0x7613b000)
	liblog4cpp.so.5 => /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5 (0x760fb000)
	libgnuradio-audio.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-audio.so.3.7.10 (0x760a6000)
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0x76081000)
	libasound.so.2 => /usr/lib/arm-linux-gnueabihf/libasound.so.2 (0x75f94000)
	libjack.so.0 => /usr/lib/arm-linux-gnueabihf/libjack.so.0 (0x75f4b000)
	libportaudio.so.2 => /usr/lib/arm-linux-gnueabihf/libportaudio.so.2 (0x75f12000)
	liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0x75e9a000)
	libicudata.so.57 => /usr/lib/arm-linux-gnueabihf/libicudata.so.57 (0x7460d000)
	libicui18n.so.57 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.57 (0x743fd000)
	libicuuc.so.57 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.57 (0x74293000)
	libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0x7426f000)
	libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0x74252000)
	libopus.so.0 => /usr/lib/arm-linux-gnueabihf/libopus.so.0 (0x741ff000)
```

> The .so files are dynamic libraries similar to DLL in Windows, so stands for shared object.
> 
> http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html
> http://www.yolinux.com/TUTORIALS/LibraryArchives-StaticAndDynamic.html

Thanks you for these informations and your help, I'll get a look to these links. The famous mysterious big virtual was....a big virtual hug :).

velichkov, 2019-07-15:
Hi @Krounet,

> My gnuradio version is 3.7.10. I installed gnuradio via an Aptitude Repository with Debian (sudo apt-get install gnuradio). I tried first to installed gnuradio from source with cmake, But I had some [...]

OK

> > How did you install gr-osmosdr?
> 
> I installed gr-osmosdr from source with cmake

Is there any particular reason to choose installation from source?  There is a [gr-osmosdr package](https://packages.debian.org/stretch/gr-osmosdr) in Debian Stretch. If you decide to try the version from the package first remove the one installed from source (sudo make uninstall) as having multiple versions installed at the same time could cause hard to debug problems.

> ```
> git clone git://git.osmocom.org/gr-osmosdr
> cd gr-osmosdr/
> mkdir build
> cd build/
> cmake ../
> make
> sudo make install
> sudo ldconfig
> ```

Installation steps are OK.
 
> Before that i installed the rtl-sdr packages, I did a rtl_test with my sdr device and it worked well.

OK. Have you tried [osmocom_fft](https://osmocom.org/projects/gr-osmosdr/wiki#Spectrum-Browser)?

> ```
> linux-vdso.so.1 (0x7e98d000)
> 	/usr/lib/arm-linux-gnueabihf/libarmmem.so (0x76ee2000)
> 	libpython2.7.so.1.0 => /usr/lib/arm-linux-gnueabihf/libpython2.7.so.1.0 (0x76ba9000) [...]

It finds all necessary libraries but maybe there are some undefined symbols. Could you run `ldd -r /usr/local/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.so` and provide the full output. Without the `-r` flag `ldd` does not perform functions and data object relocations.

Have you updated any of the `gr-osmosdr`'s dependency libraries (the one listed above) after installing `gr-osmosdr` as this could cause undefined symbols? If so you could try reinstalling `gr-osmosdr` by first removing its build directory and then executing the same steps as before.

You could also try adding `/usr/local/lib/python2.7/dist-packages/osmosdr/` to `LD_LIBRARY_PATH` environment variable and then start `gnuradio-companion` from the same terminal but I doubt that this will solve the problem.

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib/python2.7/dist-packages/osmosdr/
gnuradio-companion
```

Krounet, 2019-07-20:
Hi velichkov,

Well I remove my gr-osmosdr version, and I replaced it with the Debian package, and it works !!!!

Thanks you very much for your help and have a nice week end :)

ptrkrysik, 2026-11-09:
This is a gr-osmosdr installation problem, not a gr-gsm one, and it was solved by installing the distribution's gr-osmosdr package.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik