ptrkrysik/gr-gsm #472: Can't find my TMSI after Decode SDCCH

haryows, 2019-07-17:
Hi Guys.
I have a problem, I already capturing with ARFCN 2 after Decode it for BCCH with this command :
[image]
[image]

I can Find My TMSI, but after second decode
[image]

[image]

I Can't find my TMSI anymore but i still can find the chiper, for this case is A5/1
I Continue for decode with my Kc and got nothing. I capturing 30 second and send at least 5 sms n i make sure that ARFCN,TMSI and KC is not change.
i have no idea what i missing/mistake, hope u can help me with this issue. (im using hackrf one)

TMSI 0x2430B7E0
Kc	 096F6C34DEB99195
cfile : 
https://drive.google.com/open?id=1fpe0yJuWN7KKmAdmucQOAnF9YV63J4_p

velichkov, 2019-07-17:
Hi @haryows,

You should post questions like this one to [our mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

> I have a problem, I already capturing with ARFCN 2 after Decode it for BCCH with this command :
> [image]
> [image]

Please never post screenshots of your terminal or Wireshark. You could copy-paste the text from the terminal and in Wireshark right click on a packet or a decoded field and then Copy -> All Visible .....

How you make sure that this Immediate Assignment is for you phone as it does not contains TMSI/IMSI?
There are 4 Immediate Assignment messages for an SDCCH8 channel in your capture, have you tried them all?

> I can Find My TMSI, but after second decode

I can't find any message containing your TMSI. I've used "gsm_a.tmsi == 0x2430B7E0" as Wireshark filter after decoding timeslot 0, 1, 2 and 3 and there are no any messages that contains it.

> [image]

The above Immediate Assignment is for timeslot 2 so you need to decode timeslot 2 (`-t 2`) and not 1.

> [image]
> 
> I Can't find my TMSI anymore but i still can find the chiper, for this case is A5/1

This Ciphering Mode Command  most probably is not for your phone.

> I Continue for decode with my Kc and got nothing. I capturing 30 second and send at least 5 sms

Try receiving SMSes or Voice Calls as this will trigger Paging Request messages with your TMSI/IMSI. 

>  n i make sure that ARFCN,TMSI and KC is not change.

How do you make sure that these are correct and do not change during the test?

> i have no idea what i missing/mistake, hope u can help me with this issue. (im using hackrf one)
> 
> TMSI 0x2430B7E0
> Kc 096F6C34DEB99195

How did you get TMSI and KC? 

Regards,
Vasil

haryows, 2019-07-19:
> Please never post screenshots of your terminal or Wireshark. You could copy-paste the text from the terminal and in Wireshark right click on a packet or a decoded field and then Copy -> All Visible .....

im sorry for this :)

> Try receiving SMSes or Voice Calls as this will trigger Paging Request messages with your TMSI/IMSI. 

should I just send 1 sms/capture?

> How do you make sure that these are correct and do not change during the test?

> How did you get TMSI and KC?

i using blackberry engineering screen that contain ARFCN,TMSI and KC. I checked before and after I send d SMS

I already got the GSM SMS and there is no SMS in it. the description show : 

> GSM SMS TPDU (GSM 03.40) SMS-STATUS REPORT
.0.. .... = TP-UDHI: The TP UD field contains only the short message
..0. .... = TP-SRQ: SMS STATUS REPORT is the result of a SMS SUBMIT.
.... 0... = TP-LP: The message has not been forwarded and is not a spawned message
.... .1.. = TP-MMS: No more messages are waiting for the MS in this SC
.... ..10 = TP-MTI: SMS-STATUS REPORT (2)
TP-MR: 43
TP-Recipient-Address - (destination number)
TP-Service-Centre-Time-Stamp
TP-Discharge-Time
TP-Status
0... .... = Definition of bits 0-6: as follows
.000 0000 = Error: Short message transaction completed (0)
.000 0000 = Reason: Short message received by the SME (0)

is that because i send multiple SMS in 1 time?

velichkov, 2019-07-22:
Hi @haryows,

> > Try receiving SMSes or Voice Calls as this will trigger Paging Request messages with your TMSI/IMSI.
> 
> should I just send 1 sms/capture?

Try sending one SMS from another phone to this one, or send one SMS to your number so you are going to both send and receive one SMS.

> > How do you make sure that these are correct and do not change during the test?
> 
> > How did you get TMSI and KC?
> 
> i using blackberry engineering screen that contain ARFCN,TMSI and KC. I checked before and after I send d SMS

OK. I don't have blackberry and I'm not familiar with their engineering screen but note that it's possible your phone to choose a different ARFCN then the one it display on this screen.

> I already got the GSM SMS and there is no SMS in it. the description show :
> 
> > GSM A-I/F DTAP - CP-DATA
> > Protocol Discriminator: SMS messages (9)
> > .... 1001 = Protocol discriminator: SMS messages (0x9)
> > 0... .... = TI flag: allocated by sender
> > .000 .... = TIO: 0
> > DTAP Short Message Service Message Type: CP-DATA (0x01)
> > CP-User Data
> > Length: 37
> > RPDU (not displayed)
> 
> > GSM SMS TPDU (GSM 03.40) SMS-STATUS REPORT
> > .0.. .... = TP-UDHI: The TP UD field contains only the short message
> > ..0. .... = TP-SRQ: SMS STATUS REPORT is the result of a SMS SUBMIT.
> > .... 0... = TP-LP: The message has not been forwarded and is not a spawned message
> > .... .1.. = TP-MMS: No more messages are waiting for the MS in this SC
> > .... ..10 = TP-MTI: SMS-STATUS REPORT (2)
> > TP-MR: 43
> > TP-Recipient-Address - (destination number)
> > TP-Service-Centre-Time-Stamp
> > TP-Discharge-Time
> > TP-Status
> > 0... .... = Definition of bits 0-6: as follows
> > .000 0000 = Error: Short message transaction completed (0)
> > .000 0000 = Reason: Short message received by the SME (0)
> 
> is that because i send multiple SMS in 1 time?

Nope. When you send an SMS it is sent on the so called "uplink" (from phone to the network) and the response is send back on the so called "downlink" (from network to phone) and as you are capturing only the downlink it is expected to see only the response and not the original SMS. That's why I've asked you to test with receiving an SMS.

For the record a [link to your question](https://groups.google.com/d/topic/gr-gsm/IcNP8ZLv0gQ/discussion) in the mailing list. I prefer to keep the discussion in one place only.

haryows, 2019-07-23:
> Nope. When you send an SMS it is sent on the so called "uplink" (from phone to the network) and the response is send back on the so called "downlink" (from network to phone) and as you are capturing [...]

> SMS.For the record a link to your question in the mailing list. I prefer to keep the discussion in one place only.

I already move this case in mailing list, thank you very much for your suggest, very appreciate that :)

closed 2019-08-20 by ptrkrysik

samnq, 2020-04-18:
KC is wrong

nurzhan-zz, 2023-11-01:
My problem 😭