ptrkrysik/gr-gsm #473: Problem with HackRF

Dimtios, 2019-07-21:
Hello, I recently bought HackRF ONE and faced a problem, when using grgsm_livemon, everything goes well, but when you capture a channel, it becomes impossible to run it using grgsm_decode, wireshark does not see anything. With Rtl-Sdr v3, everything goes fine. Tell me what could be the problem with?

Capturing with:
`grgsm_capture -f 950.2M -g 40 -T 15 -s 1M out_hackrf_1S_15T.cfile`

Console info output:
```
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware git-509c8f4 
Decoding with:
grgsm_decode -f 950.2M -s 1M -c out_hackrf_1S_15T.cfile 
```

Responding to command hackrf_info
```

hackrf_info version: git-509c8f4
libhackrf version: git-509c8f4 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8385e1dcf
Board ID Number: 2 (HackRF One)
Firmware Version: git-509c8f4 (API:1.03)
Part ID Number: 0xa000cb3c 0x00554358
CPLD checksum: 0x05829db7
```

When using Rtl-Sdr v3 capturing with:
`grgsm_capture -f 950.2M -g 40 -T 15 -s 1M out_rtlsdr_1S_15T.cfile`

```
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
Reattached kernel driver

```
Decoding with:
`grgsm_decode -f 950.2M -s 1M -c out_rtlsdr_1S_15T.cfile `

[HackRF cfile ](https://drive.google.com/file/d/1snwBnhXliieVPY9IguH2Q5fZA75j07Nz/view?usp=drivesdk)
[Rtl-Sdr cfile](https://drive.google.com/file/d/108tV39Tclo6XPGgPyl9gYOtia8TrXhWO/view?usp=drivesdk)

velichkov, 2019-07-22:
Hi @Dimtios,

> I recently bought HackRF ONE and faced a problem, when using grgsm_livemon, everything goes well, but when you capture a channel, it becomes impossible to run it using grgsm_decode, wireshark does not see anything. With Rtl-Sdr v3, everything goes fine. Tell me what could be the problem with?

The HackRF has a big DC offset/spike that most probably causes this problem. 
[image]

See:
https://github.com/mossmann/hackrf/wiki/FAQ#what-is-this-big-spike-in-the-center-of-my-received-spectrum
https://github.com/mossmann/hackrf/wiki/FAQ#how-do-i-deal-with-the-dc-offset

`grgsm_livemon` works because it captures with an 400kHz offset but latest versions of `grgsm_capture` does not support capturing with an offset. This feature was removed in https://github.com/ptrkrysik/gr-gsm/commit/81ba56f59dcefb2d39d023c68d3a077b4a321520

See also.
https://groups.google.com/d/msg/gr-gsm/Zt47sSi3NOM/USE9IGlWAAAJ
https://groups.google.com/d/msg/gr-gsm/FhqNPtk3GME/28pMXO8SAgAJ
https://dsp.stackexchange.com/questions/40734/what-does-correcting-iq-do?newreg=aea502529d314116984a46b8e8b34e3a

> 
> Capturing with:
> `grgsm_capture -f 950.2M -g 40 -T 15 -s 1M out_hackrf_1S_15T.cfile`
> 
> Console info output:

One possible solution to the DC offset problem is to capture with some offset, on a different center frequency then the one that you need and then use `grgsm_channelize` script to extract the needed ARFCN or create a simple gnuradio flowgraph an use the Rotator block to shift the center frequency similar to what `grgsm_livemon.grc` does, together with a file source block and a file sink blocks.

Another solution that you could try is to use an older `grgsm_capture` version (https://github.com/ptrkrysik/gr-gsm/commit/57431ed50cb3e2530e6f5337fea9a98c4e6b4b2d) or try reverting https://github.com/ptrkrysik/gr-gsm/commit/81ba56f59dcefb2d39d023c68d3a077b4a321520

P.S.
We have a [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list) where you could post questions/problems like this one.

ptrkrysik, 2026-11-09:
The cause is the HackRF DC offset spike at the tuned frequency. Tune the HackRF away from the wanted channel (grgsm_livemon does this with a 400 kHz offset) or use another SDR. There is nothing to fix in gr-gsm.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik