ptrkrysik/gr-gsm #474: Compile crash on arfcn

usagi87, 2019-08-11:
Have been having problems compiling gr-gsm. It giving an error about arfcn:

Validation failed:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
        Param - Cell allocation(cell_allocation):
                Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
                name 'arfcn' is not defined

Block - import_1 - Import(import):
        Param - Import(import):
                Import "from grgsm import arfcn" failed.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1245: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

velichkov, 2019-08-12:
Hi @usagi87,

> Have been having problems compiling gr-gsm. It giving an error about arfcn:
>
> Validation failed:
>
> Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
> Param - Cell allocation(cell_allocation):
> Value [...]

Which OS do you use and what version?
How did you install gnuradio and gr-osmosdr - from package repository, manually or using pybombs and what versions?
Which `gr-gsm` version are you trying to compile - current master or older one?

See also:
- https://github.com/ptrkrysik/gr-gsm/issues/355
- https://osmocom.org/projects/gr-gsm/wiki/Installation

usagi87, 2019-08-12:
I'm using raspberry stretch, gnuradio installed from package repository and gr-osmosdr from git.osmocom.org/gr-osmosdr.  The thing is I was able to install it succesfully a few weeks ago when testing software.

velichkov, 2019-08-12:
> I'm using raspberry stretch, gnuradio installed from package repository and gr-osmosdr from git.osmocom.org/gr-osmosdr. 

Why from source and not from the apt repository?

> The thing is I was able to install it succesfully a few weeks ago when testing software.

Have you updated gnuradio or other system packages after installing gr-osmosdr? Try reinstalling it and then run `sudo ldconfig` just in case.

You haven't answered the last question from my previous post. Which gr-gsm version are you trying to compile - current master or older one?

usagi87, 2019-08-13:
Hey, I'm using the the  https://git.osmocom.org/gr-gsm version .

velichkov, 2019-08-13:
> Hey, I'm using the the https://git.osmocom.org/gr-gsm version .

OK, but could you tell me which commit id you are trying to compile - 54580d148d4c184483a5a398b8ca64545aadee4b (current master) or a different one?

Have you tried reinstalling `gr-osmosdr` and/or `sudo ldconfig`?

usagi87, 2019-08-13:
Hey, got into compile when using gr-osmosdr installed from package repository. I installed from git.osmocom.org/gr-osmosdr because the intructions I use to make my project.

velichkov, 2019-08-13:
> Hey, got into compile when using gr-osmosdr installed from package repository.

Cool :smile: 

>  I installed from git.osmocom.org/gr-osmosdr because the intructions I use to make my project.

Installing from source should work as well. Have you tried reinstalling and `sudo ldconfig` before installing the version from the apt repository?

jsa91, 2019-10-30:
Having the same issue:

`Scanning dependencies of target pygen_apps
[ 90%] Generating grgsm_livemon
Validation failed:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
        Param - Cell allocation(cell_allocation):
                Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
                name 'arfcn' is not defined

Block - import_1 - Import(import):
        Param - Import(import):
                Import "from grgsm import arfcn" failed.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1338: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
linaro@gsmtap:~/gr-gsm/build$
`
Trying to compile the latest version from github. 
Running Debian Stretch on a Tinker Board.

Any solution to this problem?

velichkov, 2019-10-30:
Hi @jsa91,

> Any solution to this problem?

Have you read all the previous messages? See https://github.com/ptrkrysik/gr-gsm/issues/474#issuecomment-520812428

How did you install `gnuradio` and `gr-osmosdr` and which versions? Have you attempted installations from source and have you uninstalled them afterwards?

Provide the full output of `find /usr/local/ -name "*osmosdr*" -o -name "*gnuradio*"` and from the gr-gsm's build directory `ldd -r lib/libgrgsm.so`

jsa91, 2019-10-31:
Installed gnuradio from repositories and gr-osmosdr from soruce. Tried also to rm gr-osmosdr once and tried again with no luck.

Gnuradio 3.7.10 and gr-osmosdr latest from Github.

> Provide the full output of find /usr/local/ -name "*osmosdr*" -o -name "*gnuradio*"

```
:~$ find /usr/local/ -name "*osmosdr*" -o -name "*gnuradio*"
/usr/local/share/doc/gr-osmosdr
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr__src__c.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1device.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1range__t.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1sink.xml
/usr/local/share/doc/gr-osmosdr/xml/namespaceosmosdr.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1device__t.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1time__spec__t.xml
/usr/local/share/doc/gr-osmosdr/xml/osmosdr__src__c_8h.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1source.xml
/usr/local/share/doc/gr-osmosdr/xml/structosmosdr_1_1meta__range__t.xml
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device.png
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1range__t-members.html
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t.html
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t.png
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t.js
/usr/local/share/doc/gr-osmosdr/html/osmosdr__src__c_8h.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source.png
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink.png
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1range__t.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1range__t.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c.png
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t.png
/usr/local/share/doc/gr-osmosdr/html/osmosdr__src__c_8h_source.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source-members.html
/usr/local/share/doc/gr-osmosdr/html/namespaceosmosdr.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t.png
/usr/local/share/doc/gr-osmosdr/html/namespaceosmosdr.html
/usr/local/share/doc/gr-osmosdr/html/osmosdr__src__c_8h.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink.html
/usr/local/share/gnuradio
/usr/local/share/gnuradio/grc/blocks/osmosdr_source.xml
/usr/local/share/gnuradio/grc/blocks/osmosdr_sink.xml
/usr/local/include/osmosdr
/usr/local/include/osmosdr/swig/osmosdr_swig.i
/usr/local/include/osmosdr/swig/osmosdr_swig_doc.i
/usr/local/lib/python2.7/dist-packages/osmosdr
/usr/local/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.so
/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.pyc
/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py
/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.pyo
/usr/local/lib/libgnuradio-osmosdr.so.0.1.5git
/usr/local/lib/pkgconfig/gnuradio-osmosdr.pc
/usr/local/lib/libgnuradio-osmosdr.so
```
[... 15 lines cut ...]
        libgnuradio-filter.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-filter.so.3.7.10 (0xb6c58000)
        libgnuradio-fft.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-fft.so.3.7.10 (0xb6c27000)
        libvolk.so.1.3 => /usr/lib/arm-linux-gnueabihf/libvolk.so.1.3 (0xb6bbf000)
        libosmocore.so.6 => /usr/lib/arm-linux-gnueabihf/libosmocore.so.6 (0xb6b9b000)
        libosmocodec.so.0 => /usr/lib/arm-linux-gnueabihf/libosmocodec.so.0 (0xb6b88000)
        libosmogsm.so.5 => /usr/lib/arm-linux-gnueabihf/libosmogsm.so.5 (0xb6b55000)
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb6a49000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb69d1000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb69a8000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb68ba000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb68a4000)
        /lib/ld-linux-armhf.so.3 (0xb6f81000)
        libboost_program_options.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_program_options.so.1.62.0 (0xb683e000)
        libboost_regex.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.62.0 (0xb677b000)
        liblog4cpp.so.5 => /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5 (0xb6744000)
[... cut ...]
        libfftw3f_threads.so.3 => /usr/lib/arm-linux-gnueabihf/libfftw3f_threads.so.3 (0xb6419000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6406000)
        liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0xb63a7000)
        libicudata.so.57 => /usr/lib/arm-linux-gnueabihf/libicudata.so.57 (0xb4b1a000)
        libicui18n.so.57 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.57 (0xb496d000)
        libicuuc.so.57 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.57 (0xb4844000)
        libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0xb4824000)
```

jsa91, 2019-10-31:
Installed gnuradio from repositories and gr-osmosdr from soruce initially. 
Now I also have tried with repo gr-osmosdr, compilation still fails.

Gnuradio 3.7.10 and gr-osmosdr latest from github/repositories.

> find /usr/local/ -name "*osmosdr*" -o -name "*gnuradio*"

```
~$ find /usr/local/ -name "*osmosdr*" -o -name "*gnuradio*"
/usr/local/share/doc/gr-osmosdr
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr__src__c.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1device.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1range__t.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1sink.xml
/usr/local/share/doc/gr-osmosdr/xml/namespaceosmosdr.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1device__t.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1time__spec__t.xml
/usr/local/share/doc/gr-osmosdr/xml/osmosdr__src__c_8h.xml
/usr/local/share/doc/gr-osmosdr/xml/classosmosdr_1_1source.xml
/usr/local/share/doc/gr-osmosdr/xml/structosmosdr_1_1meta__range__t.xml
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device.png
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1range__t-members.html
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t.html
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t.png
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device__t.js
/usr/local/share/doc/gr-osmosdr/html/osmosdr__src__c_8h.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source.png
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1device.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink.png
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1range__t.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1range__t.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c.png
/usr/local/share/doc/gr-osmosdr/html/structosmosdr_1_1meta__range__t.png
/usr/local/share/doc/gr-osmosdr/html/osmosdr__src__c_8h_source.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1source-members.html
/usr/local/share/doc/gr-osmosdr/html/namespaceosmosdr.js
/usr/local/share/doc/gr-osmosdr/html/classosmosdr__src__c-members.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1time__spec__t.png
/usr/local/share/doc/gr-osmosdr/html/namespaceosmosdr.html
/usr/local/share/doc/gr-osmosdr/html/osmosdr__src__c_8h.html
/usr/local/share/doc/gr-osmosdr/html/classosmosdr_1_1sink.html
/usr/local/share/gnuradio
/usr/local/share/gnuradio/grc/blocks/osmosdr_source.xml
/usr/local/share/gnuradio/grc/blocks/osmosdr_sink.xml
/usr/local/include/osmosdr
/usr/local/include/osmosdr/swig/osmosdr_swig.i
/usr/local/include/osmosdr/swig/osmosdr_swig_doc.i
/usr/local/lib/python2.7/dist-packages/osmosdr
/usr/local/lib/python2.7/dist-packages/osmosdr/_osmosdr_swig.so
/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.pyc
/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.py
/usr/local/lib/python2.7/dist-packages/osmosdr/osmosdr_swig.pyo
/usr/local/lib/libgnuradio-osmosdr.so.0.1.5git
/usr/local/lib/pkgconfig/gnuradio-osmosdr.pc
/usr/local/lib/libgnuradio-osmosdr.so
[... 15 lines cut ...]
        libgnuradio-filter.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-filter.so.3.7.10 (0xb6c73000)
        libgnuradio-fft.so.3.7.10 => /usr/lib/arm-linux-gnueabihf/libgnuradio-fft.so.3.7.10 (0xb6c42000)
        libvolk.so.1.3 => /usr/lib/arm-linux-gnueabihf/libvolk.so.1.3 (0xb6bda000)
        libosmocore.so.6 => /usr/lib/arm-linux-gnueabihf/libosmocore.so.6 (0xb6bb6000)
        libosmocodec.so.0 => /usr/lib/arm-linux-gnueabihf/libosmocodec.so.0 (0xb6ba3000)
        libosmogsm.so.5 => /usr/lib/arm-linux-gnueabihf/libosmogsm.so.5 (0xb6b70000)
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb6a64000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb69ec000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb69c3000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb68d5000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb68bf000)
        /lib/ld-linux-armhf.so.3 (0xb6f9d000)
        libboost_program_options.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_program_options.so.1.62.0 (0xb6859000)
        libboost_regex.so.1.62.0 => /usr/lib/arm-linux-gnueabihf/libboost_regex.so.1.62.0 (0xb6796000)
        liblog4cpp.so.5 => /usr/lib/arm-linux-gnueabihf/liblog4cpp.so.5 (0xb675f000)
[... cut ...]
        libfftw3f_threads.so.3 => /usr/lib/arm-linux-gnueabihf/libfftw3f_threads.so.3 (0xb6434000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6421000)
        liborc-0.4.so.0 => /usr/lib/arm-linux-gnueabihf/liborc-0.4.so.0 (0xb63c2000)
        libicudata.so.57 => /usr/lib/arm-linux-gnueabihf/libicudata.so.57 (0xb4b35000)
        libicui18n.so.57 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.57 (0xb4988000)
        libicuuc.so.57 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.57 (0xb485f000)
        libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0xb483f000)
```

jsa91, 2019-11-01:
Tried again from scratch, reinstalled OS and installed all dependencies from repositories. Now everything works as it should. Thanks for your help.

All3xJ, 2020-07-17:
I have the same problem. I'm on macOS. Can't compile gr-gsm with the same error.

ptrkrysik, 2026-11-09:
This was caused by conflicting source and distro installs of gr-osmosdr/GNU Radio and was solved by reinstalling from the distro packages. The new release uses a different module layout (gnuradio.gsm), so the old error no longer applies.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik