ptrkrysik/gr-gsm #476: cmake error

msfu101, 2019-08-27:
Hi there,
i have installed grgsm and gnuradio with pybombs, but it doesnt works good with my Hackrf.
if i compiled gr-gsm from source and type in cmake ..
it gives me this Output:

root@HackRF:~/gr-gsm/build# cmake ..
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Extracting version information from git describe...
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gr-gsm/build/CMakeFiles/CMakeError.log".

Can someone tell me what i have done wrong and how to solve this?
Thanks in advance!
msfu

msfu101, 2019-08-27:
Yes of course, i use Kali. 
This is a good advice, but can you tell me how to remove all pybombs related installations the right way?
I don't want to produca a failure. That would be nice!

msfu101, 2019-08-27:
Yes, i think it is a Problem with the nwéw gnuradio version.
I deinstalled pybombs and its related installations and installed gnuradio and gqrx from apt.
but if i try to compile gr-gsm it gave me the same Output as before:

root@HackRF:~/gr-gsm/build# cmake ..
-- The CXX compiler identification is GNU 9.2.1
-- The C compiler identification is GNU 9.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.23.0.rc1") 
-- Extracting version information from git describe...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'volk'
--   Found volk, version 2.0
-- Found VOLK: /usr/lib/x86_64-linux-gnu/libvolk.so  
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
-- Checking for module 'libosmocore'
--   Found libosmocore, version 0.12.0
-- Found libosmocore: /usr/lib/x86_64-linux-gnu/libosmocore.so  
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 0.12.0
-- Found libosmocodec: /usr/lib/x86_64-linux-gnu/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 0.12.0
-- Found libosmocoding: /usr/lib/x86_64-linux-gnu/libosmocoding.so  
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 0.12.0
-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so  
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gr-gsm/build/CMakeFiles/CMakeError.log".

Maybe i have to wait till the Support for gnuradio 3.8 is ready? Or do you know another solution? Because i dont want to change my Distribution at this time...

velichkov, 2019-08-27:
Hi @msfu101,

> i have installed grgsm and gnuradio with pybombs, but it doesnt works good with my Hackrf.

In my opinion the pybombs installation method is the most error prone and I do not recommend it.
It looks like you are using Debian/Ubuntu/Kali so I recommend installing `gnuradio` from the `apt` repository as described [here](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition) and then compile `gr-gsm` from source but first you need to remove everything installed by pybombs and pybombs itself.

> if i compiled gr-gsm from source and type in cmake ..
> it gives me this Output:
>
> root@HackRF:~/gr-gsm/build# cmake ..
> -- Build type not specified: defaulting to release.
> CMake Deprecation Warning at [...]

It seems you already have installed` gnuradio` from the apt repository and most probably you have two versions - one from apt in `/usr/lib` and one from pybombs in `/usr/local/` which could be a source of many hard to debug problems. If you decided to keep pybombs installation please remove all gnuradio packages, delete `/root/gr-gsm/build/` directory, create it again and repeat other steps.

> but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
> be NOT FOUND. Reason given by package:
>
> Gnuradio could not be found because dependency Volk could not be found.
>
> CMake Error [...]

Please post questions and installation problems like this one to our [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

velichkov, 2019-08-27:
> This is a good advice, but can you tell me how to remove all pybombs related installations the right way?

From https://github.com/ptrkrysik/gr-gsm/issues/447#issuecomment-467043085
```
pybombs inv
pybombs remove apache-thrift uhd gnuradio # add the names of other packages listed by `pybombs inv`
pip uninstall pybombs
cd /gsm/gr-gsm-master/
rm -rf build
mkdir build
cd build
cmake ..
make
```

velichkov, 2019-08-27:
Kali is based on Debian testing and few days ago they have updated gnuradio to 3.8 

https://pkg.kali.org/pkg/gnuradio

Support for gnuradio 3.8 is not ready and most probably you are going to face the same problems as in #475

So my next recommendation is to use Debian Buster or  Ubuntu 18.04 / 19.04.

velichkov, 2019-08-27:
> Maybe i have to wait till the Support for gnuradio 3.8 is ready? 

You could.

> Or do you know another solution? Because i dont want to change my Distribution at this time...

- You could try installing gnuradio 3.7 using pybombs :) but I'm not sure exactly how.
- You could try installing gr-gsm in a docker container based on a distribution with gnuradio 3.7.
- You could try installing in a VM based on a distribution with gnuradio 3.7.

Ghaz75, 2019-12-29:
Well I built complete gnuradio 3.8 from source without any error. Then I downloaded gr-gsm from this branch in root of my Kali linux 2019 .  When I build using cmake .. command it gives the same error that gnuradio runtime required to compile gr-gsm
Although gnuradio-runtime folder is there and everything is properly installed that is gnuradio and gnuradio-dev etc.

velichkov, 2020-01-03:
Hi @Ghaz75,

Could you provide the full cmake output attached as file? 

> Whereas it has found all necessary files .so etc in usr/lib/x86..... etc.

Installations from source usually goes to `/usr/local/lib` and not `/usr/lib`. Do you have the binary packages from the apt repository installed as well?

closed 2026-11-09 by ptrkrysik