ptrkrysik/gr-gsm #477: grgsm_decode pmt::deserialize

hackituria, 2019-08-28:
Hi,

I'm trying to decode obtained trafic but when i execute the following code:

`grgsm_decode -a 5 -b filename -t 0 -m BCCH`

The following error appear:

> terminate called after throwing an instance of 'pmt::exception'
>  what(): pmt::deserialize: malformed input stream, tag value = : 152
> Aborted

How can I fix this error?

Thank you.

velichkov, 2019-08-28:
Hi @hackituria,

> I'm trying to decode obtained trafic but when i execute the following code:
> 
> `grgsm_decode -a 5 -b filename -t 0 -m BCCH`

You are using bursts files. Which gr-gsm version do you use and how did you capture the burst file as this functionality was removed from the latest version of grgsm_capture ?
 
> The following error appear:
> 
> > terminate called after throwing an instance of 'pmt::exception'
> > what(): pmt::deserialize: malformed input stream, tag value = : 152
> > Aborted
> 
> How can I fix this error?

Are you sure that your burst file is valid. If you are c++ programmer you could try adding exception handling in https://github.com/ptrkrysik/gr-gsm/blob/54580d148d4c184483a5a398b8ca64545aadee4b/lib/misc_utils/burst_file_source_impl.cc#L87-L96

hackituria, 2019-09-26:
Hi,

Using the cfiles obtained by grgsm_capture script of your fork, `maint-3.8`, as you mentined on #480, I fixed this issue.

Thank you

closed 2026-11-09 by ptrkrysik