ptrkrysik/gr-gsm #480: gnuradio error

nocomp, 2019-09-17:
hi folks,
first of all thxx for this great tool i used to have working, but no idea what i ve did, almost spend a day to get it back on rails with no luck.
you are my last chance
here is the situation:
i ve installed on last kali version (used to work) using apt-get and now when i launch it i  get:

nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm$ grgsm_scanner 
Traceback (most recent call last):
  File "/usr/bin/grgsm_scanner", line 24, in <module>
    from gnuradio import blocks
ImportError: No module named gnuradio

so i ve pulled the git and when i try to build it i get for cmake ..

CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

i ve uninstall and reinstall gnuradio gr-gsm many times still same error :'(

grc is in  /usr/lib/python3/dist-packages/gnuradio/grc

and here are my paths:

sh -c 'set | grep -i path'
DBUS_SESSION_BUS_ADDRESS='unix:path=/run/user/1000/bus'
PATH='/home/nocomp/.local/bin:/home/nocomp/bin:/usr/local/sbin:/usr/sbin:/sbin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/usr/local/go/bin'
WINDOWPATH='2'

all kind of helps are welcome, receiving my limesdr in 2 days and my whole world is broken... #HELPPPPPPPPPPPPPPPPPP

thxx for your time

velichkov, 2019-09-17:
Hi @nocomp,

> first of all thxx for this great tool i used to have working, but no idea what i ve did, almost spend a day to get it back on rails with no luck.
> you are my last chance
> here is the situation:
> i ve installed on last kali version (used to work) using apt-get and now when i launch it i get:

It's quite typical for Kali (and other rolling distributions) something that have worked yesterday to no longer works today.

> nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm$ grgsm_scanner
> Traceback (most recent call last):
> File "/usr/bin/grgsm_scanner", line 24, in
> from gnuradio import blocks
> ImportError: No module named gnuradio

In Debian Testing and Kali they have updated gnuradio to 3.8 but still have not updated all dependent packages.

> so i ve pulled the git and when i try to build it i get for cmake ..
>
> CMake Warning at CMakeLists.txt:135 (find_package):
> Found package configuration file:
>
> ``` [...]

We are working on adding support for gnuradio 3.8 in #475. Try building from `maint-3.8` branch from [my fork](https://github.com/velichkov/gr-gsm/tree/maint-3.8) and have a look at [Kali.docker](https://github.com/velichkov/gr-gsm/blob/maint-3.8/tests/dockerfiles/Kali.docker) that is used in our [travis builds](https://travis-ci.org/velichkov/gr-gsm/builds/584836514)

nocomp, 2019-09-17:
hi
thx a lot for your reponse
i check that in a hour
thxx again for your time

nocomp, 2019-09-17:
-- Found libosmogsm: /usr/local/lib/libosmogsm.so  
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/home/nocomp/tools/sdr/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/nocomp/tools/sdr/gr-gsm/build/CMakeFiles/CMakeError.log".
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm/build$ 

so i guess there is nothing i can do ?

velichkov, 2019-09-17:
> CMake Error at CMakeLists.txt:150 (message):
GnuRadio Runtime required to compile gr-gsm

> so i guess there is nothing i can do ?

As I said in my previous message you need to checkout and build `maint-3.8` branch from my fork.

nocomp, 2019-09-17:
oh i thought i just had to git clone your repo and then go to the install process like the on in installation page.
do you have a tutorial i can follow?
thxx for your time

velichkov, 2019-09-17:
```
rm -rf gr-gsm
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
mkdir build
cd build
cmake ...
make
sudo make install
sudo ldconfig
```

nocomp, 2019-09-17:
damm , no luck :/
 62%] Building CXX object lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o
make[2]: ***  Aucune règle pour fabriquer la cible « /usr/lib/x86_64-linux-gnu/liborc-0.4.so », nécessaire pour « lib/libgrgsm.so.v0.41.1-342-g70cf13d6 ». Arrêt.
make[1]: *** [CMakeFiles/Makefile2:353: lib/CMakeFiles/grgsm.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

nocomp, 2019-09-17:
i ve installed liborcus-dev no luck same error

nocomp, 2019-09-17:
tried after installing sudo apt-get install liborc-0.4-0 no luck, same error

nocomp, 2019-09-17:
hmm just tried with doker, i ve saved te Kali-docker.txt and renamed it as docker-compose.yml but dot an error (not surprise noobs when it comes to docker)
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr$ docker-compose up -d
ERROR: yaml.parser.ParserError: expected '<document start>', but found '<scalar>'
  in "./docker-compose.yml", line 28, column 5
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr$

nocomp, 2019-09-17:
solved building docker, i keep you posted

nocomp, 2019-09-17:
well got docker error too :/

Removing intermediate container f7f273aac7ed
 ---> d8f696f11bbf
Step 7/7 : RUN PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH cmake .. &&     mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ &&     make -j $(nproc) &&     make install &&     ldconfig &&     make CTEST_OUTPUT_ON_FAILURE=1 test
 ---> Running in 377a86ee05f4
CMake Error: The source directory "/src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
The command '/bin/sh -c PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH cmake .. &&     mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ &&     make -j $(nproc) &&     make install &&     ldconfig &&     make CTEST_OUTPUT_ON_FAILURE=1 test' returned a non-zero code: 1
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/docker-grgsm$

velichkov, 2019-09-17:
> i ve installed liborcus-dev no luck same error
> tried after installing sudo apt-get install liborc-0.4-0 no luck, same error

You need to install `liborc-dev` and not `liborcus-dev` nor `liborc-0.4-0`. You could find the full list of packages in the docker file.

```
apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python3-docutils \
    python3-scipy \
    gnuradio-dev gnuradio libgmp-dev libboost-regex-dev liborc-dev \
    libosmocore-dev \
    gr-osmosdr
```

> well got docker error too :/
> 
> CMake Error: The source directory "/src" does not appear to contain CMakeLists.txt.

From the top gr-gsm directory you need to execute

```
docker build -f tests/dockerfiles/Kali.docker .
```
Note that there is a dot (.) at the end of the command, you need it.

nocomp, 2019-09-17:
> You need to install `liborc-dev` and not `liborcus-dev` nor `liborc-0.4-0`. You could find the full list of packages in the docker file.
>
> ```
> apt-get update && DEBIAN_FRONTEND=noninteractive apt-get [...]

hi
thxx a lot for your reply
i ve been able to compile following your advice, but when i run it, i get this module error:

nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm/build$ grgsm_scanner 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
ModuleNotFoundError: No module named 'grgsm'

;
what did i do wrong?

compilation worked like a charm with your fork, well done!
thxx for your time
cant find using pip or apt any module named like this

velichkov, 2019-09-17:
> nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm/build$ grgsm_scanner
> Traceback (most recent call last):
> File "/usr/local/bin/grgsm_scanner", line 33, in
> import grgsm
> ModuleNotFoundError: No module named 'grgsm'

You need to add `/usr/local/lib/python3/dist-packages/` to `PYTHONPATH`, in your shell execute

```
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
```
or to make it permanent add this to your shell profile (`~/.bashrc` or `~/.bash_profile`)

That's needed because of [this debian's patch](https://salsa.debian.org/bottoms/pkg-gnuradio/blob/unstable/debian/patches/debian-python-install#L8)

nocomp, 2019-09-17:
yeaaaaaaaaaaaaaaaaaaaaaaaaaa works like a charm,
thank you so much:!!!!
the docker installed work, but wont use it, need to fgure out how clean that in order to get some hd space.
thx again for your time, truelly appreciate.
next challenge: decode sms :))

velichkov, 2019-09-17:
> but wont use it, need to fgure out how clean that in order to get some hd space.

Run `docker ps -a` and then for every `CONTAINER ID` or `NAMES` run `docker rm 895e7a7b445d gnuradio-ubuntu-16.04 ....`

Then run `docker images` and for every `IMAGE ID` run `docker rmi c3e52da505c6 ...

> next challenge: decode sms :))

See our [Decoding-How-To](https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To). Please close this issue and post your questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list).

nocomp, 2019-09-17:
thx for your help!

nocomp, 2019-09-18:
hi,
just find out that grgsm_scanner is not working
no error message, just nothing showin up one you press enter (no hardware detection either as you can have with livemmon)

nocomp, 2019-09-18:
if i sudo i get 
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/phone-hack$ sudo grgsm_scanner 
[sudo] Mot de passe de nocomp : 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
ModuleNotFoundError: No module named 'grgsm'
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/phone-hack$

nocomp, 2019-09-18:
just tried using docker version:

build went fine:
Total Test time (real) =   4.71 sec
Removing intermediate container 5e20fdd9d7d4
 ---> 5047a68a9fcd
Successfully built 5047a68a9fcd

but it doesn t run
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm$ docker run 5047a68a9fcd
nocomp@8uR34ud3sL1c0rn35:~/tools/sdr/gr-gsm$ docker ps -a
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS                     PORTS               NAMES
c6b767394bd4        5047a68a9fcd        "bash"              3 seconds ago       Exited (0) 

not much logs either
nocomp@8uR34ud3sL1c0rn35:~/Documents$ docker logs fac618223384
nocomp@8uR34ud3sL1c0rn35:~/Documents$ 

hope this help
thxx for your time

nocomp, 2019-09-20:
just reinstalled and now it works, just takes a bit of time
thxx @velichkov for your priceless time

nocomp, 2019-09-20:
in fact there is an issue (sorry again) 
when i grgsm_scanner with your fork version i get only one result
if i grgsm_scanner on another computer that has it preinstalled i get 14 results

any idea why?

thxx for your help

velichkov, 2019-09-23:
> when i grgsm_scanner with your fork version i get only one result
> if i grgsm_scanner on another computer that has it preinstalled i get 14 results
>
> any idea why?

Do you use the same SDR device with the same antenna on both computers?
What are the hardware specs of the two computers - CPU, ram, USB port types?

nocomp, 2019-09-23:
hi,
yes same sdr device and same antenna and same computer lenovo x280 in multiboot
your fork on kali boot, and the normal version on HamOS 
8gb ram USB 3 ssd hd i3 8th gen

velichkov, 2019-09-23:
> and the normal version on HamOS

What is the exact version?
What SDR device do you use and how is connected to the host - USB 2/3 or ethernet?

nocomp, 2019-09-23:
hamos runs 0.42.2-1+b2
i ve used a standard nooelec blue rtlsdr to usb3

velichkov, 2019-09-23:
> i ve used a standard nooelec blue rtlsdr to usb3

NESDR Mini 2 or NESDR Mini 2+ ?

Run the following commands on both HamOS and Kali, wait them to complete, upload the full output somewhere and provide a link. Do not copy-past the whole output here as the long posts makes the discussion hard to follow.

```
lsusb -d 0bda:2832 -v
lsusb -d 0bda:2838 -v
timeout 1m rtl_test -p
timeout 5m rtl_test -s 2000000
```

Could you try compiling from source and testing `grgsm_scanner` the version from the `master` branch on HamOS? If there is a regression we need to determine if it the bug introduced between version `0.42.2` and `maser` or between `master` an my  `maint-3.8` branch.

> hamos runs 0.42.2-1+b2

Unfortunately this version unfortunately does not support the `--debug` flag.  As root open `/usr/bin/grgsm_scanner` with your favorite text editor, comment out lines 320 and 321, run `grgsm_scanner` as usual, wait it to finish, upload the full output and provide a link to it.

```
320             #os.dup2(null_fds[0], 1)
321             #os.dup2(null_fds[1], 2)
``` 

Also run `grgsm_scanner --debug ... ` on Kali, and again upload and provide a link to it.

nocomp, 2019-09-23:
hi,
thx a lot for your time, i ll do that tonite @work atm.
thxx again for your help

velichkov, 2019-09-23:
> 8gb ram USB 3 ssd hd i3 8th gen

Dose your CPU support AVX2 or FMA instructions?  Provide the output of `cat /proc/cpuinfo  | grep flags`

There is a known VOLK bug (https://github.com/gnuradio/volk/pull/278) and the fix for it is still not merged. The workaround is to set VOLK_GENERIC environment variable to 1.
```
export VOLK_GENERIC=1
grgsm_scanner
```

nocomp, 2019-09-23:
hi, here are the flags
nocomp@8uR34ud3sL1c0rn35:~/Documents/secu/emails/fraud1$ cat /proc/cpuinfo | grep flags
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d

velichkov, 2019-09-23:
> **fma** cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 **avx2**

So it supports both AVX2 and FMA instructions. My advice is to first try `export VOLK_GENERIC=1` and if it does not resolve the issue then execute all the other commands and provide their full outputs.

nocomp, 2019-09-23:
you were right!!!
works like a charm!
nocomp@8uR34ud3sL1c0rn35:~/Documents/secu/emails/fraud1$ export VOLK_GENERIC=1
nocomp@8uR34ud3sL1c0rn35:~/Documents/secu/emails/fraud1$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:0316 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04ca:7070 Lite-On Technology Corp. 
Bus 001 Device 002: ID 8087:0a2b Intel Corp. 
Bus 001 Device 014: ID 17ef:6019 Lenovo 
Bus 001 Device 015: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
nocomp@8uR34ud3sL1c0rn35:~/Documents/secu/emails/fraud1$ grgsm_scanner 

ARFCN:  979, Freq:  926.0M, CID:     0, LAC:     0, MCC:   0, MNC:   0, Pwr: -39
ARFCN: 1013, Freq:  932.8M, CID:     0, LAC:     0, MCC:   0, MNC:   0, Pwr: -33
ARFCN: 1016, Freq:  933.4M, CID: 18389, LAC: 22540, MCC: 208, MNC:  20, Pwr: -39
ARFCN: 1023, Freq:  934.8M, CID: 18258, LAC: 22540, MCC: 208, MNC:  20, Pwr: -46
ARFCN:    6, Freq:  936.2M, CID:     0, LAC:     0, MCC:   0, MNC:   0, Pwr: -28
ARFCN:   15, Freq:  938.0M, CID: 63138, LAC: 20993, MCC: 208, MNC:   1, Pwr: -33
ARFCN:  116, Freq:  958.2M, CID:     0, LAC: 26540, MCC: 208, MNC:  10, Pwr: -39
ARFCN:  120, Freq:  959.0M, CID:  7492, LAC: 26540, MCC: 208, MNC:  10, Pwr: -36
nocomp@8uR34ud3sL1c0rn35:~/Documents/secu/emails/fraud1$ 

do i need to add export VOLK_GENERIC=1 in my .bashrc to make it permanent?

velichkov, 2019-09-23:
> do i need to add export VOLK_GENERIC=1 in my .bashrc to make it permanent?

Yes, you could add it to .bashrc but then if you care about performance you need to monitor the [volk package](https://pkg.kali.org/pkg/volk) in the Kali repository and when they release a fixed version you'll need to remove this line.

And again please use our mailing list. Thanks.

nocomp, 2019-09-23:
many many manyyyyyyyyyyyyyyyyyyyy thanks!!! 
glad we spoted somethig that wasn t quite right
thanks again for your time

jsa91, 2019-12-05:
I followed the instructions above and successfully build it on Kali 2019.4.
i5, 8GB RAM, USB 2.0

Unfortunately, i get the following errors.

```
 ~# grgsm_scanner -b GSM900 -d
[ERROR] avahi_service_browser_new() failed: Bad state
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.8.0.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airsp                                             y airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
Reattached kernel driver
Args=
[ERROR] avahi_service_browser_new() failed: Bad state
Assertion 'close_nointr(fd) != -EBADF' failed at src/basic/fd-util.c:71, function safe_cl                                             ose(). Aborting.
Aborted
```
And with livemon

```
~# grgsm_livemon_headless -f 952.6M
[INFO] [UHD] linux; GNU C++ version 9.2.1 20191109; Boost_106700; UHD_3.14.1.1-1+b1
[ERROR] avahi_service_browser_new() failed: Bad state
Found Rafael Micro R820T tuner
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.8.0.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
 15 06 21 00 01 f0 2b 30 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 30 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
```
livemon seems to be running okey, although the initial error.
Any solutions for this issue?

velichkov, 2019-12-10:
Hi @jsa91,

>  ~# grgsm_scanner -b GSM900 -d
> [ERROR] avahi_service_browser_new() failed: Bad state

> [ERROR] avahi_service_browser_new() failed: Bad state
> Assertion 'close_nointr(fd) != -EBADF' failed at src/basic/fd-util.c:71, function safe_close(). Aborting.

This seems like either avahi or systemd bug.

Try running both `grgsm_scanner` and `grgsm_livemon_headless` with `--args=rtl` argument.

```
grgsm_scanner --args=rtl -d
grgsm_livemon_headless  --args=rtl -f 952.6M
```
and provide the full output if you still see the above errors.

jsa91, 2019-12-10:
Hi @velichkov ,

Took you tips from #475 and installed Debian 10 instead. Now running gr-gsm without problems.
Thanks anyway.

Lohith12, 2019-12-25:
Strucked at 70 % when typed make command. What should i do now ?

Lohith12, 2019-12-25:
Getting error while executing grgsm_livemon it says 
Traceback( most recent call last): 
File "/usr/local/bin/grgsm_livemon",line 37 in <module> from grgsm import arfcn 
ModuleNotFoundError: No module named 'grgsm'

velichkov, 2019-12-27:
Hi @Lohith12,

> Strucked at 70 % when typed make command. What should i do now ?

Give us the last 20 lines of the output.

> Getting error while executing grgsm_livemon it says
> Traceback( most recent call last):
> File "/usr/local/bin/grgsm_livemon",line 37 in from grgsm import arfcn
> ModuleNotFoundError: No module named 'grgsm'

See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

kiransgk1si18lvs04, 2020-01-28:
Sir, everyone is having a problem in installing gr-gsm please make a vedio tutorial of present installation using GitHub library.
Thank you

anandpandey786, 2020-02-11:
oot@kaali:~# grgsm_livemon  --args=rtl -f 952.4M
gr-osmosdr 0.1.5 (0.1.4) gnuradio 3.8.0.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!

** (python3:2745): WARNING **: 13:01:29.983: Invalid borders specified for theme pixmap:
        /usr/share/themes/Kali-Dark/gtk-2.0/assets/trough-scrollbar-horiz.png,
borders don't fit within the image

** (python3:2745): WARNING **: 13:01:29.983: invalid source position for vertical gradient

** (python3:2745): WARNING **: 13:01:29.983: invalid source position for vertical gradient

** (python3:2745): WARNING **: 13:01:29.983: invalid source position for vertical gradient

** (python3:2745): WARNING **: 13:01:29.983: invalid source position for vertical gradient

avahi bad state error was also seen which was eradicated by args argument but invalid source error...what is the reason for that... livemon monitor is running but gsm data being captured so far

anandpandey786, 2020-02-11:
grgsm_livemon could not capturing GSM data

kiransgk1si18lvs04, 2020-02-11:
Hi @anandpandey786 
Use this command grgsm_livemon_headless --args=rtl -f 952.4M
you have missed headless in command.

anandpandey786, 2020-02-11:
Thanks for reply
I deliberated omitted headless .... wanted to see the pixmap/monitor
Every thing is running in my build but despite that gsm data is not visible on monitor tried several frequencies...

velichkov, 2020-02-11:
Hi @anandpandey786,

> avahi bad state error was also seen which was eradicated by args argument

What args did you used to solve the avahi errors?

> but invalid source error...what is the reason for that...

I don't know.

> livemon monitor is running but no gsm data being captured so far
> Every thing is running in my build but despite that gsm data is not visible on monitor tried several frequencies...

First try finding a frequency of a nearby C0 BCCH channel using `grgsm_scanner` and then specify that frequency when running `grgsm_livemon`. Also make sure there are GSM networks where you live in the bands supported by your RTL device.

anandpandey786, 2020-02-13:
> Hi @anandpandey786,
>
>
> What args did you used to solve the avahi errors?
>
>
> I don't know.
>
>
> First try finding a frequency of a nearby C0 BCCH channel using `grgsm_scanner` and then specify that [...]

Hi velichkov
Thanks for your reply .... issue was resolved there was some glitch in my antenna i am using your 3.8maint .
regards

twigs538, 2020-03-15:
Hi,

I'm using a RTL2838 SDR with Ubuntu 16. I followed your installation instructions from the osmo website hyperlink from ptrksky grmgsm github repository to the letter.

Everything works except when I run grgsm_livemon I keep getting 0's displayed.

grgsm_scanner returns no results when scanning all available bands. 
Kalibrate-rtl RETURNS GSM stations and no other bands.
Kalibrate has given me my ppm. 

Rtl_test returns no errors in my signals. I have changed sample rates, used VM and a desktop with Ubuntu 16 returning the same results.

What else can I do?

velichkov, 2020-03-16:
Hi @twigs538,

Please avoid posting in an existing issues not directly related to your issue. Please post such questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list) and do not send me  emails directly.
 
> I'm using a RTL2838 SDR with Ubuntu 16. I followed your installation instructions from the osmo website hyperlink from ptrksky grmgsm github repository to the letter.
>
> Everything works except when I [...]

Kalibrate-rtl is known to return false positive results.

> Kalibrate has given me my ppm.
> 
> Rtl_test returns no errors in my signals. I have changed sample rates, used VM and a desktop with Ubuntu 16 returning the same results.
> 
> What else can I do?

Do you have a suitable antenna attached to your RTL dongle? Are you sure that there are GSM networks in the area where you are and in the band you are searching in? In case you are inside in a big building try moving near a window or outside.

In case you live in USA note that AT&T [stopped their 2G network](https://about.att.com/innovationblog/2g_sunset) in the GSM850 band back in 2017 and for T-Mobile network in the GSM1900 band you need a dongle with Elonics E4000 tuner or another SDR device that supports that band.

kiransgk1si18lvs04, 2020-04-10:
Hi @velichkov 
I am successful in installing grgsm from your maint branch and I am get all BCCH data like MCC,MNC,LAC and some other things, but there is problem when designing simple gui application using tkinter. I am attaching code below  

import tkinter as tk
import subprocess
import shlex

def BCCH():
                cmd=["grgsm_scanner","--args=rtl","-b","GSM900"]
                output= subprocess.check_output(cmd)
                result=print('>', output)
                # put result in label
                result['text'] = output.decode('utf-8')
               
my_gui = tk.Tk()
my_gui.geometry('600x600')
my_gui.title("Get output inside GUI") 

tk.Button(my_gui,text="GET BCCH", command=BCCH).pack() 

# label for BCCH result
result = tk.Label(my_gui,font=50)
result.pack()

my_gui.mainloop()

I am getting excemptions thinking that some files are missing.
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/kali/test.py", line 7, in BCCH
    output= subprocess.check_output(cmd)
  File "/usr/lib/python3.7/subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['grgsm_scanner', '--args=rtl', '-b', 'GSM900']' returned non-zero exit status 1.

when I type in command line interface as kal -s GSM900 the output is coming. I think I will get solution. 
Thanks

kiransgk1si18lvs04, 2020-04-10:
When I try adding exclusions.

import tkinter as tk
import subprocess
import shlex
def BCCH():
    while True:
        try:
                cmd=["grgsm_scanner","--args=rtl","-b","GSM900"]
                output= subprocess.getoutput(cmd)
                result=print('>', output)
                # put result in label
                if not output is None:
                    result['text'] = output.decode('utf-8')
        except subprocess.CalledProcessError as exc:
           print(exc.returncode)
        
my_gui = tk.Tk()

my_gui.geometry('600x600')
my_gui.title("Get output inside GUI") 
tk.Button(my_gui,text="GET BCCH", command=BCCH).pack() 
# label for BCCH result
result = tk.Label(my_gui,font=50)
result.pack()

my_gui.mainloop()

The errors are like this
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
ModuleNotFoundError: No module named 'grgsm'

velichkov, 2020-04-10:
Hi @kiransgk1si18lvs04,

> ModuleNotFoundError: No module named 'grgsm'

Make sure you are running your application with the same python version that gr-gsm and gnuradio was compiled with (most probably python3), add the following as the first line in your scripts.
```
#!/usr/bin/python3
```
If it still does not work you may need to add the path where the gr-gsm's python module was installed to the `PYTHONPATH` environment variable, see https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

P.S.
Please next time try using [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages as right now they are hardly readable and avoid hijacking unrelated github issues.

kiransgk1si18lvs04, 2020-04-10:
Hi@velichkov
I have added python path permanently by using your instruction.
# export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH

I am running the scripts on python 3.7.6

kiransgk1si18lvs04, 2020-04-27:
Hi@velichkov 
I am using python 3.7.6.
pythonpath added still program is not working.

#!/usr/bin/python3
import tkinter as tk
import subprocess
import shlex
def BCCH():
    while True:
        try:
                cmd=["grgsm_scanner","--args=rtl","-b","GSM900"]
                output= subprocess.getoutput(cmd)
                result=print('>', output)
                # put result in label
                if not output is None:
                    result['text'] = output.decode('utf-8')
        except subprocess.CalledProcessError as exc:
           print(exc.returncode)
        
                
                
          
my_gui = tk.Tk()

my_gui.geometry('600x600')
my_gui.title("Get output inside GUI") 
tk.Button(my_gui,text="GET BCCH", command=BCCH).pack() 
# label for BCCH result
result = tk.Label(my_gui,font=50)
result.pack()

my_gui.mainloop()

velichkov, 2020-04-27:
Hi @kiransgk1si18lvs04,

> pythonpath added still program is not working.

What do you mean by "not working"?

Double check that the environment variable is exported correctly and that `grgsm` module is installed in `/usr/local/lib/python3/dist-packages/`
```
echo $PYTHONPATH
ls `/usr/local/lib/python3/dist-packages/grgsm/
```

kiransgk1si18lvs04, 2020-04-27:
sorry @velichkov  
grgsm is inside the folder of `/usr/local/lib/python3/dist-packages/'
grgsm_scanner works on command terminal in kali linux, but not works on above program.

The error details are,
```
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 33, in <module>
    import grgsm
ModuleNotFoundError: No module named 'grgsm'
```

This is bash_profile 

```
 GNU nano 4.5                                                                                      .bash_profile                                                                                       Modified  
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH

```

velichkov, 2020-04-27:
Hi @kiransgk1si18lvs04,

> grgsm_scanner works on command terminal in kali linux, but not works on above program.

How do you start the above program?  In your program somewhere at the top add the following and give me the output
```
import sys
print(sys.path)
```
and also give me the full output of `ls /usr/local/lib/python3/dist-packages/grgsm`

kiransgk1si18lvs04, 2020-04-27:
> How do you start the above program?  In your program somewhere at the top add the following and give me the output
By using .py file

`['/home/kali', '/usr/bin', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.7/dist-packages']`

> and also give me the full output of ls /usr/local/lib/python3/dist-packages/grgsm
```
kali@kali:~$ ls /usr/local/lib/python3/dist-packages/grgsm
arfcn.py                           clock_offset_corrector_tagged.pyo  fcch_detector.pyc  _grgsm_swig.so                     gsm_gmsk_mod.pyc         __init__.py              txtime_bursts_tagger.pyc
arfcn.pyc                          device.py                          fcch_detector.pyo  gsm_bcch_ccch_demapper.py          gsm_gmsk_mod.pyo         __init__.pyc             txtime_bursts_tagger.pyo
arfcn.pyo                          device.pyc                         fn_time.py         gsm_bcch_ccch_demapper.pyc         gsm_input.py             __init__.pyo
chirpz.py                          device.pyo                         fn_time.pyc        gsm_bcch_ccch_demapper.pyo         gsm_input.pyc            sch_detector.py
chirpz.pyc                         fcch_burst_tagger.py               fn_time.pyo        gsm_bcch_ccch_sdcch4_demapper.py   gsm_input.pyo            sch_detector.pyc
chirpz.pyo                         fcch_burst_tagger.pyc              grgsm_swig.py      gsm_bcch_ccch_sdcch4_demapper.pyc  gsm_sdcch8_demapper.py   sch_detector.pyo
clock_offset_corrector_tagged.py   fcch_burst_tagger.pyo              grgsm_swig.pyc     gsm_bcch_ccch_sdcch4_demapper.pyo  gsm_sdcch8_demapper.pyc  trx
clock_offset_corrector_tagged.pyc  fcch_detector.py                   grgsm_swig.pyo     gsm_gmsk_mod.py                    gsm_sdcch8_demapper.pyo  txtime_bursts_tagger.py

```

velichkov, 2020-04-27:
> By using .py file

How exactly? Do you run it from some IDE or from the command line?

> ```
> ['/home/kali', '/usr/bin', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.7/dist-packages']
> ```
The `/usr/local/lib/python3/dist-packages/grgsm` is not in `sys.path` and this explains why it can't find the grgsm module.

> This is bash_profile
>```
> export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
>```

Try adding the export line to your `~/.profile` and logout and login again. See also https://help.ubuntu.com/community/EnvironmentVariables#A.2BAH4-.2F.profile

[9 more replies]

closed 2026-11-09 by ptrkrysik