ptrkrysik/gr-gsm #481: qr-qtgui checked failed

hackituria, 2019-09-18:
Hi,

I am trying to install _gr-gsm_ on my _Kali Linux 2019.3_. When I execute the following command:

`sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default`

I receive the following result:

```
...
-- Python checking for PyQt4
-- Python checking for PyQt4 - found
-- 
-- Configuring gr-qtgui support...
--   Dependency Boost_FOUND = 1
--   Dependency QT4_FOUND = TRUE
--   Dependency QWT_FOUND = FALSE
--   Dependency ENABLE_VOLK = ON
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency PYQT4_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gr-qtgui but configuration checked failed
Call Stack (most recent call first):
  gr-qtgui/CMakeLists.txt:44 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gnuradio/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```
What can I do to fix this issue?

Thank you..

velichkov, 2019-09-18:
Hi @hackituria ,

> I am trying to install _gr-gsm_ on my _Kali Linux 2019.3_. When I execute the following command:
> 
> `sudo pybombs prefix init /usr/local -a default_prx -R gnuradio-default`

In my opinion the `pybombs` installation method is the hardest and most error prone one so I do not recommend it to non programmers. 

You could install gnuradio and gr-osmosdr from the apt repository and then clone my [fork](https://github.com/velichkov/gr-gsm/tree/maint-3.8/) checkout branch `maint-3.8` and build it.

Also have a look at [Kali.docker](https://github.com/velichkov/gr-gsm/blob/maint-3.8/tests/dockerfiles/Kali.docker),  #480 and #475.  

> I receive the following result:
>
> ```
> ...
> -- Python checking for PyQt4
> -- Python checking for PyQt4 - found
> --
> -- Configuring gr-qtgui support...
> --   Dependency Boost_FOUND = 1
> --   Dependency [...]

See  https://github.com/gnuradio/gr-recipes/issues/156

Which gnuradio version did you try to install and ave you modified any of the pybomb's recipes?

If you decide to use pybombs and have other problems not directly related to compilation of gr-gsm, please report them in the [gr-recipes](https://github.com/gnuradio/gr-recipes/) project.

And in the future please use out [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list) for installation problems, questions, etc...

hackituria, 2019-09-26:
Hi,

I install your fork and the issue is fixed. The problem was the incompatibility with Gnuradio 3.8.

Thank you.

RJHughes, 2020-06-08:
For anyone else trying to make this work, I had the same issue in Ubuntu 18.04 and the exact same error message as @hackituria. The error points out that QWT_FOUND = False. So I resolved my issue by install QWT via

> sudo apt install libqwt-dev

ptrkrysik, 2026-11-09:
This is a GNU Radio/pybombs build issue (missing libqwt-dev), solved in the thread, and not a gr-gsm defect.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik