ptrkrysik/gr-gsm #482: Only seeing one Mobile Network Code (MNC) within area

gooroo363, 2019-10-01:
Good evening, 
I was wondering if anyone had any thoughts as to why I might be only seeing a single MNC (260: T-Mobile) within my area. I'm not located in a major city, but I would think I should at least also be able to pickup AT&T cells as well. 

**Area:**
https://opencellid.org/#zoom=14&lat=43.45092&lon=-71.58555

**SDR Hardware:**
Hackrf

**Commands Used:**
- grgsm_scanner -b PCS1900 -40  --args=hackrf -v
- grgsm_scanner -b GSM850 -g 40 --args=hackrf -v

**Frequency Table:**
https://en.wikipedia.org/wiki/GSM_frequency_bands

I'm assuming that I'm probably missing something in the learning curve, but I thought I'd reach out and ask since my "Googling" isn't returning much luck.

velichkov, 2019-10-02:
Hi @gooroo363 

> I was wondering if anyone had any thoughts as to why I might be only seeing a single MNC (260: T-Mobile) within my area. I'm not located in a major city, but I would think I should at least also be able to pickup AT&T cells as well.

AT&T stopped their GSM 2G network back in 2017. See https://about.att.com/innovationblog/2g_sunset

gooroo363, 2019-10-02:
Thanks @velichkov, 
I was thinking that something like that might have been the case. So our options are to use gr-gsm for the 2G networks and something like OpenLTE for the 3G and 4G LTE networks?

velichkov, 2019-10-02:
> So our options are to use gr-gsm for the 2G networks and something like OpenLTE for the 3G and 4G LTE networks?

Yes, `gr-gsm` supports only GSM 2G. For LTE 4G you could use [gr-lte](https://github.com/kit-cel/gr-lte), [srsLTE](https://github.com/srsLTE/srsLTE), [OpenAirInterface](https://www.openairinterface.org/) or [OpenLTE](http://openlte.sourceforge.net/). The only 3G UMTS open source project that I'm aware of is [OpenBTS-UMTS](https://github.com/RangeNetworks/OpenBTS-UMTS).

Please post your further questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list).

gooroo363, 2019-10-02:
Awsome thanks!

closed 2019-10-02 by gooroo363