ptrkrysik/gr-gsm #483: In Kali linux, it says problem occured while building package qwt6

kburke15, 2019-10-04:
I have tried everything to get gr-gsm installed on my kali linux, but it keeps coming up with the same thing. what the heck is qwt6? It just needs to be fixed so I can use it.
Thanks,
Kevin Burke

velichkov, 2019-10-04:
Hi @kburke15,

> I have tried everything to get gr-gsm installed on my kali linux, 

You could find instructions how to install `gr-gsm` on Kali in the following issues:

- https://github.com/ptrkrysik/gr-gsm/issues/481#issuecomment-532782543
- https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532162198
- https://github.com/ptrkrysik/gr-gsm/issues/476#issuecomment-525407538

> but it keeps coming up with the same thing. 

Report the qwt6 problem in the [gr-recipes](https://github.com/gnuradio/gr-recipes) project and don't forget to provide logs.

> what the heck is qwt6?

https://qwt.sourceforge.io/

kburke15, 2019-10-04:
Thanks, I just finished reporting the problem in the gr-recipes project including the logs.

velichkov, 2019-10-04:
For the record the qwt6 issue is https://github.com/gnuradio/gr-recipes/issues/161

closed 2019-10-07 by kburke15