ptrkrysik/gr-gsm #484: while installing gr-gsm via pybombs, it says that it is not compatible with gnuradio 3.8.0

kburke15, 2019-10-07:
That's basically what it says and then it aborts the installation.

velichkov, 2019-10-07:
Hi @kburke15,

> while installing gr-gsm via pybombs, it says that it is not compatible with gnuradio 3.8.0
> That's basically what it says and then it aborts the installation.

In https://github.com/ptrkrysik/gr-gsm/issues/483#issuecomment-538446970 I gave you three links, have you read them?

From https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532162198

> We are working on adding support for gnuradio 3.8 in #475. Try building from `maint-3.8` branch from my fork ....

P.S.
Please post any further questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

ptrkrysik, 2026-11-09:
The old pybombs recipe was incompatible with GNU Radio 3.8; the new release targets GNU Radio 3.10 and builds with CMake.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik