ptrkrysik/gr-gsm #485: Error : Validation failed when compiling gr-gsm

r00tb3, 2019-10-27:
Hi all I use the `gnuradio verison 3.7.13.5` compiled manually also `gr-osmosdr version 0.1.4-14build1` installed from Ubuntu repo when I compile the master branch of the gr-gsm I receive the following error:

CMake:
```
cmake ../
.
.
.
-- Extracting version information from git describe...
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: FILTER
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-filter.so;/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/usr/local/include
 * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_PMT_FOUND = TRUE
-- Loading build date Sun, 27 Oct 2019 10:48:54 into constants...
-- Loading version 0.42.2. into constants...
-- Using install prefix: /usr/local
-- Building for version: 0.42.2. / 0.42.2.
-- Configuring done
CMake Warning at lib/CMakeLists.txt:98 (add_library):
  Cannot generate a safe runtime search path for target grgsm because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libosmocodec.so.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

CMake Warning at /usr/local/lib/cmake/gnuradio/UseSWIG.cmake:263 (add_library):
  Cannot generate a safe runtime search path for target _grgsm_swig because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libosmocodec.so.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:189 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:49 (GR_SWIG_MAKE)

-- Generating done

```

For make I receive following error:
```
make -j4
[  2%] Built target pygen_python_8b6fd
[  4%] Built target pygen_python_misc_utils_34496
[  4%] Built target grgsm_swig_swig_doc
[  9%] Built target pygen_python_demapping_648e2
[ 10%] Built target pygen_python_transmitter_394d4
[  9%] Built target pygen_python_receiver_4ed28
[ 73%] Built target grgsm
[ 75%] Built target pygen_apps_helpers
[ 76%] Built target man_grgsm_livemon.rst
[ 79%] Built target pygen_python_trx_a3216
[ 81%] Built target _grgsm_swig_swig_tag
[ 84%] Built target grgsm_swig_swig_2d0df
[ 87%] Built target pygen_swig_95332
[ 92%] Built target _grgsm_swig
[ 93%] Generating grgsm_livemon
[ 94%] Generating grgsm_livemon_headless
Validation failed:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
	Param - Cell allocation(cell_allocation):
		Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
		name 'arfcn' is not defined

Block - import_1 - Import(import):
	Param - Import(import):
		Import "from grgsm import arfcn" failed.
Error during file compilation.
Validation failed:

Block - gsm_receiver_0 - GSM Receiver(gsm_receiver):
	Param - Cell allocation(cell_allocation):
		Value "[arfcn.downlink2arfcn(fc)]" cannot be evaluated:
		name 'arfcn' is not defined

Block - import_1 - Import(import):
	Param - Import(import):
		Import "from grgsm import arfcn" failed.
Error during file compilation.
apps/CMakeFiles/pygen_apps.dir/build.make:89: recipe for target 'apps/grgsm_livemon_headless' failed
make[2]: *** [apps/grgsm_livemon_headless] Error 1
make[2]: *** Waiting for unfinished jobs....
apps/CMakeFiles/pygen_apps.dir/build.make:84: recipe for target 'apps/grgsm_livemon' failed
make[2]: *** [apps/grgsm_livemon] Error 1
CMakeFiles/Makefile2:1267: recipe for target 'apps/CMakeFiles/pygen_apps.dir/all' failed
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm not able to figure out whats wrong!!

Is it the SWIG version or something else?

Regards.

velichkov, 2019-10-27:
Hi @r00tb3,

> Hi all I use the `gnuradio verison 3.7.13.5` compiled manually also `gr-osmosdr version 0.1.4-14build1` installed from Ubuntu repo 

You can't mix them like this as `gr-osmosdr` depends on `gnuradio` and have to be compiled with the same version you've compiled from source.

Why do you wan't to install gnuradio from source and not from the Ubuntu repository?

> I'm not able to figure out whats wrong!!
> 
> Is it the SWIG version or something else?

The mix is the most probable explanation. See also 

- https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition
- https://github.com/ptrkrysik/gr-gsm/blob/master/tests/dockerfiles/Ubuntu_16_04.docker

r00tb3, 2019-10-27:
> You can't mix them like this as `gr-osmosdr` depends on `gnuradio` and have to be compiled with the same version you've compiled from source.

Okay, I've uninstalled gr-osmosdr that I had installed from ubuntu repo now git cloned and installed manually still I get the same error.

How do I proceed ahead?

> Why do you wan't to install gnuradio from source and not from the Ubuntu repository?

I need the latest version of the maint-3.7 branch for srsLTE setup also I'm on Ubuntu 18.04.3 LTS which has gnuradio version 3.7.11 in the Ubuntu repo.

> The mix is the most probable explanation. See also

I've followed the link you've provided but I haven't installed the gnuradio-dev, gr-osmosdr, libosmocore-dev as I had installed them manually through git.

velichkov, 2019-10-27:
> Okay, I've uninstalled gr-osmosdr that I had installed from ubuntu repo now git cloned and installed manually still I get the same error.
>
> How do I proceed ahead?

Have you also uninstalled all gnuradio related packages as well? [gr-osmosdr](https://packages.ubuntu.com/bionic/gr-osmosdr) depends on `gnuradio`, `libgnuradio-pmt3.7.11`,   `libgnuradio-runtime3.7.11` and `libgnuradio-osmosdr0.1.4` packages so make sure they are not installed or remove them if they are. Having two versions of gnuradio installed on the same time, one from source and one from packages, could lead to multiple hard to resolve problems.

> > Why do you wan't to install gnuradio from source and not from the Ubuntu repository?
> 
> I need the latest version of the maint-3.7 branch for srsLTE setup

AFAIK `srsLTE` does **not** depends on gnuradio but only on `UHD` and only in case you want to use it with USRP SDR device.

>  also I'm on Ubuntu 18.04.3 LTS which has gnuradio version 3.7.11 in the Ubuntu repo.

If this version does not work for you my advice would be to contact the package maintainers, report to them all problems you are having with this version and request version update.

> > The mix is the most probable explanation. See also
> 
> I've followed the link you've provided but I haven't installed the gnuradio-dev, gr-osmosdr, libosmocore-dev as I had installed them manually through git.

After removing all gnuradio's packages do a clean rebuild (remove build directory and start from scratch) to ensure everything is build against the versions installed from source and if it sill does not work execute `ldd -r ./lib/libgrgsm.so` from the build directory and provide the full output.

closed 2019-11-05 by r00tb3