ptrkrysik/gr-gsm #486: GnuRadio Runtime required to build gr-osmosdr

serj053, 2019-10-29:
Good time of day. Wandering around the Internet in search of a solution to your question, I met your explanation of the problem associated with gnuradio, I had hope for help from a knowledgeable person, if you do not mind.. The problem is that I can not compile the block gr-osmosdr, won a mountain of errors but I can not win the lack of runtime, if you can help solve the problem please. (Translated with the help of an electronic translator because there may be inaccuracies in the meanings). Sincerely Sergey.

serj@serj:~/gr-osmosdr/build$ cmake ../ -DCMAKE_CXX_COMPILER=/usr/bin/g++-7 -DCMAKE_C_COMPILER=/usr/bin/gcc-7
-- The CXX compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/g++-7
-- Check for working CXX compiler: /usr/bin/g++-7 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Extracting version information from git describe...
-- Configuring Boost C++ Libraries...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.65.1") found components:  thread system chrono date_time atomic 
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.65.1") found components:  date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- Found VOLK: Volk::volk  
CMake Warning (dev) at /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:38 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  CMakeLists.txt:154 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:
  Policy CMP0011 is not set: Included scripts do automatic cmake_policy PUSH
  and POP.  Run "cmake --help-policy CMP0011" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  The included script

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake

  affects policy settings.  CMake is implying the NO_POLICY_SCOPE option for
  compatibility, so the effects are applied to the including context.
Call Stack (most recent call first):
  CMakeLists.txt:154 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'gnuradio-iqbalance'
--   Found gnuradio-iqbalance, version 0
-- Found GNURADIO_IQBALANCE: /usr/lib/x86_64-linux-gnu/libgnuradio-iqbalance.so  
-- Checking for module 'uhd'
--   Found uhd, version 003.010.003.000-0-unknown
-- Found UHD: /usr/lib/x86_64-linux-gnu/libuhd.so  
-- Checking for module 'gnuradio-uhd'
--   Found gnuradio-uhd, version 3.7.11
-- Found gnuradio-uhd: /usr/include, /usr/lib/x86_64-linux-gnu/libgnuradio-uhd.so
-- Found GNURADIO_UHD: /usr/lib/x86_64-linux-gnu/libgnuradio-uhd.so  
-- Checking for module 'gnuradio-fcd'
--   Found gnuradio-fcd, version 3.7.11
-- Found gnuradio-fcd: /usr/include, /usr/lib/x86_64-linux-gnu/libgnuradio-fcd.so
-- Found GNURADIO_FCD: /usr/lib/x86_64-linux-gnu/libgnuradio-fcd.so  
-- Checking for module 'gnuradio-fcdproplus'
--   Found gnuradio-fcdproplus, version 3.7.11
-- Found gnuradio-fcdproplus: /usr/include, /usr/lib/x86_64-linux-gnu/libgnuradio-fcdproplus.so
-- Found GNURADIO_FCDPP: /usr/lib/x86_64-linux-gnu/libgnuradio-fcdproplus.so  
-- Checking for module 'libosmosdr'
--   Found libosmosdr, version 
-- Found libosmosdr: /usr/include, /usr/lib/x86_64-linux-gnu/libosmosdr.so
-- Checking for module 'librtlsdr'
--   Found librtlsdr, version 0.6.0-5-g8183
-- Found librtlsdr: /usr/local/include, /usr/local/lib/librtlsdr.so
-- Checking for module 'libmirisdr'
--   Found libmirisdr, version 0.0.0
-- Found libmirisdr: /usr/include, /usr/lib/x86_64-linux-gnu/libmirisdr.so
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.5
-- Found LIBHACKRF: /usr/lib/x86_64-linux-gnu/libhackrf.so  
-- Checking for module 'libairspy'
--   Found libairspy, version 1.0
-- Found LIBAIRSPY: /usr/lib/x86_64-linux-gnu/libairspy.so  
-- Checking for module 'libbladeRF'
--   Found libbladeRF, version 2.2.1-git-f03d8433
-- Found libbladeRF: /usr/local/include, /usr/local/lib/libbladeRF.so
-- Checking for module 'libfreesrp'
--   No package 'libfreesrp' found
-- Found libfreesrp: /usr/include, /usr/lib/x86_64-linux-gnu/libfreesrp.so
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen missing components:  dot
CMake Error at CMakeLists.txt:179 (message):
  GnuRadio Runtime required to build gr-osmosdr

-- Configuring incomplete, errors occurred!
See also "/home/serj/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/serj/gr-osmosdr/build/CMakeFiles/CMakeError.log".
......................................................

CMakeErrorLog.txt says that-cannot find lpthread, 
collect2: error: ld returned 1 exit status, although when I run the command apt-get install libpthread-stubs0-dev, ubuntu tells me that such a library is already installed

velichkov, 2019-10-29:
Hi @serj053,

> ubuntu tells me that such a library is already installed

Which ubuntu version do you use?
What version of `gnuradio` do you have and how did you installed it - from the ubuntu's repositories, using pybombs or manually?
Where do you get `gr-osmosdr` from (there are quite a lot of forks out there) and which version are you trying to build?
Why are you trying to build gr-osmosdr from source and not installing it from the ubuntu's repository?

See also https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

serj053, 2019-10-29:
Thank you for responding to my request, ubuntu version-Ubuntu 18.04.3 LTS,
 Gnuradio version 3.9.0.0, installed manually using the page
 first - https://habr.com/ru/post/354780/,
 then tried - https://wiki.gnuradio.org/index.php/InstallingGR#Linux, 
then again tried on old , in General already four days'm beating as fish about the ice,
 already strayed with ultimately, Gnuradio runs, and here is further begin problems,
 am asking forgiveness for lyrics. Gr-osmosdr I got
 from - https://osmocom.org/projects/gr-osmosdr/wiki/GrOsmoSDR.
 Gr-osmosdr I took from the source because I did not know other options. 
I little studied exile that you have proposed in late your letters and 
found that I have not established package libosmocore-dev, can be this one of reasons, 
' ll wait your answer and build on a new . Sincerely Sergey (Sorry for my crooked English)

velichkov, 2019-10-29:
> Thank you for responding to my request, ubuntu version-Ubuntu 18.04.3 LTS,
> Gnuradio version 3.9.0.0, installed manually using the page
> first - https://habr.com/ru/post/354780/,

So you have installed the latest version from the gnuradio's master branch and this explains a lot of the errors.

> then tried - https://wiki.gnuradio.org/index.php/InstallingGR#Linux,

And then you have installed gnuradio from the apt-repository so now you have two different versions - one from source in `/usr/local/` and one from the apt repository in `/usr/`. Having two versions could be a source of many hard to fix errors and I really do not recommend this.

> then again tried on old , in General already four days'm beating as fish about the ice,
> already strayed with ultimately, Gnuradio runs, and here is further begin problems,
> am asking forgiveness for lyrics. 

> Gr-osmosdr I got from - https://osmocom.org/projects/gr-osmosdr/wiki/GrOsmoSDR.
> Gr-osmosdr I took from the source because I did not know other options.

`gr-osmosdr` from the osmocom repository is still not ported to gnuradio 3.8 and 3.9 versions and that's why you are seeing the `GnuRadio Runtime required to build gr-osmosdr` error. It looks for gnuradio, finds the newer one that is not compatible and prints the error.

> I little studied exile that you have proposed in late your letters and
> found that I have not established package libosmocore-dev, can be this one of reasons,
> ' ll wait your answer and build on a new . 

My advice is to remove everything installed from source, run `make uninstall` from the gnuradio's build directory. Then make sure there is no gnuradio leftovers in `/usr/local` directory and remove them manually.  Alternatively if this ubuntu installation is a new one and  you don't need it for something else you could try formatting and reinstalling the whole OS.

And then install everything you need from the apt repository. Install from source only when you have no other options and when you really need the latest versions.

```
sudo apt-get install gnuradio gr-osmosdr gr-gsm
```

> (Sorry for my crooked English)

Don't worry too much about your English but please spend some time to format you messages using [markdown](https://guides.github.com/features/mastering-markdown/) and use new lines (paragraphs) from time to time as right now your messages are hardly readable.

serj053, 2019-10-29:
Thank you, I begin to implement your recommendations

serj053, 2019-10-29:
I decided to upgrade to ubuntu 19.10, can I get your opinion on my choice to work with gnuradio.

velichkov, 2019-10-29:
> I decided to upgrade to ubuntu 19.10, can I get your opinion on my choice to work with gnuradio.

Install gnuradio, gr-osmosdr and everything else you need from the apt repository. The gnuradio version in Ubuntu 19.10 is 3.7.13.4-4build2 so you won't have problems with out-of-tree modules like gr-osmosdr that are still not migrated to gnuradio 3.8.

Install from source only when:
- A given software is not available from the Ubuntu repositories.
- You need some specific functionality that is only available in a newer version then the one available from the repositories.
- You plan to develop and contribute new features or bug fixes.

serj053, 2019-11-02:
I welcome you, thanks to your recommendations everything is established and works. Now I want to try to study in pyboms (sudo pybombs install gr-gsm), as your opinion can there be conflicts with the previously installed environment ? Sincerely Sergeyю

velichkov, 2019-11-02:
> I welcome you, thanks to your recommendations everything is established and works.

Cool.

> Now I want to try to study in pyboms (sudo pybombs install gr-gsm), as your opinion can there be conflicts with the previously installed environment ?

In my opinion with `pybombs` you can easily create a mess so first try using it in a virtual machine or in a docker container. Install `pybombs` using `pip3` and not `pip` in order to use `python3` and do not attempt installations in `/usr/local` prefix but rather install in your home directory - `~/gnuradio` for example.

You will need to edit `gr-iqbal.lwr`, `gr-osmosdr`, `gr-gsm.lwr` recipes to change the repository URLs and branch names as currently versions from the official repositories still do not support gnuradio 3.8. You could try using `maint-3.8` branches from my forks or search for another forks.

See also:
- https://osmocom.org/projects/gr-gsm/wiki/Installation#Compilation-and-installation-with-use-of-Pybombs
- https://wiki.gnuradio.org/index.php/InstallingGR#Using_PyBOMBS
- Issue https://github.com/ptrkrysik/gr-gsm/issues/475

thebugslayers, 2020-01-20:
> Good time of day. Wandering around the Internet in search of a solution of my question, I met your explanation of the problem associated with gnuradio, I had hope for help from a knowledgeable person, if you do not mind.. The problem is that I can not compile the block gr-osmosdr, won a mountain of errors but I can not win the lack of runtime, if you can help solve the problem please. (Translated with the help of an electronic translator because there may be inaccuracies in the meanings). Sincerely Sergey.
> 
> serj@serj:~/gr-osmosdr/build$ cmake ../ -DCMAKE_CXX_COMPILER=/usr/bin/g++-7 -DCMAKE_C_COMPILER=/usr/bin/gcc-7
> -- The CXX compiler identification is GNU 7.4.0
> -- Check for working CXX compiler: /usr/bin/g++-7
> -- Check for working CXX compiler: /usr/bin/g++-7 -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Build type not specified: defaulting to release.
> -- Found Git: /usr/bin/git (found version "2.17.1")
> -- Extracting version information from git describe...
> -- Configuring Boost C++ Libraries...
> -- Looking for pthread.h
> -- Looking for pthread.h - found
> -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
> -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
> -- Looking for pthread_create in pthreads
> -- Looking for pthread_create in pthreads - not found
> -- Looking for pthread_create in pthread
> -- Looking for pthread_create in pthread - found
> -- Found Threads: TRUE
> -- Found Boost: /usr/include (found version "1.65.1") found components: thread system chrono date_time atomic
> -- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
> -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
> -- Checking for module 'gmp'
> -- No package 'gmp' found
> -- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so
> -- Checking for module 'mpir >= 3.0'
> -- No package 'mpir' found
> -- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_INCLUDE_DIR)
> -- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so
> -- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.65.1") found components: date_time program_options filesystem system regex thread unit_test_framework chrono atomic
> -- Found VOLK: Volk::volk
> CMake Warning (dev) at /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:38 (if):
> if given arguments:
> 
> ```
> "ON"
> ```
> 
> An argument named "ON" appears in a conditional statement. Policy CMP0012
> is not set: if() recognizes numbers and boolean constants. Run "cmake
> --help-policy CMP0012" for policy details. Use the cmake_policy command to
> set the policy and suppress this warning.
> Call Stack (most recent call first):
> CMakeLists.txt:154 (find_package)
> This warning is for project developers. Use -Wno-dev to suppress it.
> 
> CMake Warning (dev) in /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:
> Policy CMP0011 is not set: Included scripts do automatic cmake_policy PUSH
> and POP. Run "cmake --help-policy CMP0011" for policy details. Use the
> cmake_policy command to set the policy and suppress this warning.
> 
> The included script
> 
> ```
> /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake
> ```
> 
> affects policy settings. CMake is implying the NO_POLICY_SCOPE option for
> compatibility, so the effects are applied to the including context.
> Call Stack (most recent call first):
> CMakeLists.txt:154 (find_package)
> This warning is for project developers. Use -Wno-dev to suppress it.
> 
> -- Checking for module 'gnuradio-iqbalance'
> -- Found gnuradio-iqbalance, version 0
> -- Found GNURADIO_IQBALANCE: /usr/lib/x86_64-linux-gnu/libgnuradio-iqbalance.so
[... 26 lines cut ...]
> -- Found LIBHACKRF: /usr/lib/x86_64-linux-gnu/libhackrf.so
> -- Checking for module 'libairspy'
> -- Found libairspy, version 1.0
> -- Found LIBAIRSPY: /usr/lib/x86_64-linux-gnu/libairspy.so
> -- Checking for module 'libbladeRF'
> -- Found libbladeRF, version 2.2.1-git-f03d8433
> -- Found libbladeRF: /usr/local/include, /usr/local/lib/libbladeRF.so
> -- Checking for module 'libfreesrp'
> -- No package 'libfreesrp' found
> -- Found libfreesrp: /usr/include, /usr/lib/x86_64-linux-gnu/libfreesrp.so
> -- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components: doxygen missing components: dot
> CMake Error at CMakeLists.txt:179 (message):
> GnuRadio Runtime required to build gr-osmosdr
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/serj/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
> See also "/home/serj/gr-osmosdr/build/CMakeFiles/CMakeError.log".
> ......................................................
> 
> CMakeErrorLog.txt says that-cannot find lpthread,
> collect2: error: ld returned 1 exit status, although when I run the command apt-get install libpthread-stubs0-dev, ubuntu tells me that such a library is already installed

I received the same error too, i need to install osmosdr for adding osmosdr sink in my gnuradio block
but unfortunately i cant do it.When i tried cmake it created a lot of errors.I'm using Ubuntu 18.4 LTS and Gnuradio 3.9.Please help me resolve my problem.
Thanks in advance:)

velichkov, 2020-01-20:
Hi @thebugslayers,

> I received the same error too, i need to install osmosdr for adding osmosdr sink in my gnuradio block
> but unfortunately i cant do it.When i tried cmake it created a lot of errors.I'm using Ubuntu 18.4 LTS and Gnuradio 3.9.Please help me resolve my problem.

Better report the exact errors you get to the gr-osmosdr project:
* https://osmocom.org/projects/gr-osmosdr/wiki
* https://lists.osmocom.org/mailman/listinfo/osmocom-sdr

thebugslayers, 2020-01-21:
Thank you @velichkov,I will .
Can i know the proper way to add osmocom sink to my gnuradio .
:)

thebugslayers, 2020-01-21:
I can not compile the block gr-osmosdr, obtained block of errors ,if you can help me solve the problem ,please.Sincerely thebugslayers

The following are the errors

-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Extracting version information from git describe...
-- Configuring Boost C++ Libraries...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gruel'
--   No package 'gruel' found
-- Could NOT find GRUEL (missing: GRUEL_LIBRARIES GRUEL_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-core'
--   No package 'gnuradio-core' found
-- Could NOT find GNURADIO_CORE (missing: GNURADIO_CORE_LIBRARIES GNURADIO_CORE_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-iqbalance'
--   Found gnuradio-iqbalance, version 0
-- Could NOT find GNURADIO_IQBALANCE (missing: GNURADIO_IQBALANCE_INCLUDE_DIRS) 
-- Checking for module 'uhd'
--   No package 'uhd' found
-- Could NOT find UHD (missing: UHD_LIBRARIES UHD_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-uhd'
--   Found gnuradio-uhd, version 3.9.0
-- gnuradio-uhd not found.
-- Could NOT find GNURADIO_UHD (missing: GNURADIO_UHD_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-fcd'
--   No package 'gnuradio-fcd' found
-- gnuradio-fcd not found.
-- Could NOT find GNURADIO_FCD (missing: GNURADIO_FCD_LIBRARIES GNURADIO_FCD_INCLUDE_DIRS) 
-- Checking for module 'gnuradio-fcdproplus'
--   Found gnuradio-fcdproplus, version 3.7.11
-- Found gnuradio-fcdproplus: /usr/include, /usr/lib/x86_64-linux-gnu/libgnuradio-fcdproplus.so
-- Found GNURADIO_FCDPP: /usr/lib/x86_64-linux-gnu/libgnuradio-fcdproplus.so  
-- Checking for module 'libosmosdr'
--   No package 'libosmosdr' found
-- libosmosdr not found.
-- Checking for module 'librtlsdr'
--   No package 'librtlsdr' found
-- librtlsdr not found.
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- libmirisdr not found.
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.5
-- Found LIBHACKRF: /usr/local/lib/libhackrf.so  
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- Could NOT find LIBAIRSPY (missing: LIBAIRSPY_LIBRARIES LIBAIRSPY_INCLUDE_DIRS) 
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Error at CMakeLists.txt:166 (message):
  Gruel required to build gr-osmosdr

-- Configuring incomplete, errors occurred!
See also "/home/shan/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/shan/gr-osmosdr/build/CMakeFiles/CMakeError.log".
 
Thanks:)

MountainLogic, 2020-02-05:
I'm installing on a new 64 bit Kali Raspberry Pi4.  I started from their "meta" SDR & Wireless packages then did a bunch of apt-gets to support source builds for gr-foo, gr-ieee802-11 and gr-ieee802-15-4.  I have not performed any other binary installs of GR from source.  The only odd GR source, besides the foo and 802x packages are a bunch of YAML files that I'm seeing in /usr/local/share/gnuradio/grc/blocks Even after updating apt I can not see gr-gsm so, I've been trying to build gr-gsm source.  And I'm getting the ubiquitous "GnuRadio Runtime required to compile to gr-gsm" along with the Gnradio_Found to false because dependency Volk could not be found.  I'm not seeing a maint 3.8 branch any more in git?
Also, I'm getting the same Gnuradio runtime error trying to compile gr-lte.

velichkov, 2020-02-05:
Hi @MountainLogic,

> And I'm getting the ubiquitous "GnuRadio Runtime required to compile to gr-gsm" along with the Gnradio_Found to false because dependency Volk could not be found
>  I'm not seeing a maint 3.8 branch any more in git?

See #475 and #480. For now you need to compile from `maint-3.8` of [my fork](https://github.com/velichkov/gr-recipes/tree/maint-3.8)

> Also, I'm getting the same Gnuradio runtime error trying to compile gr-lte.

Report this in the [gr-lte](https://github.com/kit-cel/gr-lte) project.

And please next time try to [format](https://help.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax) your messages, everything on one line is hardly readable.

ptrkrysik, 2026-11-09:
This is a gr-osmosdr/GNU Radio build problem and was resolved by using the distribution packages; it belongs to the gr-osmosdr project, not gr-gsm.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik