ptrkrysik/gr-gsm #487: CMAKE Error

aceboy2020, 2019-10-31:
Installing gr-gsm
I have done this.
sudo apt-get update && \
sudo apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    python-docutils \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python-scipy \
    python-gtk2 \
    gnuradio-dev \
    gr-osmosdr \
    libosmocore-dev

But after doing this
git clone https://git.osmocom.org/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake .. --------------------------------- After executing this

This error showed

root@kali:~/gr-gsm/build# cmake ..
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Extracting version information from git describe...
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/aarch64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gr-gsm/build/CMakeFiles/CMakeError.log".

velichkov, 2019-10-31:
Hi @aceboy2020,

> root@kali:~/gr-gsm/build# cmake ..

Based on your hostname you are using Kali Linux, is that correct?

> Found package configuration file:
>
> /usr/lib/aarch64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake
>
> but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
> be NOT FOUND. Reason [...]

In Debian Testing and Kali they have updated gnuradio to version 3.8 which is incompatible with the previous versions 3.7.*. We are working on adding support for gnuradio 3.8 in #475 but the changes are still not merged in the `master` branch. For now you need to checkout `maint-3.8` branch from [my fork](https://github.com/velichkov/gr-recipes/tree/maint-3.8)

Also please read all messages in #475, #476, #480 and #481.

> By the way I have installed via PyBombs but too much error. And I haven't continue it anymore and followed the manual installation.

Have you uninstalled everything installed by `pybombs`? Having two versions - one from the apt repository and one from pybombs installed in /usr/local, could result in various hard to fix errors, so is not something I recommend. 

P.S.
Please post any further questions to our [mailing list](https://github.com/ptrkrysik/gr-gsm#mailing-list)

aceboy2020, 2019-11-01:
I uninstalled pybombs. I dont know whats repository and i think i havent installed it. If i have installed it, how to remove the repository of pybombs?

aceboy2020, 2019-11-01:
I cant clone and cmake maint-3.8 
root@kali:~# git clone https://github.com/velichkov/gr-recipes/tree/maint-3.8
Cloning into 'maint-3.8'...
fatal: repository 'https://github.com/velichkov/gr-recipes/tree/maint-3.8/' not found

aceboy2020, 2019-11-01:
I have installed all the dependencies of gr-gsm.. just can cmake it.

aceboy2020, 2019-11-01:
how to install maint-3.8

velichkov, 2019-11-01:
Hi @aceboy2020,

> how to install maint-3.8

Have you read all messages in all the issues I gave you?
See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532195846

> I uninstalled pybombs.

I haven't asked you to uninstall pybombs but everything installed by pybombs.

> I dont know whats repository and i think i havent installed it. If i have installed it, how to remove the repository of pybombs?

Check pybombs' documentation.

aceboy2020, 2019-11-01:
Yes sir. Sorry late reply.
I have done this
rm -rf gr-gsm
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

Everything went well with no errors. But The only thing i see at Wireless Attack-SoftwareDefinedRadio is chirp and gnuradio-companion. I can't see the others. 

I dont know what #480 meant docker error. Do I have to install Docker?

aceboy2020, 2019-11-01:
I think they are installed because theres no error. But i can't see them gqrx, gr-scan, rfcat, rtl_adsb, rtl_fm and others..

aceboy2020, 2019-11-01:
GQRX is installed. Thank you so much! :)

aceboy2020, 2019-12-02:
everything is installed. the only problem is i cant launch grgsm livemon and grgsm scanner. and airprobe rtl :/ i am using kali on rpi 4.

aceboy2020, 2019-12-02:
i cant launch grgsm livemon and grgsm scanner. and airprobe rtl :/ i am using kali linux dist booted on rpi 4.

velichkov, 2019-12-02:
Have you read all messages in all the issues I gave you?

What errors do you get when trying to run grgsm_livemon from the terminal?

kiransgk1si18lvs04, 2020-01-29:
sir, at 70% it is getting strucked. I am using on raspberry platform

velichkov, 2020-01-29:
Hi @kiransgk1si18lvs04,

When it gets stuck check that there is free memory and if not try adding some swap space.

kiransgk1si18lvs04, 2020-01-30:
sorry for replying late, when I type make command it is getting stuck and please tell me how to  increase swap size.

kiransgk1si18lvs04, 2020-01-30:
I checked that while it gets stuck, it shows  available memory is 22Mi.
when I check again memory is decreasing.
 
Thank you for your time @velichkov

velichkov, 2020-01-30:
> please tell me how to increase swap size.

https://www.google.com/search?q=raspberry+pi+add+swap

kiransgk1si18lvs04, 2020-01-31:
Hi @ velichkov 
Thanks for your solution, the installation is completed.

when i type grgsm_livemon the following error comes
Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'.

I know that ,export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH this had to be done.
do you have a tutorial i can follow?
Thanks for your time

kiransgk1si18lvs04, 2020-02-03:
Thank you @velichkov 
I have installed gr-gsm. But when i type grgsm_livemon_headless --args=rtl -f 945.4M. 
I am getting some values at start few seconds and rest all are 0's. The below attachment are results
[image]

velichkov, 2020-02-05:
Hi @kiransgk1si18lvs04,

>  The below attachment are results

Please do not attach screenshots of your terminal, you can copypaste the text here.

> I have installed gr-gsm. But when i type grgsm_livemon_headless --args=rtl -f 945.4M.
> I am getting some values at start few seconds and rest all are 0's.

These are not zeros (0) but capital letter o and indicates **O**verflows/**O**verruns. Your RPi device is probably not powerful enough to process all the samples in the real time and that's why the internal buffers get full and the device starts dropping samples.

kiransgk1si18lvs04, 2020-02-06:
Thanks for replying sir @velichkov 
Is there any method to flush when O comes at output window, like flushing or any methods. And please give tutorials for that.
I will follow copypaste avoiding screenshots.

kiransgk1si18lvs04, 2020-02-06:
I have taken samples with a rate of 0.8e6 and 1.6e6. The results are less O's .

velichkov, 2020-02-06:
> Is there any method to flush when O comes at output window, like flushing or any methods. 

I don't know.

> I have taken samples with a rate of 0.8e6 and 1.6e6. The results are less O's .

Cool! You may try running `volk_profile` in your terminal to help `volk` to determine the optimal kernels, this may speed up GNU Radio blocks a bit.

Also check whether the CPU on your raspberry supports [NEON instructions](https://en.wikipedia.org/wiki/ARM_architecture#Advanced_SIMD_(NEON)) and if yes make sure that the `volk`, `gnuradio` and `gr-gsm` are compiled with NEON instructions enabled. Also read the [Compiler optimisation for RPi 3](https://groups.google.com/d/topic/gr-gsm/FcUi-vg7ZBI/discussion) discussion in our [Mailing List](https://github.com/ptrkrysik/gr-gsm#mailing-list).

kiransgk1si18lvs04, 2020-02-07:
Thank you @velichkov 
I will work on it.

evso9816, 2021-12-08:
@velichkov, Thanks to the above discussion and this link <<https://medium.com/@Dr.Signal/gr-gsm-with-gnuradio-3-8-in-ubuntu-20-04-using-limesdr-e4ef9c9377a8>>, i successfully installed gr-gsm on Ubuntu 20.04, GR 3.8.

sumerpepper45, 2022-05-08:
i have problems when i do MAKE i get this error
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/gr-gsm/python/grgsm/__init__.py", line 64, in <module>
    from .device import *
  File "/root/gr-gsm/python/grgsm/misc_utils/device.py", line 24, in <module>
    import osmosdr
  File "/usr/lib/python3/dist-packages/osmosdr/__init__.py", line 21, in <module>
    from .osmosdr_python import *
ImportError: generic_type: type "sink" referenced unknown base type "gr::hier_block2"
>>> Load Error: /root/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:97: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:864: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

tyler-1987, 2022-10-10:
> i have problems when i do MAKE i get this error During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "/root/gr-gsm/python/grgsm/**init**.py", line 64, in from .device import * File "/root/gr-gsm/python/grgsm/misc_utils/device.py", line 24, in import osmosdr File "/usr/lib/python3/dist-packages/osmosdr/**init**.py", line 21, in from .osmosdr_python import * ImportError: generic_type: type "sink" referenced unknown base type "gr::hier_block2"
> 
> > > > Load Error: /root/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
> > > > Compilation error
> > > > make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:97: apps/grgsm_livemon] Error 1
> > > > make[1]: *** [CMakeFiles/Makefile2:864: apps/CMakeFiles/pygen_apps.dir/all] Error 2
> > > > make: *** [Makefile:146: all] Error 2

i have the same problem

evso9816, 2022-10-11:
Hey Tyler,
Which Gnu radio version are you using? With 3.7(old version) the
installation goes without any issue. However, with 3.8+, it requires
additional updates.

ptrkrysik, 2026-11-09:
This report was against the GNU Radio 3.8 / pre-port code and build system. The project has since been ported to GNU Radio 3.10; please try the new release and open a new issue with full details if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.