ptrkrysik/gr-gsm #488: cmake failure

prodragster, 2019-11-09:
Hello,

Can you help me please with that : 

cmake ..
-- The CXX compiler identification is GNU 9.2.1
-- The C compiler identification is GNU 9.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.23.0.rc1") 
-- Extracting version information from git describe...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   chrono
--   date_time
--   atomic
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'volk'
--   Found volk, version 2.0
-- Found VOLK: /usr/lib/x86_64-linux-gnu/libvolk.so  
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/x86_64-linux-gnu/libcppunit.so;dl  
-- Checking for module 'libosmocore'
--   Found libosmocore, version 0.12.0
-- Found libosmocore: /usr/lib/x86_64-linux-gnu/libosmocore.so  
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 0.12.0
-- Found libosmocodec: /usr/lib/x86_64-linux-gnu/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 0.12.0
-- Found libosmocoding: /usr/lib/x86_64-linux-gnu/libosmocoding.so  
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 0.12.0
-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so  
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gr-gsm/build/CMakeFiles/CMakeError.log".

velichkov, 2019-11-09:
Hi @prodragster,

> Gnuradio could not be found because dependency Volk could not be found.

> GnuRadio Runtime required to compile gr-gsm

See https://groups.google.com/d/topic/gr-gsm/rAhicfOsqzw/discussion and in the future please post your questions to our mailing list.

velichkov, 2019-11-09:
The correct link is https://groups.google.com/d/msg/gr-gsm/rAhicfOsqzw/_nuGdEJdAQAJ 
See also #475.

prodragster, 2019-11-09:
Hi @velichkov thank you for you reply !
I've dont like it's exaplin here https://groups.google.com/d/msg/gr-gsm/rAhicfOsqzw/_nuGdEJdAQAJ and thank you, but i'm still blocked ! 
I've create a question here if you can help me please :
https://groups.google.com/forum/#!topic/gr-gsm/4axbu00y7Qo

f5gkw, 2020-12-30:
hi,
I meet the same problem, but I do not understand... 
at cmake ..  :

kali@tetra:~$ git clone https://git.osmocom.org/gr-gsm
Cloning into 'gr-gsm'...
Fetching objects: 9302, done.
kali@tetra:~$ cd gr-gsm
kali@tetra:~/gr-gsm$ mkdir build
kali@tetra:~/gr-gsm$ cd build
kali@tetra:~/gr-gsm/build$ cmake ..
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.29.2")
-- Extracting version information from git describe...
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.35") found components: filesystem system thread
--
-- Checking for module SWIG
-- Disabling SWIG because version check failed.
CMake Error at CMakeLists.txt:106 (message):
  SWIG required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/home/kali/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/kali/gr-gsm/build/CMakeFiles/CMakeError.log".
kali@tetra:~/gr-gsm/build$

I am not enough clever to know what I must do ...
Does somebody would like to try to give the hand ?

Thanks, happy new year

ptrkrysik, 2026-11-09:
Duplicate of #475. This was caused by GNU Radio 3.8 not being supported by the old code. GNU Radio 3.10 is now supported in the new release; please try it and open a new issue if the build still fails.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.