ptrkrysik/gr-gsm #489: make error

dablitz, 2019-11-14:
I have followed instruction for maint-3.8 completely. I have all nessesary packages installed and when I do a make I get this:

/source/gr-gsm/build$ make
Scanning dependencies of target grgsm
[  1%] Building CXX object lib/CMakeFiles/grgsm.dir/constants.cc.o
[  2%] Building C object lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_conv.c.o
[  3%] Building C object lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o
/home/icmt/source/gr-gsm/lib/decoding/osmocom/coding/gsm0503_coding.c:44:10: fatal error: osmocom/gsm/gsm0503.h: No such file or directory
 #include <osmocom/gsm/gsm0503.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
lib/CMakeFiles/grgsm.dir/build.make:110: recipe for target 'lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o' failed
make[2]: *** [lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o] Error 1
CMakeFiles/Makefile2:347: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed
make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

velichkov, 2019-11-14:
Hi @dablitz,

> I have followed instruction for maint-3.8 completely.

Which instructions exactly? Which Linux distro and version do you use?

> I have all nessesary packages installed and when I do a make I get this:

How did you install all the dependencies - using the distro package manager or manually?

> /source/gr-gsm/build$ make
> Scanning dependencies of target grgsm
> [ 3%] Building C object lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o
> /home/icmt/source/gr-gsm/lib/decoding/osmocom/coding/gsm0503_coding.c:44:10: fatal error: osmocom/gsm/gsm0503.h: No such file or directory

The build with the embeded libosmocore version seems to be broken, I'm usually installing libosmocore using the package manager or manually and I do not build with the included copy. Thanks for reporting this, I'm going to fix it.

My advice is to try install libosmocore using the package manager (`apt-get install libosmocore-dev` or `dnf install libosmocore-devel`) then delete the gr-gsm's build directory to clear the cmake's cache and any leftovers and start from scratch.

dablitz, 2019-11-14:
Which instructions exactly? Which Linux distro and version do you use?

I am using Ubuntu 18.04: 

                

https://github.com/velichkov/gr-gsm/blob/5c1e805a624ac2acd43c3c12fc7acb3b48b8e0c3/tests/dockerfiles/Kali.docker#L24-L46  <-- these instructions

 

I have been installing dependencies via apt install

 

 

Thankyou for looking into this.

velichkov, 2019-11-14:
> I am using Ubuntu 18.04:
>
> https://github.com/velichkov/gr-gsm/blob/5c1e805a624ac2acd43c3c12fc7acb3b48b8e0c3/tests/dockerfiles/Kali.docker#L24-L46  <-- these instructions

I'm a bit confused. In Debian Testing and Kali they have updated gnuradio to 3.8 while in Ubuntu 18.04 the gnuradio package is [still 3.7](https://packages.ubuntu.com/bionic/gnuradio)

> I have been installing dependencies via apt install

So if you are installing from the apt how how did you install gnuradio 3.8?

Have you tried installing [libosmocore-dev](https://packages.ubuntu.com/bionic/libosmocore-dev) and then cleaning the build dir?

dablitz, 2019-11-14:
I compiled 3.8 from source and used support packages for most of my dependencies.

 

And yes I have libosmocore-dev package installed

velichkov, 2019-11-14:
> I compiled 3.8 from source and used support packages for most of my dependencies.

OK, next time please try to be more precise. 

> And yes I have libosmocore-dev package installed

When did you install it - before or after running `cmake` in the gr-gsm's build directory?

If you have installed it after then it is important to clean the build directory as cmake keeps a cache (CMakeCache.txt) in it and no matter how many times you run it after that it will takes some of the values from this cache.

If you did clean the build directory and it still does not work then please create another sub-directory with a different name, run `cmake ..` in it and provide the full output.

dablitz, 2019-11-15:
When did you install it - before or after running cmake in the gr-gsm's build directory?

I had it installed before running cmake.

 

I have deleted the build directory completely and ran cmake again.

dablitz, 2019-11-15:
:~/source/gr-gsm/build$ sudo apt install libosmocore-dev

[sudo] password for icmt:

Reading package lists... Done

Building dependency tree

Reading state information... Done

libosmocore-dev is already the newest version (0.9.0-7).

The following packages were automatically installed and are no longer required:

  linux-headers-5.0.0-31 linux-headers-5.0.0-31-generic linux-image-5.0.0-31-generic linux-modules-5.0.0-31-generic linux-modules-extra-5.0.0-31-generic

Use 'sudo apt autoremove' to remove them.

0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

 

Then:

 

/source/gr-gsm/build$ cmake ../

-- The CXX compiler identification is GNU 7.4.0

-- The C compiler identification is GNU 7.4.0

-- Check for working CXX compiler: /usr/bin/c++

-- Check for working CXX compiler: /usr/bin/c++ -- works

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- Check for working C compiler: /usr/bin/cc

-- Check for working C compiler: /usr/bin/cc -- works

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Detecting C compile features

-- Detecting C compile features - done

-- Build type not specified: defaulting to release.

-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")

-- Checking for module 'gmp'

--   No package 'gmp' found

-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so

-- Checking for module 'mpir >= 3.0'

--   No package 'mpir' found

-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_INCLUDE_DIR)

-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so

-- Looking for pthread.h

-- Looking for pthread.h - found

-- Looking for pthread_create

-- Looking for pthread_create - not found

-- Looking for pthread_create in pthreads

-- Looking for pthread_create in pthreads - not found

-- Looking for pthread_create in pthread

-- Looking for pthread_create in pthread - found

-- Found Threads: TRUE

-- Boost version: 1.65.1

-- Found the following Boost libraries:

--   date_time

--   program_options

--   filesystem

--   system

--   regex

--   thread

--   unit_test_framework

--   chrono

--   atomic

-- Found VOLK: Volk::volk

-- User set python executable /usr/bin/python3

-- Found PythonInterp: /usr/bin/python3 (found version "3.6.8")

-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.8")

-- Checking for module 'fftw3f >= 3.0'

--   Found fftw3f , version 3.3.7

-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so

-- Found Git: /usr/bin/git

-- Extracting version information from git describe...

-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen missing components:  dot

-- Checking for module 'libosmocore'

--   Found libosmocore, version 0.9.0

-- Found libosmocore: /usr/lib/x86_64-linux-gnu/libosmocore.so

-- Checking for module 'libosmocodec'

--   Found libosmocodec, version 0.9.0

-- Found libosmocodec: /usr/lib/x86_64-linux-gnu/libosmocodec.so

-- Checking for module 'libosmocoding'

--   No package 'libosmocoding' found

-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR)

-- Checking for module 'libosmogsm'

--   Found libosmogsm, version 0.9.0

-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so

-- Loading build date Fri, 15 Nov 2019 15:23:51 into constants...

-- Loading version v0.41.1-342-g70cf13d6 into constants...

-- Using install prefix: /usr/local

-- Building for version: v0.41.1-342-g70cf13d6 / 1.0.0git

-- No C++ unit tests... skipping

--

-- Checking for module SWIG

-- Found SWIG version 3.0.12.

-- Found SWIG: /usr/bin/swig3.0

-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.8")

-- Configuring done

-- Generating done

-- Build files have been written to: /home/$user/source/gr-gsm/build

 

Then:

 

~/source/gr-gsm/build$ make

Scanning dependencies of target grgsm

[  1%] Building CXX object lib/CMakeFiles/grgsm.dir/constants.cc.o

[  2%] Building C object lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_conv.c.o

[  3%] Building C object lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o

/home/icmt/source/gr-gsm/lib/decoding/osmocom/coding/gsm0503_coding.c:44:10: fatal error: osmocom/gsm/gsm0503.h: No such file or directory

#include <osmocom/gsm/gsm0503.h>

          ^~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

lib/CMakeFiles/grgsm.dir/build.make:110: recipe for target 'lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o' failed

make[2]: *** [lib/CMakeFiles/grgsm.dir/decoding/osmocom/coding/gsm0503_coding.c.o] Error 1

CMakeFiles/Makefile2:347: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed

make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2

Makefile:140: recipe for target 'all' failed

make: *** [all] Error 2

 

I cannot figure this out.

velichkov, 2019-11-16:
Hi @dablitz,

> libosmocore-dev is already the newest version (0.9.0-7).

> -- Found libosmocodec: /usr/lib/x86_64-linux-gnu/libosmocodec.so
> -- Checking for module 'libosmocoding'
> --   No package 'libosmocoding' found
> -- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR)

It turns out this version is quite old and does not provide the `libosmocoding` library.

I just push few commits to the `maint-3.8` branch (2efaa49e09c8f0306fc0baf7db095beb6e83db32) that hopefully fix this problem and also added a [Dockerfile for Ubuntu 18.04](https://github.com/velichkov/gr-gsm/blob/d049963627ce6766c96c7dfbebe7ab039eb9f854/tests/dockerfiles/Ubuntu_18_04_gnuradio_ppa.docker) and the [travis tests](https://travis-ci.org/velichkov/gr-gsm/builds/612647167) were successful.

ptrkrysik, 2026-11-09:
The bundled libosmocore copy no longer exists and a fix was made on the maint-3.8 branch; the new release requires a system libosmocore (with libosmocoding). Please reopen with logs if you still see a problem.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik