ptrkrysik/gr-gsm #490: Problem with sctp_bindx

Woxazu, 2019-11-16:
I tried to install gr gsm using pybombs. I already installed gnu radio, set up prefix as prefix/default, then ran `source ./setup_env.sh` and tried `pybombs install gr-gsm`, but here what I got while installing libosmocore as the first dependence:
```
checking for library containing sctp_bindx... no
configure: error: sctp_bindx not found in searched libs
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
```
What does it mean and what should I do? If this matter, I had trouble while installing gnuradio. Pybombs not installed qwt6 (as far as I figured out it was 6.1.*) because there was mistake in .h file (qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type) and I manually installed freshest version and replaced original qwt folder with my in prefix.

velichkov, 2019-11-16:
Hi @Woxazu,

> configure: error: sctp_bindx not found in searched libs
>
> What does it mean and what should I do?

Try installing [lksctp](http://lksctp.sourceforge.net/), for Debian/Ubuntu/Kali install `libsctp-dev` and for Fedora/Centos install `lksctp-tools-devel`

> If this matter, I had trouble while installing gnuradio. Pybombs not installed qwt6 (as far as I figured out it was 6.1.*) because there was mistake in .h file (qwt_transform.h:111:5: error: [...]

Consider reporting this `pybombs` problem in the correct place - https://github.com/gnuradio/pybombs/ and if it is a problem with one of the pybombs' recipes then https://github.com/gnuradio/gr-recipes. And don't forget to report which Linux distro you are using and which version. There is nothing we could do about this here.

P.S.
In the future please post questions or problems like this to our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list)

Woxazu, 2019-11-16:
sorry for incorrect posting. I was looking some your answers and

Woxazu, 2019-11-16:
sorry for incorrect posting. 
Kali GNU/Linux Rolling 64x
Release:        2019.4
Codename:       kali-rolling
gnuradio: 3.9.0.0-git
I already installed libsctp-dev, but forgot to write it didn't change anything.  I was looking some your previous answers and if I get it right I should manually install all packages. Now I installed libosmocore, but pyBombs refuses working with it. So i'll try install other packages myself in pyBombs changed environment and close the issue if everything will be good

Woxazu, 2019-11-17:
Well, idk how, but it has installed. I just rebooted system. For now I released that everything is completely messed up with pybombs installations and also problem intersects with [incompatable versions of gr-osmocom and gnuradio.](https://github.com/gnuradio/gnuradio/issues/2838) I will uninstall everything later, but now there is deadline and it forces to get quick solution. What do you recommend? gnuRadio rollback to 3.7?

velichkov, 2019-11-18:
> For now I realized that everything is completely messed up with pybombs installations 

With pybombs that happens quite often. 

> and also problem intersects with [incompatable versions of gr-osmocom and gnuradio.](https://github.com/gnuradio/gnuradio/issues/2838)

There are some forks out there that add 3.8 support.

> I will uninstall everything later, but now there is deadline and it forces to get quick solution. What do you recommend? gnuRadio rollback to 3.7?

Yes, first clean up or reinstall the OS in a clean partition, and then install everything from the apt repository.  Debian Testing and Kali have updated gnuradio to 3.8 so you are going to have some problems. If you need something stable then consider using a different distribution - Debian stable, Ubuntu 18.04 or Fedora 30.

If you decide to stay on Kali then have a look at [Kali.docker](https://github.com/velichkov/gr-gsm/blob/maint-3.8/tests/dockerfiles/Kali.docker) and install `gr-gsm` from the `maint-3.8` branch of [my fork](https://github.com/velichkov/gr-gsm/tree/maint-3.8/)

Woxazu, 2019-11-19:
i haven't worked with docker yet. It looks like it is test that indicates what's wrong and what's not, but doesn't install anything, at least on the host system. If not, should I run `docker build` or it just help me to determine which versions of packages to install?

velichkov, 2019-11-21:
Hi @Woxazu,

> i haven't worked with docker yet. It looks like it is test that indicates what's wrong and what's not, but doesn't install anything, at least on the host system.

Yes, we are using `docker` to test `gr-gsm` in a clean environment on different Linux distribution.
 
> If not, should I run `docker build` or it just help me to determine which versions of packages to install?

You could build the container and use gr-gsm that way although it could be tricky to configure X/GUI  access and USB/network access.

> I done everything using your instructions and osmocom guide. However, i got the error at step `make`:
> 
> ```
> make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgrgsm.so.v0.41.1-346-g2efaa49e'.  Stop.

You need to install `liborc-dev` package.

https://github.com/velichkov/gr-gsm/blob/2efaa49e09c8f0306fc0baf7db095beb6e83db32/tests/dockerfiles/Kali.docker#L16
 
> I found same [error](https://github.com/ptrkrysik/gr-gsm/issues/480). trying to repeat steps...

Always search in the github issues and in the mailing list archive.

Woxazu, 2019-11-23:
YES. for now it all had installed and worked, thank you very much for your help! Now I close the issue. For seekers who also can't install everything I write commands that worked for me:
```
apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
    cmake \
    autoconf \
    libtool \
    pkg-config \
    build-essential \
    libcppunit-dev \
    swig \
    doxygen \
    liblog4cpp5-dev \
    python3-docutils \
    python3-scipy \
    gnuradio-dev gnuradio libgmp-dev libboost-regex-dev liborc-dev \
    libosmocore-dev \
    gr-osmosdr
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
cmake ..
make
sudo make install
sudo ldconfig
```
after loading OS next time don't forget to change pytnonpath again: `export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH`

closed 2019-11-23 by Woxazu