ptrkrysik/gr-gsm #491: pybombs install error

artem222414, 2019-11-21:
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
[INFO] Executing command with elevated privileges: `sudo -H apt-get -y install libqwt-qt5-dev'
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqwt-qt5-dev is already the newest version (6.1.4-1.1).
The following packages were automatically installed and are no longer required:
  android-framework-res android-libaapt android-libandroidfw libantlr-java libantlr3-runtime-java libkf5xmlgui-bin
  libsmali-java libstringtemplate-java libvpx5 libxmlunit-java libyaml-snake-java
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 493 not upgraded.
Install tree:
|
\- gr-gsm
   |
   +- gr-osmosdr
   |  |
   |  +- airspy
   |  |
   |  +- hackrf
   |  |
   |  +- uhd
   |  |
   |  +- soapysdr
   |  |
   |  +- gnuradio
   |  |  |
   |  |  +- uhd
   |  |  |
   |  |  \- qwt6
   |  |
   |  +- bladeRF
   |  |
   |  +- gr-iqbal
   |  |  |
   |  |  +- libosmo-dsp
   |  |  |
   |  |  \- gnuradio
   |  |     |
   |  |     +- uhd
   |  |     |
   |  |     \- qwt6
   |  |
   |  +- osmo-sdr
   |  |
   |  \- rtl-sdr
   |
   +- gnuradio
   |  |
   |  +- uhd
   |  |
   |  \- qwt6
   |
   \- libosmocore
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: libosmocore
[INFO] Install python-apt to speed up apt processing.
Cloning into 'libosmocore'...
[INFO] Install python-apt to speed up apt processing.
Configuring: (100%) [=======================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
Makefile.am:43: warning: DOXYGEN was already defined in condition TRUE, which includes condition !HAVE_DOXYGEN ...
configure.ac:84: ... 'DOXYGEN' previously defined here
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking whether make sets $(MAKE)... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for pkg-config... /usr/bin/pkg-config
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.20... yes
checking for ANSI C header files... (cached) yes
checking execinfo.h usability... yes
checking execinfo.h presence... yes
checking for execinfo.h... yes
checking sys/select.h usability... yes
checking sys/select.h presence... yes
checking for sys/select.h... yes
checking sys/socket.h usability... yes
checking sys/socket.h presence... yes
checking for sys/socket.h... yes
checking sys/timerfd.h usability... yes
checking sys/timerfd.h presence... yes
checking for sys/timerfd.h... yes
checking syslog.h usability... yes
checking syslog.h presence... yes
checking for syslog.h... yes
checking ctype.h usability... yes
checking ctype.h presence... yes
checking for ctype.h... yes
checking netinet/tcp.h usability... yes
checking netinet/tcp.h presence... yes
checking for netinet/tcp.h... yes
checking netinet/in.h usability... yes
checking netinet/in.h presence... yes
checking for netinet/in.h... yes
checking for size_t... yes
checking for working alloca.h... yes
checking for alloca... yes
checking for library containing dlopen... -ldl
checking for library containing dlsym... -ldl
checking for backtrace in -lexecinfo... no
checking whether gcc is Clang... no
checking whether pthreads work with -pthread... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking whether more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking for library containing clock_gettime... none required
checking for doxygen... /usr/bin/doxygen
checking whether SYS_getrandom is declared... yes
checking if gcc supports -fvisibility=hidden... yes
checking for clock_gettime... yes
checking for localtime_r... yes
checking whether struct tm has tm_gmtoff member... yes
checking whether to workaround TLS bug in old gcc on ARM platforms... 
checking for TALLOC... yes
checking for PCSC... yes
checking for LIBGNUTLS... yes
checking for library containing sctp_bindx... no
configure: error: sctp_bindx not found in searched libs
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package libosmocore:
Configuration failed
[ERROR] Error installing package libosmocore. Aborting.

velichkov, 2019-11-21:
Hi @artem222414,

Please report pybombs' related problems in https://github.com/gnuradio/pybombs/ or https://github.com/gnuradio/gr-recipes as there is nothing really we could do about them here.

> checking for library containing sctp_bindx... no
configure: error: sctp_bindx not found in searched libs

Install the `libsctp-dev` package manually, see #490

Why did you choose to install `gr-gsm` using pybombs? In my opinion this is the hardest and most error prone way. Why not installing everything from the `apt` repository? 

See https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

If you are on Debian Testing or Kali you need to install `gr-gsm` from the `maint-3.8` branch of my fork, see #480 and #475

artem222414, 2019-11-21:
Thanks for your answer, I use pybombs because the method via git also has errors...

[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: libosmocore
[INFO] Install python-apt to speed up apt processing.
Configuring: (100%) [=========================================================================================================================================================================]
Building:    (100%) [=========================================================================================================================================================================]
Installing:  (100%) [=========================================================================================================================================================================]
[INFO] Installation successful.
[INFO] Installing package: qwt6
[INFO] Install python-apt to speed up apt processing.
03469 kB / 03469 kB (100%)
Configuring: (100%) [=========================================================================================================================================================================]
Building:    (100%) [=========================================================================================================================================================================]
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /usr/local/src/qwt6/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/usr/local/src/qwt6/src'
compiling qwt_abstract_scale_draw.cpp
In file included from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qlocale.h:43,
                 from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qtextstream.h:46,
                 from ../../../../include/x86_64-linux-gnu/qt5/QtCore/qdebug.h:49,
                 from qwt_interval.h:17,
                 from qwt_scale_div.h:14,
                 from qwt_abstract_scale_draw.h:14,
                 from qwt_abstract_scale_draw.cpp:10:
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h: In constructor ‘QVariant::QVariant(QVariant&&)’:
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:275:25: warning: implicitly-declared ‘constexpr QVariant::Private& QVariant::Private::operator=(const QVariant::Private&)’ is deprecated [-Wdeprecated-copy]
  275 |     { other.d = Private(); }
      |                         ^
../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:401:16: note: because ‘QVariant::Private’ has user-provided ‘QVariant::Private::Private(const QVariant::Private&)’
  401 |         inline Private(const Private &other) Q_DECL_NOTHROW
      |                ^~~~~~~
In file included from qwt_scale_map.h:14,
                 from qwt_abstract_scale_draw.cpp:14:
qwt_transform.h: At global scope:
qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type
  110 |     QT_STATIC_CONST double LogMin;
      |     ^~~~~~~~~~~~~~~
qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type
  111 |     QT_STATIC_CONST double LogMax;
      |     ^~~~~~~~~~~~~~~
make[1]: *** [Makefile:3810: obj/qwt_abstract_scale_draw.o] Error 1
make[1]: Leaving directory '/usr/local/src/qwt6/src'
make: *** [Makefile:50: sub-src-make_first-ordered] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package qwt6:
Build failed.
[ERROR] Error installing package qwt6. Aborting.

velichkov, 2019-11-21:
> I use pybombs because the method via git also has errors...

What errors? Which gnuradio version do you use, how did you installed it and which gr-gsm branch did you tried to compile from?

> ../../../../include/x86_64-linux-gnu/qt5/QtCore/qvariant.h:401:16: note: because ‘QVariant::Private’ has user-provided ‘QVariant::Private::Private(const QVariant::Private&)’
401 | inline Private(const Private &other) Q_DECL_NOTHROW
| ^~~~~~~
In file included from qwt_scale_map.h:14,
from qwt_abstract_scale_draw.cpp:14:
qwt_transform.h: At global scope:
qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type
110 | QT_STATIC_CONST double LogMin;

As I've said report those errors to `pybombs` or `gr-recipes` projects. Don't forget to search in their issues first to check whether this problem has been already reported and if not open a new issue there and don't forget to report which Linux distribution you are using, which version and all the commands you have executed so far.

artem222414, 2019-11-21:
if i use git 

cat CMakeError.log 
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_18f9d/fast"
/usr/bin/make -f CMakeFiles/cmTC_18f9d.dir/build.make CMakeFiles/cmTC_18f9d.dir/build
make[1]: Entering directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_18f9d.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_18f9d.dir/CheckSymbolExists.c.o   -c /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_18f9d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_18f9d.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_18f9d.dir/CheckSymbolExists.c.o  -o cmTC_18f9d 
/usr/bin/ld: CMakeFiles/cmTC_18f9d.dir/CheckSymbolExists.c.o: in function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_18f9d.dir/build.make:87: cmTC_18f9d] Error 1
make[1]: Leaving directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_18f9d/fast] Error 2

File /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_7a851/fast"
/usr/bin/make -f CMakeFiles/cmTC_7a851.dir/build.make CMakeFiles/cmTC_7a851.dir/build
make[1]: Entering directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_7a851.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_7a851.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.13/Modules/CheckFunctionExists.c
Linking C executable cmTC_7a851
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_7a851.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_7a851.dir/CheckFunctionExists.c.o  -o cmTC_7a851 -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_7a851.dir/build.make:87: cmTC_7a851] Error 1
make[1]: Leaving directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_7a851/fast] Error 2

cat CMakeOutput.log 
The system is: Linux - 5.2.0-2parrot1-amd64 - x86_64
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/zimplex/gr-gsm/build/CMakeFiles/3.13.4/CompilerIdCXX/a.out"

Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
[... 660 lines cut ...]
Determining if the include file pthread.h exists passed with the following output:
Change Dir: /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_d7051/fast"
/usr/bin/make -f CMakeFiles/cmTC_d7051.dir/build.make CMakeFiles/cmTC_d7051.dir/build
make[1]: Entering directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_d7051.dir/CheckIncludeFile.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_d7051.dir/CheckIncludeFile.c.o   -c /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
Linking C executable cmTC_d7051
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d7051.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_d7051.dir/CheckIncludeFile.c.o  -o cmTC_d7051 
make[1]: Leaving directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'

Determining if the function pthread_create exists in the pthread passed with the following output:
Change Dir: /home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_fd5ba/fast"
/usr/bin/make -f CMakeFiles/cmTC_fd5ba.dir/build.make CMakeFiles/cmTC_fd5ba.dir/build
make[1]: Entering directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_fd5ba.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_fd5ba.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.13/Modules/CheckFunctionExists.c
Linking C executable cmTC_fd5ba
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fd5ba.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_fd5ba.dir/CheckFunctionExists.c.o  -o cmTC_fd5ba -lpthread 
make[1]: Leaving directory '/home/zimplex/gr-gsm/build/CMakeFiles/CMakeTmp'

velichkov, 2019-11-21:
> cat CMakeError.log

Don't look at the `CMakeError.log`, most of the errors there are normal, it's just how cmake works. 

And please use [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages as pasting long logs just that makes them mostly unreadable. You need to enclose the logs in triple back quotes (```) and if you are going to paste more then lets say 20 lines consider uploading the whole file as attachment instead.

artem222414, 2019-11-21:
thanks for your answer . and I'm sorry for my ignorance . I'm too young to work with gsm ... 
if i ignore erors :

artem222414, 2019-11-21:
┌─[✗]─[root@parrot]─[/home/zimplex/gr-gsm/build]
└──╼ #mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
mkdir: cannot create directory ‘/root/.grc_gnuradio/’: File exists
mkdir: cannot create directory ‘/root/.gnuradio/’: File exists
┌─[✗]─[root@parrot]─[/home/zimplex/gr-gsm/build]
└──╼ #make
make: *** No targets specified and no makefile found.  Stop.

velichkov, 2019-11-21:
> mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
> mkdir: cannot create directory ‘/root/.grc_gnuradio/’: File exists
> mkdir: cannot create directory ‘/root/.gnuradio/’: File exists

No need to create these directories multiple times. Try to understand what a command does before executing it. Don't execute commands blindly. 

>└──╼ #make
make: *** No targets specified and no makefile found. Stop.

Create a `build` subdirectory first (if you haven't created it yet) and then from it you need to run `cmake ..` and then from the `build` directory run `make`

Also please read again all my messages and try to respond to all my questions.

artem222414, 2019-11-21:
I'm crate build , and run make . I saw 
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm
-- Configuring incomplete, errors occurred!
See also "/home/zimplex/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/zimplex/gr-gsm/build/CMakeFiles/CMakeError.log".

velichkov, 2019-11-21:
> GnuRadio Runtime required to compile gr-gsm

Thats because you have gnuradio 3.8 and `gr-gsm` from the `master` branch is still not ported to support `3.8`. You need to get the `maint-3.8` branch from my fork, it's all explained in #480 and #475

```
rm -rf gr-gsm
sudo apt-get install liblog4cpp5-dev gnuradio-dev gnuradio libgmp-dev libboost-regex-dev liborc-dev libosmocore-dev gr-osmosdr
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```

velichkov, 2019-11-21:
Also you probably need to clean up the mess created by pybombs first, which is not a trivial task because you have used `/usr/local` for your prefix. If you could try reinstalling your `parrot` in a clean partition.

artem222414, 2019-11-21:
Installation was successful, but when I write the grgsm_livemon command
  I see
Traceback (most recent call last):
   File "/ usr / local / bin / grgsm_livemon", line 37, in <module>
     from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'
how to fix it ?

velichkov, 2019-11-22:
> from grgsm import arfcn
> ModuleNotFoundError: No module named 'grgsm'

Have you tried searching the issues and the mailing list archives?
Have you read all the messages in the issues I already gave you?

The solution to your latest problem is described in https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

> You need to add `/usr/local/lib/python3/dist-packages/` to `PYTHONPATH`, in your shell execute
>
> ```
> export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH
> ```
> or to make it permanent add this to your shell profile (`~/.bashrc` or `~/.bash_profile`)

artem222414, 2019-11-24:
THANKS . WORK :heart:

closed 2019-11-24 by artem222414