ptrkrysik/gr-gsm #492: Hola a todos, alguien podria ayudarme ?

earelylopez, 2019-11-23:
I was using this comand , but i don't know what is my wrong ? can anybody help me?

 $ sudo apt-get update
$ sudo apt-get install git python-pip
$ sudo pip install PyBOMBS
$ sudo pybombs prefix init /usr/local -a default_prx
$ sudo pybombs config default_prefix default_prx
$ sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
$ sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
$ sudo pybombs install gr-gsm
$ sudo ldconfig

checking for library containing sctp_bindx... no
configure: error: sctp_bindx not found in searched libs
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package libosmocore:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package libosmocore. Aborting.

velichkov, 2019-11-23:
Please report pybombs' problems in https://github.com/gnuradio/pybombs/ or https://github.com/gnuradio/gr-recipes as there is nothing really we could do about them here.

See also #491, #480 and #475

ptrkrysik, 2026-11-09:
PyBOMBS build issues are outside this project (and missing libsctp-dev is the likely cause); the new release is built with plain CMake, please follow the README.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik