ptrkrysik/gr-gsm #493: Error 2 | Raspbian | RPi 3B+

catalinalb, 2019-11-29:
pi@rpi102:~/kalibrate-rtl $ sudo ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether ln -s works... yes
checking for ranlib... ranlib
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking sys/time.h usability... yes
checking sys/time.h presence... yes
checking for sys/time.h... yes
checking for unistd.h... (cached) yes
checking for stdbool.h that conforms to C99... yes
checking for _Bool... yes
checking for inline... inline
checking for working strtod... yes
checking for floor... no
checking for getpagesize... yes
checking for memset... yes
checking for sqrt... no
checking for strtoul... yes
checking for strtol... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for FFTW3... yes
checking for LIBRTLSDR... yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
pi@rpi102:~/kalibrate-rtl $ sudo make
make  all-recursive
make[1]: Entering directory '/home/pi/kalibrate-rtl'
Making all in src
make[2]: Entering directory '/home/pi/kalibrate-rtl/src'
g++  -I/ -g -O2   -o kal kal-arfcn_freq.o kal-c0_detect.o kal-circular_buffer.o kal-fcch_detector.o kal-kal.o kal-offset.o kal-usrp_source.o kal-util.o -lfftw3 -L -lrtlsdr -lrt
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::~usrp_source()':
/home/pi/kalibrate-rtl/src/usrp_source.cc:81: undefined reference to `rtlsdr_close'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::tune(double)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:127: undefined reference to `rtlsdr_set_center_freq'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::set_freq_correction(int)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:143: undefined reference to `rtlsdr_set_freq_correction'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::set_gain(float)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:155: undefined reference to `rtlsdr_set_tuner_gain_mode'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:160: undefined reference to `rtlsdr_set_tuner_gain'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::open(unsigned int)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:176: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:184: undefined reference to `rtlsdr_get_device_name'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:187: undefined reference to `rtlsdr_get_device_name'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:191: undefined reference to `rtlsdr_open'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:198: undefined reference to `rtlsdr_set_sample_rate'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:203: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:203: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::fill(unsigned int, unsigned int*)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:230: undefined reference to `rtlsdr_read_sync'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:401: kal] Error 1
make[2]: Leaving directory '/home/pi/kalibrate-rtl/src'
make[1]: *** [Makefile:376: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/kalibrate-rtl'
make: *** [Makefile:317: all] Error 2

I can only use sudo...

velichkov, 2019-11-30:
Hi @catalinalb,

> make[2]: Entering directory '/home/pi/kalibrate-rtl/src'

Better report this problem in the [kalibrate-rtl](https://github.com/steve-m/kalibrate-rtl) project, see https://github.com/steve-m/kalibrate-rtl/issues/19

> g++ -I/ -g -O2 -o kal kal-arfcn_freq.o kal-c0_detect.o kal-circular_buffer.o kal-fcch_detector.o kal-kal.o kal-offset.o kal-usrp_source.o kal-util.o -lfftw3 -L -lrtlsdr -lrt
> /usr/bin/ld: [...]

Which Linux distribution do you use on the RPi and which version?
How did you install the `rtlsdr` library and which version? 

> I can only use sudo...

What do you mean by this?

catalinalb, 2019-12-01:
Hello and thank you for the reply.
I use the latest version of raspbian (sudo apt update, upgrade and dist-upgrade) done before starting the installation. 

"I can only use sudo" means that if I do not type sudo in front of ./bootstrap or ./configure ore even make command has no permissions to do anything. This is a bit strange for me.... 

I think the rtlsdr library was not installed at that time but I installed it with synaptic is that ok?

Thank you,
Catalin

velichkov, 2019-12-01:
> "I can only use sudo" means that if I do not type sudo in front of ./bootstrap or ./configure ore even make command has no permissions to do anything. This is a bit strange for me....

Strange, you should definitely be able to run `./cofigure` or `make` without `sudo`. What error do you get when not using `sudo`?

> I think the rtlsdr library was not installed at that time but I installed it with synaptic is that ok?

Yes, install the `librtlsdr-dev` package.

zodoczi, 2019-12-10:
same here :( Latest rasbian on RPi 3:
`pi@raspberrypi:~/kalibrate-rtl $ sudo make
make  all-recursive
make[1]: Entering directory '/home/pi/kalibrate-rtl'
Making all in src
make[2]: Entering directory '/home/pi/kalibrate-rtl/src'
g++  -I/ -g -O2   -o kal kal-arfcn_freq.o kal-c0_detect.o kal-circular_buffer.o kal-fcch_detector.o kal-kal.o kal-offset.o kal-usrp_source.o kal-util.o -lfftw3 -L -lrtlsdr -lrt 
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::~usrp_source()':
/home/pi/kalibrate-rtl/src/usrp_source.cc:81: undefined reference to `rtlsdr_close'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::tune(double)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:127: undefined reference to `rtlsdr_set_center_freq'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::set_freq_correction(int)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:143: undefined reference to `rtlsdr_set_freq_correction'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::set_gain(float)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:155: undefined reference to `rtlsdr_set_tuner_gain_mode'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:160: undefined reference to `rtlsdr_set_tuner_gain'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::open(unsigned int)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:176: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:184: undefined reference to `rtlsdr_get_device_name'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:187: undefined reference to `rtlsdr_get_device_name'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:191: undefined reference to `rtlsdr_open'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:198: undefined reference to `rtlsdr_set_sample_rate'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:203: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: /home/pi/kalibrate-rtl/src/usrp_source.cc:203: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: kal-usrp_source.o: in function `usrp_source::fill(unsigned int, unsigned int*)':
/home/pi/kalibrate-rtl/src/usrp_source.cc:230: undefined reference to `rtlsdr_read_sync'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:401: kal] Error 1
make[2]: Leaving directory '/home/pi/kalibrate-rtl/src'
make[1]: *** [Makefile:376: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/kalibrate-rtl'
make: *** [Makefile:317: all] Error 2
`

velichkov, 2019-12-10:
Hi @zodoczi 

> same here :( Latest rasbian on RPi 3:

See https://github.com/ptrkrysik/gr-gsm/issues/493#issuecomment-559968047 and https://github.com/ptrkrysik/gr-gsm/issues/493#issuecomment-560098361

zodoczi, 2019-12-10:
thanks @velichkov meanwhile solved my problem... Might help others: I had to download rtl-sdr repo from osmocom and build that (a few functions were missing). Now running fine :) 
As in this guide: https://karelkeers.nl/karelkeers.nl/rtl-sdr-calibration-with-r820t2-chip/

Thanks

catalinalb, 2020-02-07:
So I managed to install it and it's running but it isn't able to find anything.. 

kallibrate is running without any problems. grgsm... no :(

ShitalBorganve, 2021-06-11:
@zodoczi the link https://karelkeers.nl/karelkeers.nl/rtl-sdr-calibration-with-r820t2-chip/ seems to be not working.
I floowed below steps to build and install rtl-sdr

git clone git://git.osmocom.org/rtl-sdr.git
cd rtl-sdr/
mkdir build
cd build
cmake ../
make
sudo make install
sudo ldconfig
After this the above issue solved

ptrkrysik, 2026-11-09:
This is a kalibrate-rtl / librtlsdr build problem, not a gr-gsm one; the solution in the thread was installing a recent librtlsdr. Please report kalibrate problems to its own repository.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik