ptrkrysik/gr-gsm #494: Make error

syn-ua, 2019-12-11:
Used

rm -rf gr-gsm
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

Tools version: 

cmake version 3.16.0
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0

cmake:

-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.65.1") found components: date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components: doxygen missing components: dot
-- Checking for module 'libosmocore'
--   Found libosmocore, version 0.9.0
-- Found libosmocore: /usr/lib/x86_64-linux-gnu/libosmocore.so  
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 0.9.0
-- Found libosmocodec: /usr/lib/x86_64-linux-gnu/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 0.9.0
-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so  
-- Loading build date Wed, 11 Dec 2019 14:36:32 into constants...
-- Loading version v0.41.1-346-g2efaa49e into constants...
-- Using install prefix: /usr/local
-- Building for version: v0.41.1-346-g2efaa49e / 1.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/srx315/Desktop/3.8/test/gr-gsm/build

make
....
[ 89%] Building CXX object swig/CMakeFiles/grgsm_swig.dir/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_sink_get_sub_types(PyObject*, PyObject*)’:
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:129401:24: error: ‘class gr::gsm::burst_sink’ has no member named ‘get_sub_types’; did you mean ‘get_bursts’?
       result = (arg1)->get_sub_types();
                        ^~~~~~~~~~~~~
                        get_bursts
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_sink_get_sub_slots(PyObject*, PyObject*)’:
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:129434:24: error: ‘class gr::gsm::burst_sink’ has no member named ‘get_sub_slots’; did you mean ‘get_timeslots’?
       result = (arg1)->get_sub_slots();
                        ^~~~~~~~~~~~~
                        get_timeslots
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_sink_sptr_get_sub_types(PyObject*, PyObject*)’:
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:129827:25: error: ‘class gr::gsm::burst_sink’ has no member named ‘get_sub_types’; did you mean ‘get_bursts’?
       result = (*arg1)->get_sub_types();
                         ^~~~~~~~~~~~~
                         get_bursts
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_sink_sptr_get_sub_slots(PyObject*, PyObject*)’:
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:129860:25: error: ‘class gr::gsm::burst_sink’ has no member named ‘get_sub_slots’; did you mean ‘get_timeslots’?
       result = (*arg1)->get_sub_slots();
                         ^~~~~~~~~~~~~
                         get_timeslots
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_source_set_arfcn(PyObject*, PyObject*, PyObject*)’:
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:133219:15: error: ‘class gr::gsm::burst_source’ has no member named ‘set_arfcn’
       (arg1)->set_arfcn(arg2);
               ^~~~~~~~~
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_source_sptr_set_arfcn(PyObject*, PyObject*, PyObject*)’:
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:133695:16: error: ‘class gr::gsm::burst_source’ has no member named ‘set_arfcn’
       (*arg1)->set_arfcn(arg2);
                ^~~~~~~~~
swig/CMakeFiles/grgsm_swig.dir/build.make:62: recipe for target 'swig/CMakeFiles/grgsm_swig.dir/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [swig/CMakeFiles/grgsm_swig.dir/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:1192: recipe for target 'swig/CMakeFiles/grgsm_swig.dir/all' failed
make[1]: *** [swig/CMakeFiles/grgsm_swig.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

velichkov, 2019-12-11:
Hi @syn-ua,

> -- Build files have been written to: /home/user/Desktop/3.8/test/gr-gsm/build
> 
> make
> ....
> [ 89%] Building CXX object swig/CMakeFiles/grgsm_swig.dir/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx.o
> /home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_burst_sink_get_sub_types(PyObject*, PyObject*)’:
> /home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:129401:24: error: ‘class gr::gsm::burst_sink’ has no member named ‘get_sub_types’; did you mean ‘get_bursts’?

You probably have attempted a build from the master branch in the same build folder and there are some leftovers from the previous build, `make clean` and `make` should solve this problem. Or delete `swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx` to force make to regenerate it. 

Please in the future use [markdown](https://guides.github.com/features/mastering-markdown/) to format your messages and enclose logs in a triple backticks (```)

syn-ua, 2019-12-11:
Hi @velichkov . Thanks for the quick response.
I tried to``` make clean ``` and then ```make``` . It did not help.
I deleted the gr-gsm folder and try it again
```
rm -rf gr-gsm
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```
Also did not help.

velichkov, 2019-12-11:
> I tried tomake clean and then make . It did not help.
> I deleted the gr-gsm folder and try it again

> Also did not help.

Strange! Do you have another version of gr-gsm installed at the same time - manually (in `/usr/local/` or with apt-get (in `/usr`) ? If so try removing it and then `make clean`.

Also provide the full output of
```
find /usr -name burst_sink.h 2>/dev/null
grep -R get_sub_slots /home/user/Desktop/3.8/test/gr-gsm 
cd /home/user/Desktop/3.8/test/gr-gsm 
git status
git branch
git describe
```

syn-ua, 2019-12-11:
@velichkov 
```user@user-pc:~/Desktop/3.8/555/gr-gsm/build$ find /usr -name burst_sink.h 2>/dev/null
/usr/local/include/grgsm/qa_utils/burst_sink.h
user@user-pc:~/Desktop/3.8/555/gr-gsm/build$ grep -R get_sub_slots /home/user/Desktop/3.8/test/gr-gsm 

grep: warning: /home/user/Desktop/3.8/test/gr-gsm/python/grgsm: recursive directory loop
/home/user/Desktop/3.8/test/gr-gsm/build/swig/grgsm_swig.py:    def get_sub_slots(self) -> "std::vector< uint8_t,std::allocator< uint8_t > >":
/home/user/Desktop/3.8/test/gr-gsm/build/swig/grgsm_swig.py:        """get_sub_slots(burst_sink self) -> std::vector< uint8_t,std::allocator< uint8_t > >"""
/home/user/Desktop/3.8/test/gr-gsm/build/swig/grgsm_swig.py:        return _grgsm_swig.burst_sink_get_sub_slots(self)
/home/user/Desktop/3.8/test/gr-gsm/build/swig/grgsm_swig.py:    def get_sub_slots(self) -> "std::vector< uint8_t,std::allocator< uint8_t > >":
/home/user/Desktop/3.8/test/gr-gsm/build/swig/grgsm_swig.py:        """get_sub_slots(burst_sink_sptr self) -> std::vector< uint8_t,std::allocator< uint8_t > >"""
/home/user/Desktop/3.8/test/gr-gsm/build/swig/grgsm_swig.py:        return _grgsm_swig.burst_sink_sptr_get_sub_slots(self)
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:SWIGINTERN PyObject *_wrap_burst_sink_get_sub_slots(PyObject *SWIGUNUSEDPARM(self), PyObject *args) {
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:  if(!PyArg_UnpackTuple(args,(char *)"burst_sink_get_sub_slots",1,1,&obj0)) SWIG_fail;
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:    SWIG_exception_fail(SWIG_ArgError(res1), "in method '" "burst_sink_get_sub_slots" "', argument " "1"" of type '" "gr::gsm::burst_sink *""'"); 
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:      result = (arg1)->get_sub_slots();
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:SWIGINTERN PyObject *_wrap_burst_sink_sptr_get_sub_slots(PyObject *SWIGUNUSEDPARM(self), PyObject *args) {
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:  if(!PyArg_UnpackTuple(args,(char *)"burst_sink_sptr_get_sub_slots",1,1,&obj0)) SWIG_fail;
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:    SWIG_exception_fail(SWIG_ArgError(res1), "in method '" "burst_sink_sptr_get_sub_slots" "', argument " "1"" of type '" "boost::shared_ptr< gr::gsm::burst_sink > *""'"); 
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:      result = (*arg1)->get_sub_slots();
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:	 { (char *)"burst_sink_get_sub_slots", _wrap_burst_sink_get_sub_slots, METH_VARARGS, (char *)"burst_sink_get_sub_slots(burst_sink self) -> std::vector< uint8_t,std::allocator< uint8_t > >"},
/home/user/Desktop/3.8/test/gr-gsm/build/swig/CMakeFiles/grgsm_swig.dir/grgsm_swigPYTHON_wrap.cxx:	 { (char *)"burst_sink_sptr_get_sub_slots", _wrap_burst_sink_sptr_get_sub_slots, METH_VARARGS, (char *)"burst_sink_sptr_get_sub_slots(burst_sink_sptr self) -> std::vector< uint8_t,std::allocator< uint8_t > >"},
user@user-pc:~/Desktop/3.8/555/gr-gsm/build$ cd /home/user/Desktop/3.8/test/gr-gsm 
user@user-pc:~/Desktop/3.8/test/gr-gsm$ git status
On branch maint-3.8
Your branch is up to date with 'origin/maint-3.8'.

nothing to commit, working tree clean
user@user-pc:~/Desktop/3.8/test/gr-gsm$ git branch
* maint-3.8
user@user-pc:~/Desktop/3.8/test/gr-gsm$ git describe
v0.41.1-346-g2efaa49
user@user-pc:~/Desktop/3.8/test/gr-gsm$     ```

velichkov, 2019-12-11:
> $ find /usr -name burst_sink.h 2>/dev/null
> /usr/local/include/grgsm/qa_utils/burst_sink.h

Can you try uninstalling this version and then a clean rebuild?

syn-ua, 2019-12-11:
@velichkov 
Thank you so much, after the removal of the directory everything was built.

closed 2019-12-11 by syn-ua

velichkov, 2019-12-11:
> after the removal of the directory everything was built.

Make sure you uninstalled it correctly (e.g. `make uninstall` executed from where you've previously built it) and not just `rm -rvf /usr/local/include/grgsm/` as there are other files (libgrgsm.so and varous *.xml) that could cause various other problems later.

The cmake build scripts have to be improved to add the local include directories before the global ones (`/usr/include` and `/usr/local/include`) to prevent conflicts like this one.