ptrkrysik/gr-gsm #495: gr-gsm cmake failed in arch linux

Gictorbit, 2019-12-16:
I'm trying to install gr-gsm on arch linux  but required package error ocurred
here is my result cmake

i installed these required package befor cmake but it doesn't work

```
-- Found Git: /usr/bin/git (found version "2.24.0") 
-- Extracting version information from git describe...
-- Found Boost: /usr/lib64/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.35") found components: filesystem system thread 
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found SWIG: /usr/bin/swig  
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Checking for module 'gmp'
--   Package 'gmp', required by 'virtual:world', not found
-- Found GMP: /usr/lib/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
-- Found MPIR: /usr/lib/libmpirxx.so  
-- Found MPLIB: /usr/lib/libgmpxx.so  
-- Found Boost: /usr/lib64/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Checking for module 'volk'
--   Found volk, version 2.0
-- Found VOLK: /lib/libvolk.so  
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
-- Found CPPUNIT: /usr/lib/libcppunit.so;dl  
-- Checking for module 'libosmocore'
--   Found libosmocore, version 1.0.1
-- Found libosmocore: /usr/lib/libosmocore.so  
-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 1.0.1
-- Found libosmocodec: /usr/lib/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 1.0.1
-- Found libosmocoding: /usr/lib/libosmocoding.so  
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 1.0.1
-- Found libosmogsm: /usr/lib/libosmogsm.so  
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
```

velichkov, 2019-12-16:
Hi @Gictorbit,

> CMake Warning at CMakeLists.txt:135 (find_package):
>  Found package configuration file:
>
>    /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake
>
>  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" [...]

>   GnuRadio Runtime required to compile gr-gsm

We are still porting `gr-gsm` to `gnuradio-3.8` in #475. For now you need to compile from the `maint-3.8` branch of [my fork](https://github.com/velichkov/gr-gsm/tree/maint-3.8)

Gictorbit, 2019-12-17:
Hi, thanks for your reply.
I compiled your branch and it was successful
but when I run the `gr-gsm` commands like `grgsm_livemon`, another error occurs. here are my errors:
[image]

<pre>
/usr/lib/python3.8/site-packages/gnuradio/qtgui/range.py:41: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  precision = len(temp) if temp is not '0' else 0
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libgrgsm.so.1.0.0git: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
  File "/usr/lib/python3.8/site-packages/grgsm/__init__.py", line 48, in <module>
    from .grgsm_swig import *
  File "/usr/lib/python3.8/site-packages/grgsm/grgsm_swig.py", line 13, in <module>
    from . import _grgsm_swig
ImportError: libgrgsm.so.1.0.0git: cannot open shared object file: No such file or directory

</pre>

velichkov, 2019-12-17:
> ImportError: libgrgsm.so.1.0.0git: cannot open shared object file: No such file or directory

You need to run `sudo ldconfig`, see also https://osmocom.org/projects/gr-gsm/wiki/Installation

Gictorbit, 2019-12-17:
I followed the instructions in there exactly and I ran `ldconfig` command in the latest step with root access and have no idea what's wrong. it looks like that the python script can't find something

velichkov, 2019-12-17:
> I followed the instructions in there exactly and I ran ldconfig command in the latest step with root access and have no idea what's wrong. it looks like that the python script can't find something

OK. Give me the full output of `find /usr/ -name libgrgsm.so.1.0.0git`

Gictorbit, 2019-12-17:
the output is:
```
/usr/local/lib/libgrgsm.so.1.0.0git
```
[image]

velichkov, 2019-12-17:
There is no need to attach screenshots, it's enough to copy-paste the text.

You probably don't have `/usr/local/lib/` in your ldconfig directories. I'm not an Arch user and I'm not sure which is the Arch way of doing this but you could try the following.

```
sudo sh -c "echo '/usr/local/lib/' >> /etc/ld.so.conf.d/usr_local_lib.conf"
sudo ldconfig
```
or
```
export LD_LIBRARY_PATH=/usr/local/lib/:$LD_LIBRARY_PATH
grgsm_livemon
```

Gictorbit, 2019-12-17:
> sudo sh -c "echo '/usr/local/lib/' >> /etc/ld.so.conf.d/usr_local_lib.conf"
> sudo ldconfig

I added this path to my `user_local_lib.conf` and after running `ldconfig` command the problem solved
thank you so much

closed 2019-12-17 by Gictorbit

JQ555888, 2020-06-14:
Sorry for gravedigging this issue but for future Arch users who also face this problem (GnuRadio Runtime required error when it's actually installed) I can confirm this fixed it.

I'm running this on Manjaro ARM (based on Arch) off a Raspberry Pi 4B.