ptrkrysik/gr-gsm #497: Make "cannot find -lvolk"

adrihulias, 2019-12-24:
Used:
```
rm -rf gr-gsm
git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```

cmake ..
``` 
-- Build type not specified: defaulting to release.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Extracting version information from git describe...
-- Checking for module 'libosmocore'
--   No package 'libosmocore' found
-- Could NOT find libosmocore (missing: LIBOSMOCORE_LIBRARY LIBOSMOCORE_INCLUDE_DIR) 
-- Checking for module 'libosmocodec'
--   No package 'libosmocodec' found
-- Could NOT find libosmocodec (missing: LIBOSMOCODEC_LIBRARY LIBOSMOCODEC_INCLUDE_DIR) 
-- Checking for module 'libosmocoding'
--   No package 'libosmocoding' found
-- Could NOT find libosmocoding (missing: LIBOSMOCODING_LIBRARY LIBOSMOCODING_INCLUDE_DIR) 
-- Checking for module 'libosmogsm'
--   No package 'libosmogsm' found
-- Could NOT find libosmogsm (missing: LIBOSMOGSM_LIBRARY LIBOSMOGSM_INCLUDE_DIR) 
-- Loading build date Tue, 24 Dec 2019 22:21:36 into constants...
-- Loading version v0.41.1-346-g2efaa49e into constants...
-- Using install prefix: /opt/gnuradio-3.8
-- Building for version: v0.41.1-346-g2efaa49e / 1.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/soft/3.8/new/gr-gsm/build
```

The first start 
make -j7
```
[ 90%] Building CXX object lib/CMakeFiles/grgsm.dir/trx/trx_burst_if_impl.cc.o
[ 91%] Linking CXX shared library libgrgsm.so
/usr/bin/ld: cannot find -lvolk
collect2: error: ld returned 1 exit status
lib/CMakeFiles/grgsm.dir/build.make:1272: recipe for target 'lib/libgrgsm.so.v0.41.1-346-g2efaa49e' failed
make[2]: *** [lib/libgrgsm.so.v0.41.1-346-g2efaa49e] Error 1
CMakeFiles/Makefile2:1038: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed
make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

retry make -j7
```
[ 20%] Linking CXX shared library libgrgsm.so
/usr/bin/ld: cannot find -lvolk
collect2: error: ld returned 1 exit status
lib/CMakeFiles/grgsm.dir/build.make:1272: recipe for target 'lib/libgrgsm.so.v0.41.1-346-g2efaa49e' failed
make[2]: *** [lib/libgrgsm.so.v0.41.1-346-g2efaa49e] Error 1
CMakeFiles/Makefile2:1038: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed
make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

velichkov, 2019-12-27:
Hi @adrihulias,

> git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git

> [ 91%] Linking CXX shared library libgrgsm.so
/usr/bin/ld: cannot find -lvolk

Which Linux distribution do you use and which version?
How did you install gnuradio and which version?

Try installing [libvolk2-dev](https://packages.debian.org/unstable/libvolk2-dev) package.

AllisonOge, 2023-02-02:
> Try installing [libvolk2-dev](https://packages.debian.org/unstable/libvolk2-dev) package.

I had the same linker error and installed libvolk2-dev in Debian 11

ptrkrysik, 2026-11-09:
Missing Volk development library on the build system; installing libvolk-dev (libvolk2-dev) resolves it.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik