ptrkrysik/gr-gsm #498: Package has no install method: python-click-plugins

vinvin011235, 2019-12-28:
So hello there! I'm only 16 and I'm working my way to improve my skills and when I heard about IMSI-catcher i tries installing it then it was shown that please install gr-gsm.
Then i tried to install gr-gsm by pybombs and this error pops up.

root@Retr0:~# pybombs install gr-gsm
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
[ERROR] Package has no install method: python-click-plugins
root@Retr0:~# 

I don't know what to do. Please help.

Bogdan1999, 2019-12-28:
Hi, I also encountered this problem, please help

velichkov, 2020-01-03:
Hi @vinvin011235, @Bogdan1999,

> [ERROR] Package has no install method: python-click-plugins

It seems in Debian Testing and Kali they have started removing some python2 packages as python2 is no longer maintained, see https://pythonclock.org/

So my advice is reinstall `pybombs` with `pip3` in order to use python3, see https://github.com/gnuradio/pybombs/#using-python3
```
sudo pip uninstall pybombs
sudo apt install python3-pip
sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
```

You will also need to install `gr-gsm` from the `maint-3.8` branch of [my fork](https://github.com/velichkov/gr-gsm/tree/maint-3.8) as the `master` branch has not yet been ported to `gnuradio 3.8`.

In my opinion the pybombs installation method is the hardest and most error prone so my other advice is to install `gnuradio` and `gr-osmosdr` from the apt repository and then build and install `gr-gsm` from source. https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition

See also #475 and #480.

saintofinternet, 2020-01-06:
> Hi @vinvin011235, @Bogdan1999,
>
>
> It seems in Debian Testing and Kali they have started removing some python2 packages as python2 is no longer maintained, see https://pythonclock.org/
>
> So my advice [...]

hello.. your advice is really helpful.

it would be really great if you could give the steps on installing the gr-gsm as you have mentioned.

sanja76wot, 2020-01-07:
There was the same problem as vinvin011235. I tried many instructions. Now the following comes out:

root@kali:~# sudo pybombs -p gr-gsm install gr-gsm
[INFO] Prefix Python version is: 3.7.5
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-gsm
   |
   +- libosmocore
   |
   \- gr-osmosdr
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-osmosdr
[WARNING] Build dir already exists: /usr/local/src/gr-osmosdr/build
Configuring: (100%) [============================================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Extracting version information from git describe...
-- Configuring Boost C++ Libraries...
-- Found Boost: /usr/include (found version "1.67.0") found components:  thread system chrono date_time atomic 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/include (found suitable version "1.67.0", minimum required is "1.67.0") found components:  date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
-- Found gnuradio-uhd: /usr/local/include, /usr/local/lib/libgnuradio-uhd.so                                                                       
-- Checking for module 'gnuradio-fcd'
--   No package 'gnuradio-fcd' found
-- gnuradio-fcd not found.
-- Could NOT find GNURADIO_FCD (missing: GNURADIO_FCD_LIBRARIES GNURADIO_FCD_INCLUDE_DIRS) 
-- Found gnuradio-fcdproplus: /usr/include, /usr/lib/x86_64-linux-gnu/libgnuradio-fcdproplus.so
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- libmirisdr not found.
-- Checking for module 'libfreesrp'
--   No package 'libfreesrp' found
-- libfreesrp not found.
CMake Error at CMakeLists.txt:179 (message):
  GnuRadio Runtime required to build gr-osmosdr

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gr-osmosdr/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-osmosdr:
Configuration failed                                                                                                                               
[ERROR] Error installing package gr-osmosdr. Aborting.

I'm a bad user in Linux, however, as in English)))
How to solve the installation problem?

vinvin011235, 2020-04-25:
> Hi @vinvin011235, @Bogdan1999,
>
>
> It seems in Debian Testing and Kali they have started removing some python2 packages as python2 is no longer maintained, see https://pythonclock.org/
>
> So my advice [...]

Thank You for the reply and sorry for my late reply! So i tried building it as you mentioned by installing gnuradio and gr-osmosdr. Then I went to the website you mentioned for Kali Rolling Edition but after installing and updating when I ran "cmake .." as mentioned it gave me an error.

Cloning into 'gr-gsm'...
root@Retr0:~# cd gr-gsm
root@Retr0:~/gr-gsm# mkdir build
root@Retr0:~/gr-gsm# cd build
root@Retr0:~/gr-gsm/build# cmake ..
-- The CXX compiler identification is GNU 9.2.1
-- The C compiler identification is GNU 9.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:45 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:48 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:51 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:54 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Extracting version information from git describe...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 43 lines cut ...]
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 0.12.0
-- Found libosmocoding: /usr/lib/x86_64-linux-gnu/libosmocoding.so  
-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 0.12.0
-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so  
CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gr-gsm/build/CMakeFiles/CMakeError.log".
root@Retr0:~/gr-gsm/build# mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
mkdir: cannot create directory ‘/root/.grc_gnuradio/’: File exists
mkdir: cannot create directory ‘/root/.gnuradio/’: File exists
root@Retr0:~/gr-gsm/build# make
make: *** No targets specified and no makefile found.  Stop.
root@Retr0:~/gr-gsm/build# sudo make install
make: *** No rule to make target 'install'.  Stop.

[CMakeError.log](https://github.com/ptrkrysik/gr-gsm/files/4532879/CMakeError.log)
[CMakeOutput.log](https://github.com/ptrkrysik/gr-gsm/files/4532880/CMakeOutput.log)

and these were the files the error referred to.
Thank you again!

vinvin011235, 2020-04-25:
> Hi, I also encountered this problem, please help
> 
> Error after command:
> 
> sudo pybombs prefix init / usr / local -a default_prx -R gnuradio-default.
> 
> The following content:
> 
> efault
> PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.17
> PyBOMBS - INFO - PyBOMBS Version 2.3.3
> PyBOMBS.prefix - WARNING - There already is a prefix in `/ usr / local '.
> Continue using this path Y / [N]? y
> PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.17
> Alias ​​`default_prx 'already exists, overwrite Y / [N]? y
> PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.17
> PyBOMBS.prefix - INFO - Installing default packages for prefix ...
> PyBOMBS.prefix - INFO -
>   - gnuradio
> PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
> PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
> PyBOMBS.install_manager - ERROR - Package has no install method: python-click-plugins
> 
> Notebook: dell latitude e5430
> System: Kali Linux 2019.4
> 
> P. s. Sorry if that. I am noob

Hey there, after tweaking a bit I got it to work. The problem mentioned before this comment was tackled after I found the similar problem encountered by someone else on gr-gsm forum. So thank you so much again for the support also, keep up the good work!

vinvin011235, 2020-04-25:
It's better to install it without pybombs and also I updated andupgraded gnuradio and gr-osmosdr.

closed 2020-04-25 by vinvin011235

paul0ss, 2021-02-12:
Hi @vinvin011235 
I have the same problem like u before. I have a similar error output of cmake and i have tried to install it with pybombs as well. Nothing works for me. Could u help me out or give the link to the forum where u have found your solution. thank you in advance.