ptrkrysik/gr-gsm #499: Error: module 'grgsm' has no attribute 'universal_ctrl_chans_demapper'

Rahul-Jadhav, 2019-12-30:
Hi,

Firstly, thank you for gr-gsm utility. I am reading about this utility and it seems like a great help
I am trying to run grgsm_livemon on my kali linux running on raspberry pi 2B

Kali version:
`Linux kali 4.19.81-Re4son-v7+ #1 SMP Wed Nov 6 10:16:47 AEDT 2019 armv7l GNU/Linux`

apt does not list gr-gsm package. I happened to install it using reference of [this](https://www.gitmemory.com/issue/ptrkrysik/gr-gsm/491/557219017) thread
Installation went smooth. However when I run grgsm_livemon with `rtl` as args I get attached error. Also, I tried with headless version and same error
I have PYTHONPATH variable set.

**Failure log for: grgsm_livemon_headless version**
[image]

**Failure log for: grgsm_livemon version**
[image]

I am using RTL SDR as my device and the default DVB driver is disabled. rtl_test works fine. This doesn't seem like a driver related issue

gnuradio v3.8 is installed

Reading your previous posts I have included as much information . Please let me know if you need anything more. 
Could you please help me in setting up livemon?

Regards.

Rahul-Jadhav, 2020-01-07:
@ptrkrysik / @velichkov could you help me unblock? I want to get grgsm running on my pi.

velichkov, 2020-01-08:
Hi @Rahul-Jadhav,

> Firstly, thank you for gr-gsm utility. I am reading about this utility and it seems like a great help
> I am trying to run grgsm_livemon on my kali linux running on raspberry pi 2B

RPI 2B may not be powerful enough to run all grgsm tools in realtime. For more info search the github and the [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list) archives.

> Kali version:
> `Linux kali 4.19.81-Re4son-v7+ #1 SMP Wed Nov 6 10:16:47 AEDT 2019 armv7l GNU/Linux`
> 
> apt does not list gr-gsm package.

Yes, it has been removed from the apt repositories.

>  I happened to install it using reference of [this](https://www.gitmemory.com/issue/ptrkrysik/gr-gsm/491/557219017) thread
> Installation went smooth. 

How did you install gnuradio and which version?

> However when I run grgsm_livemon with `rtl` as args I get attached error. Also, I tried with headless version and same error
> I have PYTHONPATH variable set.
> 
> **Failure log for: grgsm_livemon_headless version**
> [image]
> 
> **Failure log for: grgsm_livemon version**
> [image]

Please in the future copy-paste the text and do not upload screenshots of your terminal.

Which gr-gsm version did you install? Tell me the git commit id.
Did you have gr-gsm installed previously (which now you are trying to upgrade) or is this a new installation on a clean system?

Provide the full output of the following commands (as text, not screenshots pleas)

```
grep "class universal_ctrl_chans_demapper" /usr/local/lib/python3/dist-packages/grgsm/grgsm_swig.py 
ldd -r /usr/local/lib/python3/dist-packages/grgsm/_grgsm_swig.so
find /usr/local -name "libgrgsm.so*"
```

ptrkrysik, 2026-11-09:
This was seen with a mismatched GNU Radio 3.8 / old install. The new release has been ported to GNU Radio 3.10; please retry and open a new issue with a text log if it persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik