ptrkrysik/gr-gsm #501: Gnu Radio is not working

Angel2123194, 2020-02-02:
Hi, I'm new. I had a problem installing gr-gsm.

My system: Ubuntu 18.04.4 LTS
My rtl-sdr receiver: RTL2832U
My laptop: Dell Latitude E 5430
Installed according to the instructions: https://osmocom.org/projects/gr-gsm/wiki/Installation#Installation-from-packages-on-Debian-Testing-and-Ubuntu-1804

During installation, I noticed the following error:
[image]

When I start Gnu Radio I get the following message:
[image]

Result:
[image]

I apologize if I did something very stupid and for my English, I just recently started to get interested in gnu radio and use linux.

Not long ago, Gnu radio worked fine, but after some point started to issue this message.

I tried installing on a freshly installed system and tried installing using this instruction: https://osmocom.org/projects/gr-gsm/wiki/Installation#Compilation-and-installation-with-use-of-Pybombs.

I will be very grateful to you.

velichkov, 2020-02-05:
Hi @Angel2123194,

Please next time do not attach screenshots for your terminal but copy-paste the text here.

> Failed to load module "canberra-gtk-module"

Run `sudo apt-get install libcanberra-gtk-module libcanberra-gtk3-module`

> IBUS-WARNING **:

You should not run gnuradio with `sudo`, execute `gnuradio-companion airprobe_rtlsdr.grc`

Angel2123194, 2020-02-06:
Thanks for the answer and for helping me.

I ran the command: sudo apt-get install libcanberra-gtk-module libcanberra-gtk3-module

Then I executed these commands:
angel @ angel: ~ $ wget http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
--2020-02-06 03: 07: 59-- http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
Git.osmocom.org (git.osmocom.org) is recognized ... 78.46.96.155, 2a01: 4f8: 120: 8470 :: 2
Connection to git.osmocom.org (git.osmocom.org) | 78.46.96.155 |: 80 ... the connection is established.
HTTP request sent. Waiting for a response ... 200 OK
Length: 62212 (61K) [text / plain]
Save to: "grgsm_livemon.grc.2"

grgsm_livemon.grc.2 100% [====================>] 60.75K --.- KB / s for 0.09s

2020-02-06 03:07:59 (714 KB / s) - “grgsm_livemon.grc.2” was saved [62212/62212]

angel @ angel: ~ $ grcc -d. grgsm_livemon.grc
linux; GNU C ++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

angel @ angel: ~ $ mv grgsm_livemon.py grgsm_livemon
angel @ angel: ~ $ gnuradio-companion airprobe_rtlsdr.grc
<<< Welcome to GNU Radio Companion 3.7.11 >>>

Block paths:
/ usr / share / gnuradio / grc / blocks

I didn’t do anything else.
This did not help me in any way; the result remained the same.
What else can help me?

velichkov, 2020-02-06:
> This did not help me in any way; the result remained the same.

You are no longer seeing the two warnings that you've reported in your previous post so this actually did helped you.

> angel @ angel: ~ $ gnuradio-companion airprobe_rtlsdr.grc

Where do you get `airprobe_rtlsdr.grc` from? It's not part of the gr-gsm project since 2016. You can't use old copies of some flowgraphs with the latest version of gr-gsm.

velichkov, 2020-02-06:
> angel @ angel: ~ $ wget http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
> angel @ angel: ~ $ grcc -d. grgsm_livemon.grc
> angel @ angel: ~ $ mv grgsm_livemon.py grgsm_livemon

You don't have to download, compile and rename grgsm_livemon.grc flowgraph in order to use it. The script is already included in the package and you can start it from the terminal

```
$ grgsm_livemon -f <frequency>
```
But first you need to find a frequency of a nearby C0 GSM channel by using `grgsm_scanner` tool.

Also read [Usage](https://github.com/ptrkrysik/gr-gsm/wiki/Usage) and [Decoding How To](https://github.com/ptrkrysik/gr-gsm/wiki/Usage:-Decoding-How-To) wiki pages.

Angel2123194, 2020-02-07:
Thanks for the answer, I tried to run the command: angel @ angel: ~ $ `grgsm_scanner`
linux; GNU C ++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

angel @ angel: ~ $
I didn’t see anything, before I set the purity manually after running gnuradio

When I run the command: gnuradio-companion. I do not see blocks as in the third screenshot, is this normal?

The fact is that a week ago I started gnuradio and it worked fine, but after some time it stopped working, I tried to install the ego on absolutely clean Linux.

velichkov, 2020-02-07:
> Thanks for the answer, I tried to run the command: angel @ angel: ~ $ grgsm_scanner
> linux; GNU C ++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Do you have a suitable antenna attached to your RTL device? Also try moving near a window or outside. To get some debug messages run `grgsm_scanner --debug`

> I didn’t see anything, before I set the purity manually after running gnuradio

What do you mean by `purity`?

> When I run the command: gnuradio-companion. I do not see blocks as in the third screenshot, is this normal?

To see blocks you need to open an existing flowgraph or create a new one. In the gnuradio's wiki there are several good [tutorials](https://wiki.gnuradio.org/index.php/Tutorials), start with [Guided Tutorial GRC](https://wiki.gnuradio.org/index.php/Guided_Tutorial_GRC).

> The fact is that a week ago I started gnuradio and it worked fine, but after some time it stopped working, I tried to install the ego on absolutely clean Linux.

What is `ego`?

P.S.
Our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list) is a better place for discussions like this as more people are reading it.

Angel2123194, 2020-02-08:
> Do you have a suitable antenna attached to your RTL device? Also try moving near a window or outside. To get some debug messages run grgsm_scanner --debug

Result:
angel@angel:~$ grgsm_scanner --debug
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Usage: grgsm_scanner: [options]

grgsm_scanner: error: no such option: --debug
angel@angel:~$

> What do you mean by purity?

Sorry, I do not understand English well, so I use google translator

I mean: frequency
[image]

> To see blocks you need to open an existing flowgraph or create a new one. In the gnuradio's wiki there are several good tutorials, start with Guided Tutorial GRC.

`
Thank you, I will definitely study this material.

> What is ego?

The fact is that a week ago I launched gnuradio and it worked fine, but I tried to install it on completely pure Linux.

velichkov, 2020-02-10:
Hi @Angel2123194,

> > Do you have a suitable antenna attached to your RTL device?

Please answer the above question?

> Also try moving near a window or outside. To get some debug messages run grgsm_scanner --debug
> 
> Result:
> angel@angel:~$ grgsm_scanner --debug
> linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown
> 
> Usage: grgsm_scanner: [options]
> 
> grgsm_scanner: error: no such option: --debug

You have an old version that does not have the `--debug` flag. To see some debug output you need to open `grgsm_scanner` script and comment out lines 320 and 321

```
sudo vim /usr/bin/grgsm_scanner
```

```python
             #os.dup2(null_fds[0], 1)
             #os.dup2(null_fds[1], 2)
```

> angel@angel:~$
> 
> > What do you mean by purity?
> 
> Sorry, I do not understand English well, so I use google translator
> 
> I mean: frequency

Yes, when starting `grgsm_livemon` or `grgsm_livemon_headless` you need to specify a frequency of a known C0 BCCH cell.

In the screenshot you have specified 939.15M but that is not a valid GSM frequency, try 939.2M or 939.4M. See also [this](http://niviuk.free.fr/gsm_arfcn.php)

> `
> Thank you, I will definitely study this material.
>
>
> The fact is that a week ago I launched gnuradio and it worked fine, but I tried to install it on completely pure Linux.

All the screenshots that you've provided indicates that `gnuradio-companion` is working fine. To see some blocks try opening an existing flowgraph, `grgsm_livemon.grc` for example.

```
$ wget http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
$ gnuradio-companion ./grgsm_livemon.grc 
```

Angel2123194, 2020-02-11:
> Do you have a suitable antenna attached to your RTL device?
> 
> Please answer the above question?

I use the antenna that comes with my RTL2832U.

> You have an old version that does not have the --debug flag. To see some debug output you need to open grgsm_scanner script and comment out lines 320 and 321

like so?
[image]

I saved
Result:
```
angel@angel:~$  grgsm_scanner --debug
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Usage: grgsm_scanner: [options]

grgsm_scanner: error: no such option: --debug
angel@angel:~$
```
 

> Yes, when starting grgsm_livemon or grgsm_livemon_headless you need to specify a frequency of a known C0 BCCH cell.
```
angel@angel:~$ grgsm_livemon
Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 41, in <module>
    from PyQt4 import Qt
ImportError: No module named PyQt4
angel@angel:~$ grgsm_livemon_headless
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
/home/angel/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
vmcircbuf_createfilemapping: createfilemapping is not available
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon_headless", line 251, in <module>
    main()
  File "/usr/bin/grgsm_livemon_headless", line 246, in main
    tb.start()
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 5654, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(r, max_noutput_items)
RuntimeError: boost::filesystem::create_directory: Permission denied: "/home/angel/.gnuradio/prefs"
Reattached kernel driver
angel@angel:~$ 
```

> In the screenshot you have specified 939.15M but that is not a valid GSM frequency, try 939.2M or 939.4M. See also this

This is a screenshot from the Internet to show you how it was before when I launched gnuradio.

> All the screenshots that you've provided indicates that gnuradio-companion is working fine. To see some blocks try opening an existing flowgraph, grgsm_livemon.grc for example.
> 
> $ wget http://git.osmocom.org/gr-gsm/plain/apps/grgsm_livemon.grc
> $ gnuradio-companion ./grgsm_livemon.grc 

Result:

```
angel@angel:~$ gnuradio-companion ./grgsm_livemon.grc
<<< Welcome to GNU Radio Companion 3.7.11 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks

Loading: "/home/angel/grgsm_livemon.grc"
>>> Done
```

[image]

Thank you so much for helping me!

velichkov, 2020-02-11:
> > Do you have a suitable antenna attached to your RTL device?
> > Please answer the above question?
> 
> I use the antenna that comes with my RTL2832U.

OK.

> > You have an old version that does not have the --debug flag. To see some debug output you need to open grgsm_scanner script and comment out lines 320 and 321
> 
> like so?

Yes, that's correct.

> I saved
> Result:
> 
> ```
> angel@angel:~$  grgsm_scanner --debug
> linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown
> 
> Usage: grgsm_scanner: [options]
> 
> grgsm_scanner: error: no such option: --debug
> angel@angel:~$
> ```

Run it without `--debug` as your version does not have this flag, execute only `grgsm_scanner`

> ```
> angel@angel:~$ grgsm_livemon
> Warning: failed to XInitThreads()
> Traceback (most recent call last):
>   File "/usr/bin/grgsm_livemon", line 41, in <module>
>     from PyQt4 import Qt
> ImportError: No [...]

You need to install `python-qt4` and `python-qwt5-qt4` packages.
```
sudo apt-get install python-qt4 python-qwt5-qt4
```
> ```
> angel@angel:~$ grgsm_livemon_headless
> linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown
>
> gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
> built-in source types: file osmosdr fcd [...]

To remove these messages when running grgsm commands always specify `--args=rtl` parameter as you are having an RTL device.

> ```
> Detached kernel driver
> Found Rafael Micro R820T tuner
> Reattached kernel driver
> Using device #0 Realtek RTL2838UHIDIR SN: 00000001
> Detached kernel driver
> Found Rafael Micro R820T tuner
> [R82XX] [...]

That's quite unusual. Make sure the `/home/angel/.gnuradio/prefs` directory exist and your user owns it.

```
sudo chown -R angel:angel /home/angel/.gnuradio
mkdir -p /home/angel/.gnuradio/prefs
```

> > In the screenshot you have specified 939.15M but that is not a valid GSM frequency, try 939.2M or 939.4M. See also this
> 
> This is a screenshot from the Internet to show you how it was before when I launched gnuradio.

OK.

> Result:
>
> ```
> angel@angel:~$ gnuradio-companion ./grgsm_livemon.grc
> <<< Welcome to GNU Radio Companion 3.7.11 >>>
>
> Block paths:
> 	/usr/share/gnuradio/grc/blocks
>
> Loading: [...]

That's really strange! Right now I don't have any ideas what could have cause this. Lets first solve the above errors and after that if it still fails we will try solving this as well.

Angel2123194, 2020-02-12:
Thanks for the answer!

> Run it without --debug as your version does not have this flag, execute only grgsm_scanner

Result:

```
angel@angel:~$ grgsm_scanner
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 2 to 4000000
set_min_output_buffer on block 4 to 4000000
set_min_output_buffer on block 3 to 4000000
/home/angel/.gnuradio/prefs/vmcircbuf_default_factory: No such file or directory
vmcircbuf_createfilemapping: createfilemapping is not available
Reattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 134 to 4000000
set_min_output_buffer on block 136 to 4000000
set_min_output_buffer on block 135 to 4000000
Reattached kernel driver
Args= 
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
set_min_output_buffer on block 266 to 4000000
set_min_output_buffer on block 268 to 4000000
set_min_output_buffer on block 267 to 4000000
Reattached kernel driver
Args= 
[... 308 lines cut ...]
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 3f 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
 15 06 21 00 01 f0 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b 2b
```
This is victory?)

> That's quite unusual. Make sure the /home/angel/.gnuradio/prefs directory exist and your user owns it.

I do not know how to give you the answer
[image]

[image]

> That's really strange! Right now I don't have any ideas what could have cause this. Lets first solve the above errors and after that if it still fails we will try solving this as well.

Thank you for helping me, I don’t know what I would do without you !!!

velichkov, 2020-02-12:
> Result: 
> ```
> angel@angel:~$ grgsm_scanner
> ARFCN:   21, Freq:  939.2M, CID:  XXXX, LAC: XXXX, MCC: XXX, MNC:  XX, Pwr: -30
> ```

It works! Now you can uncomment the two lines in `grgsm_scanner` as you don't need all the debug messages. When posting logs you may consider censoring CID, LAC, MCC and MNC like I did in the line above. You can edit your previous message and redact them all.
 
> > To remove these messages when running grgsm commands always specify --args=rtl parameter as you are having an RTL device
> angel@angel:~$ grgsm_livemon_headless --args=rtl
>
> This is victory?)

Cool! When starting `grgsm_livemon_headless` or `grgsm_livemon` you can specify different frequencies  `--fc 939.2M`. You can get all the frequencies from the `grgsm_scanner` output.  You got really lucky that there is a GSM cell on the default frequency.
```
grgsm_livemon_headless --args=rtl --fc 939.2M
```
> > That's quite unusual. Make sure the /home/angel/.gnuradio/prefs directory exist and your user owns it.
> 
> I do not know how to give you the answer

The filename starts with a dot (`.gnuradio`) and this makes the directory hidden and the file explorer won't show it by default. Somewhere in the settings (the icons in the top right corner) you should be able to select "Show hidden files" to see them.

In the terminal to see all hidden files you need to execute `ls -l --all` or `ls -la`.

The following two commands will change the ownership and create this two directories if missing. Have you executed them?
```
sudo chown -R angel:angel /home/angel/.gnuradio
mkdir -p /home/angel/.gnuradio/prefs
```
> > That's really strange! Right now I don't have any ideas what could have cause this. Lets first solve the above errors and after that if it still fails we will try solving this as well.

Now are there any differences in gnuradio companion or it is still the same? 

> Thank you for helping me, I don’t know what I would do without you !!!

You are welcome!

Angel2123194, 2020-02-13:
> It works! Now you can uncomment the two lines in grgsm_scanner as you don't need all the debug messages. When posting logs you may consider censoring CID, LAC, MCC and MNC like I did in the line above. You can edit your previous message and redact them all.

Excuse me, I misunderstood what I need to do?

> Cool! When starting grgsm_livemon_headless or grgsm_livemon you can specify different frequencies --fc 939.2M. You can get all the frequencies from the grgsm_scanner output. You got really lucky that there is a GSM cell on the default frequency.

Great!

> The filename starts with a dot (.gnuradio) and this makes the directory hidden and the file explorer won't show it by default. Somewhere in the settings (the icons in the top right corner) you [...]

[image]

> The following two commands will change the ownership and create this two directories if missing. Have you executed them?

Yes

> Now are there any differences in gnuradio companion or it is still the same?

gnuradio companion nothing has changed, no blocks

velichkov, 2020-02-14:
> Excuse me, I misunderstood what I need to do?

You need to revert the changes you did in `grgsm_scanner` as it works and you don't need all these debug messages. You need to open it with a text editor and uncomment lines  lines 320 and 321 or execute `sudo apt-get install --reinstall gr-gsm`
 
> > In the terminal to see all hidden files you need to execute ls -l --all or ls -la.

Now  `.gnuradio`  has correct ownership and permissions. Do you still get the ` Permission denied: "/home/angel/.gnuradio/prefs"` error when running livemon?

Please next time do not attach screenshots for your terminal but copy-paste the text here.

> > Now are there any differences in gnuradio companion or it is still the same?
> 
> gnuradio companion nothing has changed, no blocks

Try reinstalling gnuradio, execute `sudo apt-get install --reinstall gnuradio`. If it still does not show any blocks try deleting the `.gnuradio` directory, `gnuradio-companion` will recreate it with some default values on next start.

Angel2123194, 2020-02-14:
> You need to revert the changes you did in grgsm_scanner as it works and you don't need all these debug messages. You need to open it with a text editor and uncomment lines lines 320 and 321 or execute sudo apt-get install --reinstall gr-gsm

Ok i did it

> Now .gnuradio has correct ownership and permissions. Do you still get the Permission denied: "/home/angel/.gnuradio/prefs" error when running livemon?

```
angel@angel:~$ grgsm_livemon
Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/usr/bin/grgsm_livemon", line 45, in <module>
    from gnuradio import qtgui
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/__init__.py", line 37, in <module>
    from range import Range, RangeWidget
  File "/usr/lib/python2.7/dist-packages/gnuradio/qtgui/range.py", line 67, in <module>
    class RangeWidget(QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'
angel@angel:~$ 
```

> Please next time do not attach screenshots for your terminal but copy-paste the text here.

Well, I'm sorry, I thought it would be convenient for you

> Try reinstalling gnuradio, execute sudo apt-get install --reinstall gnuradio. If it still does not show any blocks try deleting the .gnuradio directory, gnuradio-companion will recreate it with some default values on next start.

It didn't help me either.

This is somehow abnormal. If you want, I can give you access to my device.

velichkov, 2020-02-15:
> ```
> angel@angel:~$ grgsm_livemon
> Warning: failed to XInitThreads()
> Traceback (most recent call last):
>   File "/usr/bin/grgsm_livemon", line 45, in <module>
>     from gnuradio import qtgui
>   File [...]

This is a known issue :disappointed: - https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-356401976, https://github.com/ptrkrysik/gr-gsm/issues/355#issuecomment-366277650. At the end you may need to download and build `grgsm_livemon.grc`  #manually.

```
$ cd $HOME
$ wget https://raw.githubusercontent.com/ptrkrysik/gr-gsm/fa184a9447a90aefde2ca0dea1347b702551015d/apps/grgsm_livemon.grc
$ grcc -d . grgsm_livemon.grc 
$ ./grgsm_livemon.py 
```
> > Try reinstalling gnuradio, execute sudo apt-get install --reinstall gnuradio. If it still does not show any blocks try deleting the .gnuradio directory, gnuradio-companion will recreate it with some default values on next start.
> 
> It didn't help me either.

I'm running out of ideas. You may consider asking in the gnuradio's [mailing list ](https://wiki.gnuradio.org/index.php/MailingLists) or [IRC/Slack channels](https://wiki.gnuradio.org/index.php/IRC)

Angel2123194, 2020-02-16:
Hello, I don’t know how, but the problem was solved
[image]
I did not do anything.

> This is a known issue disappointed - #355 (comment), #355 (comment). At the end you may need to download and build grgsm_livemon.grc #manually.

Will I need to solve this problem?

> I'm running out of ideas. You may consider asking in the gnuradio's mailing list or IRC/Slack channels

Thank you very much for helping me! Do I need to close issues?

velichkov, 2020-02-17:
Hi @Angel2123194,

> Hello, I don’t know how, but the problem was solved

That's great!

> > This is a known issue disappointed - #355 (comment), #355 (comment). At the end you may need to download and build grgsm_livemon.grc #manually.
> 
> Will I need to solve this problem?

You've already solved it.
  
> Thank you very much for helping me!

You are welcome.

> Do I need to close issues?

Yes and in the future if you have any new problems or questions don't hesitate to open a new issue or post to our mailing list.

Angel2123194, 2020-02-19:
Many thanks for your help.

closed 2020-02-19 by Angel2123194