ptrkrysik/gr-gsm #502: cmake .. Error

greencorsairs, 2020-02-09:
[image]

velichkov, 2020-02-10:
Hi @greencorsairs,

See #480 and #475. And in the future please do not attach screenshots of your terminal, you can copy-paste the text instead.

greencorsairs, 2020-02-15:
[ 73%] Generating grgsm_livemon
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
/root/.grc_gnuradio
/home/leone/Desktop/gr-gsm/grc
/usr/share/gnuradio/grc/blocks

>>> Loading: /home/leone/Desktop/gr-gsm/apps/grgsm_livemon.grc
Exception in static block register_rfnoc_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key Block for the second time.
Exception in static block register_rfnoc_ddc_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key DDC for the second time.
Exception in static block register_rfnoc_duc_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key DUC for the second time.
Exception in static block register_rfnoc_fir_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key FIR for the second time.
Exception in static block register_rfnoc_null_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key NullSrcSink for the second time.
Exception in static block register_rfnoc_window_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key Window for the second time.
Exception in static block register_rfnoc_siggen_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key SigGen for the second time.
Exception in static block register_rfnoc_dma_fifo_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key DmaFIFO for the second time.
Exception in static block register_rfnoc_replay_block_ctrl
RuntimeError: Attempting to register an RFNoC block with key Replay for the second time.
Exception in static block reg_basic_and_lf_dboards
LookupError: KeyError: The dboard id 0x0000 is already registered to Basic TX.
Exception in static block reg_rfx_dboards
LookupError: KeyError: The dboard id pair [0x0024, 0x0028] is already registered to RFX400.
Exception in static block reg_xcvr2450_dboard
LookupError: KeyError: The dboard id pair [0x0061, 0x0060] is already registered to XCVR2450.
Exception in static block reg_sbx_dboards
LookupError: KeyError: The dboard id pair [0x0054, 0x0055] is already registered to SBX.
Exception in static block reg_ubx_dboards
LookupError: KeyError: The dboard id pair [0x0074, 0x0073] is already registered to UBX v0.3.
Exception in static block reg_wbx_simple_dboards
LookupError: KeyError: The dboard id pair [0x0053, 0x0052] is already registered to WBX.
Exception in static block reg_dbsrx_dboard
LookupError: KeyError: The dboard id 0x000d is already registered to DBSRX.
Exception in static block reg_unknown_dboards
LookupError: KeyError: The dboard id 0xfff0 is already registered to Unknown TX.
Exception in static block reg_tvrx_dboard
LookupError: KeyError: The dboard id 0x0040 is already registered to TVRX.
Exception in static block reg_dbsrx2_dboard
LookupError: KeyError: The dboard id 0x0012 is already registered to DBSRX2.
Exception in static block reg_tvrx2_dboard
LookupError: KeyError: The dboard id 0x0046 is already registered to TVRX2.
Exception in static block reg_twinrx_dboards
LookupError: KeyError: The dboard id 0x0091 is already registered to TwinRX Rev A.
Exception in static block register_rfnoc_e320_radio_ctrl
RuntimeError: Attempting to register an RFNoC block with key NeonRadio for the second time.
Exception in static block register_rfnoc_magnesium_radio_ctrl
RuntimeError: Attempting to register an RFNoC block with key MagnesiumRadio for the second time.
Exception in static block register_rfnoc_rhodium_radio_ctrl
RuntimeError: Attempting to register an RFNoC block with key RhodiumRadio for the second time.
Exception in static block register_rfnoc_x300_radio_ctrl
RuntimeError: Attempting to register an RFNoC block with key X300Radio for the second time.
>>> Generating: /home/leone/Desktop/gr-gsm/build/apps/grgsm_livemon.py
double free or corruption (!prev)
Child aborted
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:590: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

velichkov, 2020-02-15:
Try to disable building of `grgsm_livemon` and `grgsm_livemon_headless`  
```
cmake -DENABLE_GRCC=OFF ..
```
and then if it builds and installs successfully try building `apps/grgsm_livemon.grc` and `apps/grgsm_livemon_headless.grc` manually using `gnuradio-companion`.

```
gnuradio-companion apps/grgsm_livemon.grc
gnuradio-companion apps/grgsm_livemon_headless.grc
```

greencorsairs, 2020-02-17:
#grgsm_livemon
Traceback (most recent call last):
File "/usr/bin/grgsm_livemon", line 25, in <module>
from gnuradio import blocks
ImportError: No module named gnuradio

velichkov, 2020-02-17:
https://github.com/ptrkrysik/gr-gsm/search?q=%22ImportError%3A+No+module+named+gnuradio%22&type=Issues

https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

greencorsairs, 2020-02-17:
do you have a tutorial i can follow ?

velichkov, 2020-02-17:
> do you have a tutorial i can follow ?

Sorry, I don't have. The [official instructions](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition) are probably a bit outdated in regards to gnuradio 3.8 but they are still useful. Avoid the `pybombs` installation method as in my opinion it is the hardest and most error prone one.  Also read all messages in #480. 

When asking for help it's usually good to provide some more details about your system and what you did, like which Linux distribution do you use, how did you install gnuradio, gr-osmosdr and gr-gsm and which versions, which commands did you execute including all their arguments and their outout, etc... 

You may also post to our [mailing list](https://github.com/ptrkrysik/gr-gsm/#mailing-list).

greencorsairs, 2020-02-29:
how i can get tmsi and key ?

greencorsairs, 2020-02-29:
how i can get tmsi and key of the phone number ?

9600, 2020-02-29:
By being the mobile operator or other entity provisioning the SIM card and key etc.

greencorsairs, 2020-02-29:
how i can find that ?

greencorsairs, 2020-02-29:
[image]

9600, 2020-02-29:
Simple, just set up your own mobile phone company.

greencorsairs, 2020-02-29:
how ?

greencorsairs, 2020-02-29:
I have a phone number how can I get TMSI and KC !

greencorsairs, 2020-02-29:
#grgsm_livemon
Traceback (most recent call last):
File "/usr/bin/grgsm_livemon", line 25, in
from gnuradio import blocks
ImportError: No module named gnuradio

greencorsairs, 2020-03-01:
!!

ptrkrysik, 2026-11-09:
This report was against the GNU Radio 3.8 / pre-port code and build system. The project has since been ported to GNU Radio 3.10; please try the new release and open a new issue with full details if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik