ptrkrysik/gr-gsm #503: Using gr-gsm with gnuradio 3.8 with bladeRF (on Kali)

botherder, 2020-02-10:
I am trying to run gr-gsm using a bladeRF x40 on a Kali (rolling release) and a Raspberry Pi 4. Following the useful conversation at #480, I have installed the maint-3.8 branch from https://github.com/velichkov/gr-gsm/tree/maint-3.8.

I followed the instructions from here:
https://osmocom.org/projects/gr-gsm/wiki/Installation

Using other rtl-sdr dongles everything seems to be working fine, but when using the bladeRF both `grgsm_scanner` and `grgsm_livemon` fail. This is the output launching `grgsm_scanner --debug`:

```
root@kali:~# grgsm_scanner --debug

Args= 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
[ERROR] avahi_service_browser_new() failed: Bad state
gr-osmosdr 0.1.5 (0.1.4) gnuradio 3.8.0.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # [REDACTED] FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: Warning: number of channels specified on command line (4) is greater than the maximum number supported by this device (2). Resetting to 2.
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 438, in <module>
    main()
  File "/usr/local/bin/grgsm_scanner", line 435, in main
    options.ppm, options.gain, options.args, prn = printfunc, debug = options.debug)
  File "/usr/local/bin/grgsm_scanner", line 333, in do_scan
    ppm=ppm, gain=gain, args=args)
  File "/usr/local/bin/grgsm_scanner", line 212, in __init__
    str(grgsm.device.get_default_args(args)))
  File "/usr/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 1316, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: destination port 1 out of range for source_impl(1)
```

This is the output from bladeRF-cli:

```
root@kali:~# bladeRF-cli -i
bladeRF> info

  Board:                    Nuand bladeRF 2.0 (bladerf2)
  Serial #:                 [REDACTED]
  VCTCXO DAC calibration:   0x1f77
  FPGA size:                49 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  2
  USB address:              3
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> probe

  Description:    Nuand bladeRF 2.0 (currently open)
  Backend:        libusb
  Serial:         [REDACTED]
  USB Bus:        2
  USB Address:    3
```

Anyone else is having the same issue?

velichkov, 2020-02-10:
Hi @botherder,

> [bladeRF source] bladerf_source_c: Warning: number of channels specified on command line (4) is greater than the maximum number supported by this device (2). Resetting to 2.

That's really odd warning as you are not specifing `nchan=` on the command line (in the `--args`)

Try adding `bladerf` to `--args`
```
$ grgsm_scanner --args=bladerf
$ grgsm_livemon_headless --args=bladerf
``` 
What's the output of
```
$ grgsm_scanner --list-devices
```

velichkov, 2020-02-10:
Hi @botherder,

> [bladeRF source] bladerf_source_c: Warning: number of channels specified on command line (4) is greater than the maximum number supported by this device (2). Resetting to 2.

That's really odd as you are not specifying `nchan=` on the command line (in the `--args`).  What's the output of `grgsm_scanner --list-devices --args=nofake`? 

Try adding `bladerf` to `--args`
```
grgsm_scanner --args=bladerf
grgsm_livemon_headless --args=bladerf
``` 
and if it still fails provide the full output of the above commands.

botherder, 2020-02-10:
Ah, interesting. With `--args=bladerf` seems to be working fine.

```
root@kali:~# grgsm_scanner --list-devices --args=nofake
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
[ERROR] avahi_service_browser_new() failed: Bad state
bladerf=0,label='Nuand bladeRF SN [REDACTED]'
backend=libusb,device=0x02:0x03,driver=bladerf,instance=0,label='BladeRF #0 [REDACTED]',serial=[REDACTED],soapy=1
```
Thank you @velichkov.

velichkov, 2020-02-10:
> Ah, interesting. With --args=bladerf seems to be working fine.

Now that's a new mystery, in theory it should fails the same way.

Can you try
```
grgsm_scanner --args="bladerf=0" --debug
grgsm_scanner --args="bladerf=0,label='Nuand bladeRF SN [REDACTED]'" --debug
grgsm_scanner --args="backend=libusb,device=0x02:0x03,driver=bladerf,instance=0,label='BladeRF #0 [REDACTED]',serial=[REDACTED],soapy=1" --debug
```
replace `[REDACTED]` with the exact output as you see it in the `--list-devices` and tell me if any of the above commands fails the same way.

Also do you see the warning about number of channels when running with it with `--args=bladerf`?

The `RuntimeError: destination port 1 out of range for source_impl(1)` exception is called from [here](https://github.com/gnuradio/gnuradio/blob/maint-3.8/gnuradio-runtime/lib/hier_block2_detail.cc#L140-L148) and the `connect()` function is called from [here](https://github.com/osmocom/gr-osmosdr/blob/gr3.8/lib/source_impl.cc#L401-L415).

How did you install gr-osmosdr, which version and from the osmocom's repo or from some fork?
Do you have gr-iqbal installed, which version and from which repo?

botherder, 2020-02-11:
I am not seeing the warning about the channels anymore.

Here is the results for each of the 3 commands you requested.

```grgsm_scanner --args="bladerf=0" --debug```

This command ran successfully.

```grgsm_scanner --args="bladerf=0,label='Nuand bladeRF SN [REDACTED]'" --debug```

---

This commands fails with the same error:

```root@kali:~# grgsm_scanner --args="bladerf=0,label='Nuand bladeRF SN [REDACTED]'" --debug

Args= bladerf=0,label='Nuand bladeRF SN [REDACTED]'
gr-osmosdr 0.1.5 (0.1.4) gnuradio 3.8.0.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # [REDACTED] FW v2.3.2 FPGA v0.11.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: Warning: number of channels specified on command line (4) is greater than the maximum number supported by this device (2). Resetting to 2.
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 438, in <module>
    main()
  File "/usr/local/bin/grgsm_scanner", line 435, in main
    options.ppm, options.gain, options.args, prn = printfunc, debug = options.debug)
  File "/usr/local/bin/grgsm_scanner", line 333, in do_scan
    ppm=ppm, gain=gain, args=args)
  File "/usr/local/bin/grgsm_scanner", line 212, in __init__
    str(grgsm.device.get_default_args(args)))
  File "/usr/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 1316, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: destination port 1 out of range for source_impl(1)```

---

```grgsm_scanner --args="backend=libusb,device=0x02:0x03,driver=bladerf,instance=0,label='BladeRF #0 [REDACTED]',serial=[REDACTED],soapy=1" --debug```

This command fails with the following error:

```root@kali:~# grgsm_scanner --args="backend=libusb,device=0x02:0x03,driver=bladerf,instance=0,label='BladeRF #0 [REDACTED]',serial=[REDACTED],soapy=1" --debug

Args= backend=libusb,device=0x02:0x03,driver=bladerf,instance=0,label='BladeRF #0 [REDACTED]',serial=[REDACTED],soapy=1
gr-osmosdr 0.1.5 (0.1.4) gnuradio 3.8.0.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] bladerf_open_with_devinfo()
[INFO] bladerf_get_serial() = [REDACTED]
[INFO] setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[INFO] setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 438, in <module>
    main()
  File "/usr/local/bin/grgsm_scanner", line 435, in main
    options.ppm, options.gain, options.args, prn = printfunc, debug = options.debug)
  File "/usr/local/bin/grgsm_scanner", line 333, in do_scan
    ppm=ppm, gain=gain, args=args)
  File "/usr/local/bin/grgsm_scanner", line 212, in __init__
    str(grgsm.device.get_default_args(args)))
  File "/usr/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 1316, in make
    return _osmosdr_swig.source_make(*args, **kwargs)
RuntimeError: setupStream invalid channel selection```

---

I have gr-osmosdr 0.1.5-1+b1 from Kali kali-rolling apt repositories. I have gr-iqbal 0.38-4 also from kali-rolling.

velichkov, 2020-02-11:
> grgsm_scanner --args="bladerf=0,label='Nuand bladeRF SN [REDACTED]'" --debug

The problem is most probably related to the spaces in the `label` field and is probably specific to bladerf. [Here](https://lists.osmocom.org/pipermail/osmocom-sdr/2017-May/001586.html) is a patch for a similar problem but for an older version of `gr-osmosdr` and the the code there is no longer the same.

In `gqrx` they are [removing the label field](https://github.com/csete/gqrx/blob/dcd57dde2994893cbe8e72c88afb3da7e0ccdabe/src/qtgui/ioconfig.cpp#L244-L248) probably for similar reasons. I may implement a similar workaround solution in gr-gsm (in `python/misc_utils/device.py`) if there is not other good ideas how to solve this but for now I guess you need to always specify `--args=bladerf`.

botherder, 2020-02-11:
Interesting. Thanks a lot. I will stick to `--args=bladerf`. Should I close this ticket?

closed 2020-02-11 by botherder

sfrmattos, 2020-06-07:
@botherder did you managed to get grgsm to find channels with bladerf?
I have the same "symptoms" that you pointed, and using --args=bladerf get rid of them, but with no channels listed. I also noted that grgsm_livemon launch, but is unable to find anything (even on the channels I know that are working). Bladerf is working well on gqrx.
on the same system, rtlsdr works fine.
I dont want to bother you asking for help, I just wanna know if it worked like described here.Thank you!