ptrkrysik/gr-gsm #504: Error installing package libosmocore. Aborting.

960rus, 2020-02-12:
Hi! Please help
checking for library containing sctp_bindx... no
configure: error: sctp_bindx not found in searched libs
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package libosmocore:
Configuration failed                                                       
[ERROR] Error installing package libosmocore. Aborting.

960rus, 2020-02-12:
Sorry, no libsctp-dev
Installed libsctp-dev
-
now a error
### [ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-osmosdr:
Configuration failed                                                       
[ERROR] Error installing package gr-osmosdr. Aborting.

velichkov, 2020-02-12:
Hi @960rus,

Please report `pybombs` related problems in the [pybombs](https://github.com/gnuradio/pybombs) or [gr-recipes](https://github.com/gnuradio/gr-recipes) projects.

> [ERROR] Problem occurred while building package gr-osmosdr:

In case you've installed gnuradio 3.8 then you need to install gr-osmosdr from the `gr3.8` branch. You could edit `~/.pybombs/recipes/gr-recipes/gr-osmosdr.lwr` and add `gitbranch: gr3.8` or 
execute the following commands
```
    pybombs config -P gr-osmosdr gitbranch gr3.8
    pybombs refetch gr-osmosdr
    pybombs install gr-osmosdr
```
See also [this discussion](https://lists.gnu.org/archive/html/discuss-gnuradio/2020-01/msg00068.html)

960rus, 2020-02-13:
Thanks! 
Error make
 73%] Generating grgsm_livemon
<<< Welcome to GNU Radio Companion Compiler 3.8.0.0 >>>

Block paths:
        /home/kali/gr-gsm/grc
        /usr/local/share/gnuradio/grc/blocks

>>> Loading: /home/kali/gr-gsm/apps/grgsm_livemon.grc
>>> Load Error: /home/kali/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:85: apps/grgsm_livemon] Ошибка 1
make[1]: *** [CMakeFiles/Makefile2:590: apps/CMakeFiles/pygen_apps.dir/all] Ошибка 2
make: *** [Makefile:141: all] Ошибка 2

velichkov, 2020-02-14:
Provide the whole output as a text file attachment or copy-paste it in pastebin and provide a link.

velichkov, 2020-02-14:
Also may I ask why did you choose to install gnuradio using `pybombs` and not from the Kali official repository (using `apt-get`)?

See also:

- https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition
- https://osmocom.org/projects/gr-gsm/wiki/Installation#Compilation-and-installation-with-use-of-Pybombs

960rus, 2020-02-15:
I want to install gr-gsm. In gnuradio-companion grgsm_livemon.grc does not work.

960rus, 2020-02-15:
> Also may I ask why did you choose to install gnuradio using pybombs and not from the Kali official repository (using apt-get)?
> See also: [...]
Thank's you answer. 
It turns out I went the wrong way, thank you so much.
---
kali@kali:~$ grgsm_livemon
Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'
Please point in the right direction

velichkov, 2020-02-15:
> I want to install gr-gsm.

This is obvious but is not what I'm asking.  Why did you choose to install it using `pybombs` (the most difficult and error prone method in my opinion)?

> ModuleNotFoundError: No module named 'grgsm'

See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188 and https://github.com/ptrkrysik/gr-gsm/issues/28#issuecomment-569353611

960rus, 2020-02-16:
Hi! Thank all works on Kali2020! You are the best!!!

closed 2026-11-09 by ptrkrysik