ptrkrysik/gr-gsm #505: Linking CXX executable filter_qa_firdes.cc error

TeodorBankov, 2020-02-21:
I am getting an error while trying to install gr-gsm package
[ 51%] Linking CXX executable filter_qa_firdes.cc
/usr/bin/ld: libgnuradio-filter.so.3.8.1.0: undefined reference to `gr::fft::window::build(gr::fft::window::win_type, int, double)'
/usr/bin/ld: libgnuradio-filter.so.3.8.1.0: undefined reference to `gr::fft::window::max_attenuation(gr::fft::window::win_type, double)'
collect2: error: ld returned 1 exit status
make[2]: *** [gr-filter/lib/CMakeFiles/filter_qa_firdes.cc.dir/build.make:104: gr-filter/lib/filter_qa_firdes.cc] Error 1
make[1]: *** [CMakeFiles/Makefile2:8010: gr-filter/lib/CMakeFiles/filter_qa_firdes.cc.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
This is the error log, tried installing gnuradio not in a bundle, failed with the same error output.
I also tried installing it multiple times, still with the same error.
The OS is Kali-Linux on a Custom build thinkpad x200

velichkov, 2020-02-21:
Hi @TeodorBankov,

> I am getting an error while trying to install gr-gsm package
> [ 51%] Linking CXX executable filter_qa_firdes.cc
> /usr/bin/ld: libgnuradio-filter.so.3.8.1.0: undefined reference to [...]

You are having trouble compiling `gnuradio` and not `gr-gsm`.

> PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
> This is the error log, tried installing gnuradio not in a bundle, failed with the same error output.
> I also tried [...]

May I ask you why did you choose to install `gr-gsm` (and `gnuradio`) using `pybombs`? In my opinion the `pybombs` installation method is the hardest one. 

There are `gnuradio` and `gr-osmosdr` dev packages in the Kali repository and you can simply install them using `apt-get` and then build and install `gr-gsm` from source. For now you need to use the `maint-3.8` branch from [my fork](https://github.com/velichkov/gr-recipes/tree/maint-3.8) as the `master` branch does not yet support gnuradio 3.8. If you decide to try this first remove everything installed by `pybombs`.
```
pybombs inv
pybombs remove gnuradio  #and add every package from the output of previous command
```
See also [our installation instructions for Debian based distribution](https://osmocom.org/projects/gr-gsm/wiki/Installation#Debbian-based-distributions-Debian-Testing-Ubuntu-1604-Kali-Rolling-Edition), #480, #475 and [Kali.docker](https://github.com/velichkov/gr-gsm/blob/f722b32/tests/dockerfiles/Kali.docker#L4-L32)

ptrkrysik, 2026-11-09:
This is a GNU Radio / PyBOMBS build problem, not a gr-gsm one; please report it to the GNU Radio project.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik