ptrkrysik/gr-gsm #506: grgsm_scanner usage

aep, 2020-02-23:
is there a manual for grgsm_scanner?

as far as i can tell, it just sits there using cpu but never printing anything.

also --verbose does nothing

velichkov, 2020-02-24:
Hi @aep,

> is there a manual for grgsm_scanner?

Unfortunately there is no manual. 

> as far as i can tell, it just sits there using cpu but never printing anything.
> also --verbose does nothing

The latest versions have a `--debug` flag.
```
$ grgsm_scanner --help

  -v, --verbose         If set, verbose information output is printed: ccch
                        configuration, cell ARFCN's, neighbour ARFCN's
  -d, --debug           Print additional debug messages
```
and in case you are using an older version you need to open grgsm_scanner and comment out lines 320 and 321.
```
320             os.dup2(null_fds[0], 1)
321             os.dup2(null_fds[1], 2)
```
> there's no way to know if it actually opened the device either, since it does the same thing with or without usb plugged in.

Use the `--debug` flag or comment out the above two lines.

Note that due to a bug in gnuradio 3.7 old versions of gr-osmosdr were using null source in case no SDR device is found. See https://github.com/osmocom/gr-osmosdr/blob/v0.1.3/lib/source_impl.cc#L357

```
Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.
```

Which Linux distribution do you use, how did you install `gnuradio` and `gr-gsm`, and which versions?

aep, 2020-02-24:
thanks @velichkov 

```
Note that due to a bug in gnuradio 3.7 old versions of gr-osmosdr were using null source in case no SDR device is found. See https://github.com/osmocom/gr-osmosdr/blob/v0.1.3/lib/source_impl.cc#L357
```

ah, that confused me alot. 
debug output helped! its a simple permission problem.

i really think debug output should be the default, so the next person doesn't open a bug.

> Which Linux distribution do you use

installed ubuntu in a VM since i couldn't figure out the python stuff on arch.

closed 2020-02-24 by aep

velichkov, 2020-02-24:
Hi @aep,

It's great that you've been able to find and fix the problem that fast.

> i really think debug output should be the default,

I disagree here, the debug output is huge and most of the time irrelevant.

> installed ubuntu in a VM since i couldn't figure out the python stuff on arch.

In Arch Linux they have updated gnuradio to 3.8 (most probably with python3) and for now for 3.8 you need to compile gr-gsm from the `maint-3.8` branch of [my fork](https://github.com/velichkov/gr-gsm/tree/maint-3.8)