ptrkrysik/gr-gsm #507: Issue with building and installing bladeRF and gnu

talasalim, 2020-03-07:
```
**sudo apt-get install libvolk1-dev**
[sudo] password for ubuntu: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  efibootmgr libfwup1 libwayland-egl1-mesa
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libvolk1-bin libvolk1.3
The following NEW packages will be installed:
  libvolk1-bin libvolk1-dev libvolk1.3
0 upgraded, 3 newly installed, 0 to remove and 3 not upgraded.
Need to get 472 kB of archives.
After this operation, 4,001 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ae.archive.ubuntu.com/ubuntu bionic/universe amd64 libvolk1.3 amd64 1.3-3 [198 kB]
Get:2 http://ae.archive.ubuntu.com/ubuntu bionic/universe amd64 libvolk1-bin amd64 1.3-3 [182 kB]
Get:3 http://ae.archive.ubuntu.com/ubuntu bionic/universe amd64 libvolk1-dev amd64 1.3-3 [91.8 kB]
Fetched 472 kB in 3s (181 kB/s)        
Selecting previously unselected package libvolk1.3:amd64.
(Reading database ... 207084 files and directories currently installed.)
Preparing to unpack .../libvolk1.3_1.3-3_amd64.deb ...
Unpacking libvolk1.3:amd64 (1.3-3) ...
Selecting previously unselected package libvolk1-bin.
Preparing to unpack .../libvolk1-bin_1.3-3_amd64.deb ...
Unpacking libvolk1-bin (1.3-3) ...
Selecting previously unselected package libvolk1-dev:amd64.
Preparing to unpack .../libvolk1-dev_1.3-3_amd64.deb ...
Unpacking libvolk1-dev:amd64 (1.3-3) ...
Setting up libvolk1.3:amd64 (1.3-3) ...
Setting up libvolk1-dev:amd64 (1.3-3) ...
Setting up libvolk1-bin (1.3-3) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...

I'm following https://github.com/Nuand/bladeRF/wiki/Getting-Started:-Linux#Compile_GNU_Radio to download bladeRF and GNURadio but I'm having several issues with downloading it and after following some solutions from different threads, this error is still repetitive. 

ubuntu@ubuntu:~$ **pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default**
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[WARNING] There already is a prefix in `/home/ubuntu/pybombs/bladeRF'.
Continue using this path Y/[N]? y
[INFO] Prefix Python version is: 2.7.17
Alias `bladeRF' already exists, overwrite Y/[N]? n
[WARNING] Aborting.
ubuntu@ubuntu:~$ pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[WARNING] There already is a prefix in `/home/ubuntu/pybombs/bladeRF'.
Continue using this path Y/[N]? n
[ERROR] Aborting. A prefix already exists in `/home/ubuntu/pybombs/bladeRF'
ubuntu@ubuntu:~$ mkdir ~/pybombs/
mkdir: cannot create directory ‘/home/ubuntu/pybombs/’: File exists
ubuntu@ubuntu:~$ mkdir ~/pybombs/
ubuntu@ubuntu:~$ pybombs config --package gnuradio gitrev maint-3.7
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Using config file: /home/ubuntu/.pybombs/config.yml
Setting keys for packages gnuradio:
gitrev: maint-3.7
ubuntu@ubuntu:~$ pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default
[INFO] Prefix Python version is: 2.7.17
[INFO] PyBOMBS Version 2.3.4a0
[WARNING] There already is a prefix in `/home/ubuntu/pybombs/bladeRF'.
Continue using this path Y/[N]? y
[INFO] Prefix Python version is: 2.7.17
Alias `bladeRF' already exists, overwrite Y/[N]? y
[INFO] Creating new config file /home/ubuntu/pybombs/bladeRF/.pybombs/config.yml
[INFO] Prefix Python version is: 2.7.17
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- gnuradio
   |
   \- uhd
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: uhd
[INFO] Install python-apt to speed up apt processing.
Cloning into 'uhd'...
remote: Enumerating objects: 79, done.
remote: Counting objects: 100% (79/79), done.
remote: Total 120 (delta 79), reused 79 (delta 79), pack-reused 41
Receiving objects: 100% (120/120), 11.06 KiB | 2.21 MiB/s, done.
Resolving deltas: 100% (91/91), completed with 29 local objects.
[INFO] Install python-apt to speed up apt processing.
Configuring: (100%) [=======================================================================================================================]
Building:    (100%) [=======================================================================================================================]
Installing:  (100%) [=======================================================================================================================]
[INFO] Installation successful.
[INFO] Installing package: gnuradio
Cloning into 'gnuradio'...
remote: Enumerating objects: 1164, done.
remote: Counting objects: 100% (1164/1164), done.
remote: Total 2892 (delta 1164), reused 1164 (delta 1164), pack-reused 1728
Receiving objects: 100% (2892/2892), 1.28 MiB | 1.88 MiB/s, done.
Resolving deltas: 100% (2315/2315), completed with 322 local objects.
Checking connectivity: 155979, done.
Submodule 'volk' (https://github.com/gnuradio/volk.git) registered for path 'volk'
submodule 'volk' cannot add alternate: path '/home/ubuntu/.pybombs/gitcache/modules/volk/' does not existCloning into '/home/ubuntu/pybombs/bladeRF/src/gnuradio/volk'...
remote: Enumerating objects: 76, done.        
remote: Counting objects: 100% (76/76), done.        
remote: Compressing objects: 100% (54/54), done.        
remote: Total 8824 (delta 27), reused 37 (delta 14), pack-reused 8748        
Receiving objects: 100% (8824/8824), 2.43 MiB | 316.00 KiB/s, done.
Resolving deltas: 100% (6049/6049), done.
Submodule path 'volk': checked out '1299d72c396a88fd2679adfd7a919ac00d2cf678'
Branch 'maint-3.7' set up to track remote branch 'maint-3.7' from 'origin'.
Switched to a new branch 'maint-3.7'
Configuring: (100%) [=======================================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   chrono
--   atomic
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.17", minimum required is "2.7") 
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- Checking for module 'cppunit'
--   No package 'cppunit' found
-- Could NOT find CPPUNIT (missing: CPPUNIT_INCLUDE_DIRS) 
-- 
-- Configuring testing-support support...
--   Dependency CPPUNIT_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled testing-support but configuration checked failed
Call Stack (most recent call first):
  CMakeLists.txt:373 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/pybombs/bladeRF/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/pybombs/bladeRF/src/gnuradio/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.

```

velichkov, 2020-03-08:
Hi @talasalim,

> I'm following https://github.com/Nuand/bladeRF/wiki/Getting-Started:-Linux#Compile_GNU_Radio to download bladeRF and GNURadio but I'm having several issues with downloading it and after following some solutions from different threads, this error is still repetitive. 

You should report all those issue to Nuand as we are not the authors of this guide.

> ubuntu@ubuntu:~$ **pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default**
> [INFO] Prefix Python version is: 2.7.17

My advice is to reinstall pybombs using `pip3` in order to use python3. 

> [INFO] Using config file: /home/ubuntu/.pybombs/config.yml
> Setting keys for packages gnuradio:
> gitrev: maint-3.7

Which gnuradio version do you want to install - `3.7` or `3.8`?

> ubuntu@ubuntu:~$ pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default

In case you want to install `3.7` then you need to use `gnuradio-stable` and not `gnuradio-default` in the above receipe. 

> [WARNING] There already is a prefix in `/home/ubuntu/pybombs/bladeRF'.
> Continue using this path Y/[N]? y

It seems you have already run pybombs multiple times. My advice is before every attempt to remove the `prefix` path in order to start in a clean environment.

> Branch 'maint-3.7' set up to track remote branch 'maint-3.7' from 'origin'.
> Switched to a new branch 'maint-3.7'

> -- Configuring testing-support support...
> --   Dependency CPPUNIT_FOUND = FALSE
> CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
>   user force-enabled testing-support but configuration checked failed
> Call Stack (most recent call first):
>   CMakeLists.txt:373 (GR_REGISTER_COMPONENT)

You need to install `libcppunit-dev` package as the `gnuradio` from the `maint-3.7` branch depends on `cppunit`.

talasalim, 2020-03-23:
Apologies for the late reply however I adjusted some commands as you proposed and the command worked but now that I connected bladeRF it tells me that it does not open as permission is not granted, my bladeRF is connected and there is a green light but on the terminal this is what is says:

```
bladeRF-cli -i -v verbose
[VERBOSE @ libusb.c:598] Using libusb version: 1.0.21.11156
[DEBUG @ libusb.c:607] No devices available on the libusb backend.

No bladeRF device(s) available.

If one is attached, ensure it is not in use by another program
and that the current user has permission to access it.
```

velichkov, 2020-03-23:
Hi @talasalim,

Sorry, I can't help with this as I don't have and I've never used bladeRF. You better seek help in the [bladeRF](https://github.com/Nuand/bladeRF) project.

talasalim, 2020-03-23:
Alright thank you very much!

closed 2020-03-23 by talasalim