ptrkrysik/gr-gsm #508: gsgsm_scanner with R820T，nothing show up

jasonblackj, 2020-03-07:
Hi,
          I use the grgsm_scanner with R820T，but no station  display and grgsm_livemon works. I also try to use different samp_rate but it doesn't work.  please help me.

My OS：kali 2.0(physical machine)
library ：libosmocore+gr-osmosdr(installed)

here's the output:
root@xxx:/home/gr-gsm/apps# grgsm_scanner -b GSM900 -g 20 --samp-rate=0.8e6 -v -d

Args= 
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

[WARNING] SoapySSDPEndpoint failed join group udp://239.255.255.250:1900
  setsockopt(IP_ADD_MEMBERSHIP) [19: No such device]
[WARNING] SoapySSDPEndpoint failed join group udp://[ff02::c]:1900
  setsockopt(IPV6_ADD_MEMBERSHIP) [19: No such device]
Found Rafael Micro R820T tuner
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Invalid sample rate: 800000 Hz
Allocating 15 zero-copy buffers
Args=

velichkov, 2020-03-08:
Hi @jasonblackj,

> I use the grgsm_scanner with R820T，but no station display and grgsm_livemon works. 

Are you sure that there are GSM C0 Cells in the GSM900 band in the area where you are scanning?
Do you have a suitable antenna attached to your RTL dongle?
In case you are inside of a big building try moving near a window or outside where the signal is stronger.

> I also try to use different samp_rate but it doesn't work. please help me.
> 
> My OS：kali 2.0(physical machine)
> Library ：libosmocore+gr-osmosdr(installed)

How did you install gnuradio and gr-gsm and which versions?

> Here's the output:
> root@xxx:/home/gr-gsm/apps# grgsm_scanner -b GSM900 -g 20 --samp-rate=0.8e6 -v -d
>
> Args=
> [INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3 [...]

To suppress the above errors you can run `grgsm_scanner` with `--args=rtl` parameter.

ptrkrysik, 2026-11-09:
Scanner showing no cells is typically antenna/coverage/gain/sample rate related; please retry with the new release and use the mailing list for support.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik