ptrkrysik/gr-gsm #511: Grgsm_scanner not outputting any thing when using RTL-SDR

nano38, 2020-03-15:
I am using a Lubuntu 19.04 on a real machine. I installed the grgsm software using the `sudo apt-get install gr-gsm` command. When I used the command `sudo grgsm_scanner --band=GSM850 --gain=49.6 --verbose` I get no output (even when I go outside onto a high place). I know that the GSM850 band is used in the U.S. and my RTL-SDR supports that frequency. I also know that PCS1900 is used in the U.S. too, but the frequency is too high for my RTL-SDR. I have no idea what the issue is. Is the GSM850 band not used in my area, or is the software not working?

velichkov, 2020-03-16:
Hi @nano38,

> I know that the GSM850 band is used in the U.S. and my RTL-SDR supports that frequency.
> I have no idea what the issue is. Is the GSM850 band not used in my area, or is the software not working?

The GSM850 band is used only by AT&T and they [stopped their 2G network](https://about.att.com/innovationblog/2g_sunset) back in 2017 and that's why you don't find GSM cells.

> I also know that PCS1900 is used in the U.S. too, but the frequency is too high for my RTL-SDR.

Yes, the T-Mobile GSM network is in the PCS1900 band and for it you need an RTL-SDR device with Elonics E4000 tuner or another SDR device that supports that band.

nano38, 2020-03-18:
Oh, all right. So I need to use a SDR that has a wider frequency range. Thanks anyway.

closed 2020-03-18 by nano38