ptrkrysik/gr-gsm #512: Problems with maint-3.8 branch

cd-sdr, 2020-03-18:
I have just installed gr-gsm on Kali Rolling 2020.1 like this:

    apt install cmake autoconf libtool pkg-config build-essential python-docutils libcppunit-dev swig doxygen liblog4cpp5-dev python-gtk2 gnuradio-dev gr-osmosdr libosmocore-dev liborc-dev
    git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
    cd gr-gsm
    mkdir build
    cd build
    cmake ..
    make
    make install
    ldconfig
    export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH

but when I run `grgsm_scanner -d` I get the following error:

    Args= 
    [INFO] [UHD] linux; GNU C++ version 9.2.1 20200104; Boost_106700; UHD_3.15.0.0-2
    [ERROR] avahi_service_browser_new() failed: Bad state
    Found Rafael Micro R820T tuner
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
    Using device #0 Realtek RTL2838UHIDIR SN: 00000001
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Exact sample rate is: 2000000.052982 Hz
    [R82XX] PLL not locked!
    Allocating 15 zero-copy buffers
    OOOArgs= 
    [ERROR] avahi_service_browser_new() failed: Bad state

and then it repeats itself until it aborts.
Is this a dependency error?

velichkov, 2020-03-18:
Hi @cd-sdr,

> [ERROR] avahi_service_browser_new() failed: Bad state

To suppress this error run the `grgsm_scanner` with `--args=rtl` parameter as you are using an RTL-SDR device.
```
grgsm_scanner --args=rtl -d
```

> [R82XX] PLL not locked!

This message is normal and you can safely ignore it.

> and then it repeats itself until it aborts.

What do you mean by "it aborts"?

> Is this a dependency error?

No.

cd-sdr, 2020-03-19:
OK, thanks for the RTL argument, that's good to know.

What I mean is it finishes without find any frequencies. I remember using this a few years back on a different OS and it picked up 4 or 5 stations, and my phone says it has full signal here so I would expect it to find something. Is there a possibility it's not working? What can I do to test it?

velichkov, 2020-03-20:
Hi @cd-sdr,

> What I mean is it finishes without find any frequencies.

Do you have a suitable antenna attached to your RTL device?
In case you are inside of a big building try moving near a window or outside.

> I remember using this a few years back on a different OS and it picked up 4 or 5 stations,

In which country do you live? Have you tried scanning the other bands supported by your RTL device?

> and my phone says it has full signal here so I would expect it to find something. 

Is you phone in "2G Mode" only? Note that `gr-gsm` supports only 2G GSM and does not support CDMA, 3G or 4G LTE.

> Is there a possibility it's not working?

There are always such possibility.

ptrkrysik, 2026-11-09:
The avahi message is avoided with --args=rtl, and 'no cells found' is an antenna/location/band matter rather than a code defect.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik